arendst/Tasmota #5982: incorrect energy calculation sonoff pow r2

goosst, 2019-06-25:
### BUG DESCRIPTION
sorry, I had by accident typed my issue in an already closed topic: 
https://github.com/arendst/Sonoff-Tasmota/issues/5789#issuecomment-505553786

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : Sonoff pow r2
- [ ] Tasmota binary firmware version number used : 6.5 (latest available pre-compiled release)
- [x ] Development IDE - Compiler / Upload tools used : esptool
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
see: https://github.com/arendst/Sonoff-Tasmota/issues/5789#issuecomment-505553786

andrethomas, 2019-06-25:
1. The status 0 output in the referenced post appears incomplete. Please provide a complete output of Status 0.
2. Please also try the latest pre-compiled development binary from http://thehackbox.org/tasmota/
3. Can you verify the telemetry output with specific reference to the Period counter to see if this is in fact representative of the load you're expecting?

goosst, 2019-06-25:
1. apologies for not being complete:

> 21:35:43 CMD: status 0
21:35:43 MQT: stat/sonoffPowR2/STATUS = {"Status":{"Module":43,"FriendlyName":["SonoffPow"],"Topic":"sonoffPowR2","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:35:43 MQT: stat/sonoffPowR2/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"1T23:35:43","StartupUTC":"2019-06-23T20:00:00","Sleep":250,"CfgHolder":4617,"BootCount":10,"SaveCount":47,"SaveAddress":"F5000"}}
21:35:43 MQT: stat/sonoffPowR2/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:35:43 MQT: stat/sonoffPowR2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["telenet-00526",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
21:35:43 MQT: stat/sonoffPowR2/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
21:35:43 MQT: stat/sonoffPowR2/STATUS5 = {"StatusNET":{"Hostname":"sonoffPowR2-1052","IPAddress":"192.168.0.114","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"195.130.130.2","Mac":"80:7D:3A:33:44:1C","Webserver":2,"WifiConfig":4}}
21:35:43 MQT: stat/sonoffPowR2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.205","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_33441C","MqttUser":"stijn","MqttCount":7,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:35:43 MQT: stat/sonoffPowR2/STATUS7 = {"StatusTIM":{"UTC":"Tue Jun 25 19:35:43 2019","Local":"Tue Jun 25 21:35:43 2019","StartDST":"Sun Mar 24 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:47","Sunset":"21:57"}}
21:35:43 MQT: stat/sonoffPowR2/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:35:43 MQT: stat/sonoffPowR2/STATUS10 = {"StatusSNS":{"Time":"2019-06-25T21:35:43","ENERGY":{"TotalStartTime":"2019-06-24T00:00:00","Total":20.714,"Yesterday":1.616,"Today":19.099,"Power":4,"ApparentPower":15,"ReactivePower":15,"Factor":0.28,"Voltage":235,"Current":0.066}}}
21:35:43 MQT: stat/sonoffPowR2/STATUS11 = {"StatusSTS":{"Time":"2019-06-25T21:35:43","Uptime":"1T23:35:43","Vcc":3.140,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":74,"POWER":"ON","Wifi":{"AP":1,"SSId":"telenet-00526","BSSId":"5C:35:3B:70:05:2B","Channel":1,"RSSI":24,"LinkCount":10,"Downtime":"1T02:06:37"}}}

2. I'll give it a try when he stops mowing :)
3. Yes the output power is in the expected range (around 50 watt when charging the battery and in between some idle power to keep voltage on a wire). I might not fully get the comment about Period counter

ascillato, 2019-06-26:
Hi, please update to latest Tasmota (http://thehackbox.org/tasmota/). This issue has been fixed.

goosst, 2019-06-26:
thanks that fixed it

closed 2019-06-26 by goosst