arendst/Tasmota #5983: DS18B20 shows 85

tomasroud, 2019-06-25:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Sensor DS18B20 shows 85* after upgrade to version 6.5  
With version 6.11 it works fine.

### REQUESTED INFORMATION
I have one sensor DS18B20 plugged in by 2.5 jack so I allways set GPIO14 to DS18x20.
When I downgrade firmware back to 6.11 it works.
With version 6.21 it shows 85* again.

I tried to set all combinations with USE_DS18x20xxxx
I tried fresh firmware with same result.

I tried sonoff-sensor, sonoff-basic, sonoff-classic ... all same 85*

Sometimes it show right temperature after restart for a second and then again 85*

I tried core 2,3 and 2,5 too ... same result 85*

I tried development branch too ... same 85*

Please help me, what to set up, I am struggeling here 3 hours :(

Please do not recommend me to solder there 4,7 resistor, it must work as 6.11 firmware worked without soldering a resistor, if I use one sensor only.

tomasroud, 2019-06-25:
Now I can see that firware 6.1.1 shows first temperature 85* and then it is changed to right temperature and then it has right temperature for all time.

AlexTransit, 2019-06-25:
if the sensor is turned off, it will always show the temperature 85.
Use the library in which the sensor address is displayed. then it will be visible when it is disconnected.

ascillato2, 2019-06-26:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-26 by ascillato2

tomasroud, 2019-06-26:
I dont know anything about library or sensor address.
Could you be more specific please for lamas as me?