arendst/Tasmota #5984: xdrv_03_energy is not calculating current correctly when only voltage and power are available

twinotter, 2019-06-26:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
xnrg_01_hlw8012 is set up to handle not having a SEL line correctly, but xdrv_03_energy doesn't process this info correctly. I found this while working with the Etekcity Voltson ESW01. The ESW01 doesn't use the select pin of the HLW8012. Instead, it straps it high meaning CF1 is always reporting voltage.

You can get a bit of a work around using this template for the ESW01:

{"NAME":"ESW01-US","GPIO":[0,0,0,0,21,52,0,0,132,133,17,0,130],"FLAG":0,"BASE":6}

And you could simulate this on a SONOFF-POW by removing the SEL pin from its template and just outputting a 1 on that pin.

The result is a system that works, but needs to be calibrated. Put a known wattage bulb into a lamp. Plug it in, turn it on, issue the command PowerSet <watts of bulb>. Do a VoltageSet <your local voltage> to get that one too. (A Kill-A-Watt is great for getting these numbers.) (Warning: accuracy sucks when you're not drawing much current.)

Amps will always be wrong until this bug is fixed.

The code in HlwDrvInit() in xnrg_01_hlw8012.ino correctly sets energy_current_available and energy_voltage_available based on the pins you set, even if you don't select something for GPIO_NRG_SEL. It also measures the squarewaves on the CF1 pin correct (although it duplicates voltage onto the current numbers, since SEL can't actually be changed).

Unfortunately, the code in xdrv_03_energy.ino, especially in EnergyShow() doesn't deal with a situation where energy_current_available is false very well. It should be able to calculate current from the voltage and power results.

There's another issue too... When SEL is unset, nothing seems to work. It's as if the energy module just isn't running... although manually walking through the function pointer calls isn't showing me why. I'm not sure if it's because a SONOFF-POW has to have all three inputs (I can't find the code that does any check like that). Or what else could be going on.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff-POW and also Generic
- [X] Tasmota binary firmware version number used : 
10:56:38 MQT: stat/bathroomstars/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-03-19T12:47:29","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}} precompiled
- [X] Development IDE - Compiler / Upload tools used : Arduino / NodeMCU
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
10:57:41 MQT: stat/bathroomstars/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"bathroomstars","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":6,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:57:41 MQT: stat/bathroomstars/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:21:03","StartupUTC":"2019-06-26T04:36:38","Sleep":50,"CfgHolder":4617,"BootCount":32,"SaveCount":179,"SaveAddress":"F9000"}}
10:57:41 MQT: stat/bathroomstars/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-03-19T12:47:29","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
10:57:41 MQT: stat/bathroomstars/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXX","XXXXX"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
10:57:41 MQT: stat/bathroomstars/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16301C","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
10:57:41 MQT: stat/bathroomstars/STATUS5 = {"StatusNET":{"Hostname":"Sonoff 1","IPAddress":"192.168.1.237","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"XXXX","Webserver":2,"WifiConfig":4}}
10:57:41 MQT: stat/bathroomstars/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25F809","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:57:41 MQT: stat/bathroomstars/STATUS7 = {"StatusTIM":{"UTC":"Wed Jun 26 04:57:41 2019","Local":"Wed Jun 26 10:57:41 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+06:00","Sunrise":"09:48","Sunset":"01:58"}}
10:57:41 MQT: stat/bathroomstars/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:57:41 MQT: stat/bathroomstars/STATUS10 = {"StatusSNS":{"Time":"2019-06-26T10:57:41","ENERGY":{"TotalStartTime":"2019-06-22T16:44:05","Total":0.754,"Yesterday":0.342,"Today":0.412,"Power":62,"ApparentPower":646,"ReactivePower":643,"Factor":0.10,"Voltage":120,"Current":5.399}}}
10:57:41 MQT: stat/bathroomstars/STATUS11 = {"StatusSTS":{"Time":"2019-06-26T10:57:41","Uptime":"0T00:21:03","Vcc":3.543,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"XXXXX","BSSId":"XXXXX","Channel":1,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
see above. But simply: with a SONOFF-POW, disable SEL. Strap the SEL line to high. You should be able to calculate current. The code already detects the situation correctly. But the energy drivers don't use the information correctly.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Current should be calculated and displayed correctly.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

netpok, 2019-06-27:
It is not possible to calculate current from only power and current, because most devices aren't pure resistive loads. For non-resistive loads the power can be calculated from P=U*I*PF where power factor (PF) describes the phase angle between the voltage and current. So if your load is not a heating element or incandescent light bulb the reported current would be lower than the flowing current.

joba-1, 2019-06-27:
question is, what do you want to know?
I am charged for resistive load only, so power from capacitive/inductive load is just deko for me...

twinotter, 2019-06-27:
Two behaviors:

1. It's arguable whether or not to display a calculated current. I'm uncertain whether the power measurement the HLW8012 is producing is a Real Power measurement or not, but I think it is. (This is because it's only measuring current and voltage on the circuit, as far as I can tell.) If that's the case, then it would be appropriate to display Current as Power / Voltage.

If not, or if the developers would prefer not to address it, then nothing should be displayed, rather than an errored value.

2. The SEL line is not needed to get a power measurement or a voltage measurement. The code currently requires the SEL line in order to activate the display of power in xdrv_03_energy.ino.

arendst, 2019-06-30:
As answered before, you cannot deduce the current from the measured power and the voltage alone as it is dependant on resistive, capacitive or inductive load. So that rules out option 1.

The SEL line may be left undefined if the hardware is not using it. As the POW uses it to switch between voltage and current measurement you will have to define it. 

An example of device without SEL and CF1 configured is the Shelly 1PM which just outputs power

[image]

closed 2019-07-04 by ascillato2