arendst/Tasmota #5987: Help With Flashing

filoor, 2019-06-27:
Hi All.

I would like to have some help to Flash a switch.
I have tried tuya convert OTA but it wont work for this switch
How Must i connect it on which pins.
[image]
[image]
[image]
[image]

Thanks a Lot

tobox, 2019-06-27:
You should at least specify the exact name and model of the device you are trying to flash.

tobox, 2019-06-27:
Here you see all necessary connectors on the ESP8266 board; you may trace some of them to the 5 point header. Then solder wires to the remaining pins and follow the standard flash instructions.

https://user-images.githubusercontent.com/14855001/34749157-2fc38d00-f5a0-11e7-89c0-3cd91e443803.jpg

meingraham, 2019-06-27:
Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-06-27:
Google is your friend 🙄
[Module TYWE3S](https://docs.tuya.com/en/hardware/WiFi-module/wifi-e3s-module.html)

closed 2019-06-27 by ascillato2

filoor, 2019-06-27:
Thanks all