arendst/Tasmota #5988: ifan03

horviskoadrianko, 2019-06-27:
There have been release of new ifan03. It is an improved version from ifan02. Do some of you guys have it? Will there be support in tasmota?

bakman2, 2019-06-29:
The ifan03 has RF support and the base can be paired with the RF bridge, makes it an interesting device as tasmota already has support for the RF bridge.

dwest576, 2019-06-29:
I have a couple of ifan03.  Good news is I can flash it with a RPI w/o needing an external power source, or jumpering TP16 to GND, simply hold down the button while powering up the RPI.

It appears the ifan02 template doesn't work.  When I swap the module to ifan02, it will do a hard reset in about 30 seconds and revert back to factory default where I have to put in the wifi info again and it is back to basic.  Keeping it on basic never reset the device.  

When I swap to ifan02 and use the web interface to turn off/on, I can hear the relays click for Light, Speeds 2/3 and when I press off, I heard the relays go off.  I don't hear the relay on setting 1.

arendst, 2019-07-01:
I have ifan03 on the way. Once arrived I will (try to) implement in Tasmota.

patoroco, 2019-07-01:
I'm waiting too for the arrival of my ifan03 :D

anuadu, 2019-07-03:
iFan02 on tasmota had served me well for a year, now cool looking iFan03 arrives just in time....all we need is tasmota. 
@arendst I don't know how we'd have survived without you, keep up your good work!

bertmagin, 2019-07-03:
iFan03 has identical exterior dimensions as iFan02.  I could not even force fit into standard (US - Hampton Bay)_ mounting bracket.  A millimeter less width and this would be perfect.

Sawadee2u, 2019-07-09:
This is the Serial Info from an Ifan03 after connected and configured with template for Ifan02. It take about 3 minutes and a Reset is issued:
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-4257 with IP address 192.168.123.172
23:43:09 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
23:43:09 CFG: Use defaults
23:43:11 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 CFG: Loaded from flash at F9, Count 3
00:00:00 Project sonoff Sonoff Version 6.6.0(release-sonoff)-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-4257 with IP address 192.168.4.1

Above was with wificonfig set to 0.   Have tried 1 and 2 with same result.

matsube11, 2019-07-09:
I’ve received my ifan03 to replace all my ifan02, the new remote is clearly a great benefit over the previous one, installed one and waiting for tasmota implementation to flash them.

Sawadee2u, 2019-07-09:
I know there is a lot more work to do but:                                                           (Using Ifan02 Template)
Changed GPIO14 from Button4 (20) to Button4i (125) and resetting stopped.
Setting to Button4n resets as before
Setting to None(0) stops resetting also.

Sawadee2u, 2019-07-10:
Also. After installing tasmota 6.6.0. can't connect remote.

Sawadee2u, 2019-07-10:
Pics of Ifan03 if interested:
[image]
[image]
[image]
[image]
[image]

arendst, 2019-07-11:
I have the ifan03 too and can confirm the ifan02 template doesn't work. ITo me i seems the relays are not all controlled by the esp8266 but as I don't have schematics it's a wild guess.

If someone wants to look under the shield to see if the same microcontroller (as on ifan02) is there would make it easier.

For now implementing will be a challenge.

BTW even with the original firmware pairing the remote is/was a challenge too; I only managed to pair it once. After a power cycle the pairing was gone.

Sawadee2u, 2019-07-11:
Will have pics shortly
Removing shield

Sawadee2u, 2019-07-11:
Removed shield with heat gun narrow snout at 650f. Removed plastic push button first.
Hopefully not destructive.
These are quick pics. Let me know if you need better views or anything else at all. Really would like to get this working. Thanks

[image]
[image]

arendst, 2019-07-11:
Pics are fine. Pictures show a much smaller microcontroller than the ifan02 so things have definitly changed around button and relay control.

Sawadee2u, 2019-07-11:
Here's one with better contrast and focus.
[image]

Sawadee2u, 2019-07-11:
I think this is the controller chip BB10F8G
https://www.silabs.com/documents/public/data-sheets/efm8bb1-datasheet.pdf

bakman2, 2019-07-11:
@Sawadee2u does the ifan03 remote talk to the sonoff rf bridge (if you have one) ?

Sawadee2u, 2019-07-11:
I have one but haven't tried yet. Just go it. Same freq so should with the correct codes set.

arendst, 2019-07-11:
@bakman2 Yes the bridge perfectly sees all switches from the remote.

@Sawadee2u  Yes probably is as iTead uses it often. No need to investigate the chip or code; all we need is how it communicates with the esp8266 and how are the buttons and relays connected.

In iFan02 all are connected to the microcontroller and the esp8266 also has eight connections to the microcontroller (See logic ifan02). I see those same connections coming from the esp8285 this time too.

arendst, 2019-07-11:
You might want to try the below template:
```
{"NAME":"Sonoff iFan03","GPIO":[17,255,0,255,0,0,21,33,31,56,30,32,0],"FLAG":0,"BASE":44}
```

Sawadee2u, 2019-07-11:
Lookin' better. Light relay and all three fan speed relays working.
Sorry or the delay, was repairing a fridge for daughter.

arendst, 2019-07-12:
By using the template shown above the microcontroller is bypassed. This is not the end goal so I will need more time to investigate a final solution.

Sawadee2u, 2019-07-12:
This is great. Thank you again for all your hard work. If I can be of any use please let me know.

matsube11, 2019-07-13:
> You might want to try the below template:
> 
> ```
> {"NAME":"Sonoff iFan03","GPIO":[17,255,0,255,0,0,21,33,31,56,30,32,0],"FLAG":0,"BASE":44}
> ```

Hi all

I just tried this template with Tasmota 6.6.0 freshly flashed on an ifan03 and got everything mixed up.
light shows OFF when ON, and fan speed 3 is 1, 1 is 3, 2 is 2 and 0 (OFF) is 3 so not possible to turn off the fan.

Any clue about why ?

I'll flash back the original firmware in the meantime

arendst, 2019-07-13:
Then try this:
```
{"NAME":"Sonoff iFan03","GPIO":[17,255,0,255,0,0,29,33,31,56,32,30,0],"FLAG":0,"BASE":44}
```

arendst, 2019-07-13:
iFan03 info so far:
- Where the ifan02 was controlled by the second microcontroller all control for the ifan03 has been moved to the esp8285; the microcontroller is only used for RF decoding.
- The microcontroller interfaces with esp8285 using the serial interface 9600 bps 8N1. I managed to trace all codes from the remote with handshaking between the two.
- As the esp8285 does the fan relay sequencing I have to implement this in Tasmota too.
- The buzzer is now controlled by the esp8285 where it can easily be enabled/disabled (GPIO10)

So it will take some time before I have it all put together but at least I think I have the info I need.

Stay tuned.

arendst, 2019-07-13:
Latest template:
```
{"NAME":"Sonoff iFan03","GPIO":[17,255,0,255,0,0,29,33,23,56,22,24,0],"FLAG":0,"BASE":44}
```

matsube11, 2019-07-13:
Thanks,

All is good now with latest template, light and fan speed are correct :)

Just have to find a way to make the remote paired again with Tasmota, worked fine with the original firmware

arendst, 2019-07-13:
I have to find time and write code for the remote to make it functional again.

bakman2, 2019-07-13:
@arendst  Can it be powered via FTDI for flashing or does it require more power ?

arendst, 2019-07-13:
Yes, as long as you do not toggle relays.

bakman2, 2019-07-14:
> Yes, as long as you do not toggle relays.

Thanks I was able to flash it, in order to see it coming up does it need to be connect to mains as I don't see it when powered from usb.

arendst, 2019-07-14:
Mine came up just fine when connected to USB.

bakman2, 2019-07-14:
> Mine came up just fine when connected to USB.

Got it, used incorrect flash mode option (DIO) during flash. 
Template appears to be working, nice!

arendst, 2019-07-14:
Try the latest dev release. Pre-compiled will be downloadable in a few hours.

- It should support all features of the iFan03 when module 71 is selected. 
- No need for the template as shown earlier.
- Pairing remote works stable when iFan03 is powered off, remote button is pressed and held, iFan03 is powered on, remote button is released after a few seconds.
- Make sure to keep seriallogging disabled as it is used by the RF remote.

Give it a go.

Sawadee2u, 2019-07-14:
Wow, This is great. Thanks for this. Really wanted the remotes. Wife not an ‘automation’ fan....
Great work.

Sawadee2u, 2019-07-14:
Re-soldered the shield I removed a few day ago for pics and loaded 6.6.0.2-2.3.0. with ifan03 not plugged in. Held fan speed III down (any button should work) and plugged in power for ifan03. Kept button pressed for 15 seconds after powering on and all paired great.
Thanks again for your work and time spent on this. Much appreciated as always.

bakman2, 2019-07-14:
I could not detect the remote in the RF bridge (running with portish in rfraw 1), is this due to  the portisch firmware? (Not a big if the remote is detected by the ifan03)

arendst, 2019-07-14:
I have a portisch fw too and it detects just fine:
```
20:56:44 MQT: tele/bridge2/RESULT = {"RfReceived":{"Sync":12060,"Low":340,"High":1130,"Data":5147988,"RfKey":"None"}}
20:56:52 MQT: tele/bridge2/RESULT = {"RfReceived":{"Sync":12100,"Low":380,"High":1150,"Data":5147986,"RfKey":"None"}}
```

arendst, 2019-07-14:
@Sawadee2u glad it works. I'll close this issue.

closed 2019-07-14 by arendst

bpmittal, 2019-07-21:
> * 
> * Pairing remote works stable when iFan03 is powered off, remote button is pres

can we use the buzzer in this ifan03 , as a saperate entity ?
so that we can use the buzzer for other activity ?

arendst, 2019-07-21:
Yep

anuadu, 2019-07-22:
Hi Guy, Can you point out the GPIO0 pin for me, is it available on PCB or do we have to solder directly on the esp chip legs?

arendst, 2019-07-22:
No need as the button is connected to GPIO00. Just keep the button pressed during flashing procedure.

anuadu, 2019-07-22:
@arendst too easy, thanks

geejayem99, 2019-07-22:
You mentioned precompiled available. Can you point me to where that is? Can't see module 71 in the latest release sonoff.bin. Or do I just need to wait (or compile it myself). Thanks

arendst, 2019-07-22:
See README.md and this http://thehackbox.org/tasmota/

bpmittal, 2019-07-22:
if the buzzer can be used as a separate entity then how to get that entity in HA , 
i only get 5 entities in my HA when auto discovered, 

1 is light switch
2 is fan switch
3 is fan switch
4 is fan switch
5 is wifi rssi sensor

one fan switch entity is missing and one buzzer entity is missing , how to get those ?

finity69x2, 2019-07-23:
For anyone that has one of these that is in the US, can you say whether the speeds of the fan correspond to the usual speeds of the usual US fan speeds? 

There was an issue with the iFan02 where the fan speeds for medium and low were a lot slower than the normal US fan speeds requiring a new capacitor to be installed to get the speeds better.

arendst, 2019-07-23:
As far as I know the capacitors are the same value as in the iFan02 so no change.

bpmittal, 2019-07-23:
how to configure ifan03 with Home assistant ?
 i am unable to configure it with my HA .

dwest576, 2019-07-23:
Won't you use MQTT like other Tasmota devices?

bpmittal, 2019-07-23:
my two other tasmota nodes are working well with same HA , 
but after flashing tasmota in this ifan03 all works well within tasmota web server, but the entities which are auto discovered in HA does not works perfectly .

anuadu, 2019-07-24:
@bpmittal , not sure about auto discovery. but if you are using MQTT and assuming you have topic  setup as 'sonoff_fan', use the following configuration in [https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config#configure-ifan02](url)

I used code from Example 1 for Fan and the bit from Example 2 for light component. 
and bit of lovelace tweaking it looks like this
[image]

finity69x2, 2019-07-24:
> As far as I know the capacitors are the same value as in the iFan02 so no change.

Thanks for the info.

nonecool, 2019-07-27:
I can confirm, the capacitors are the same and will have to be replaced for use with fans in the USA.  The new ifan03 is a whole lot easier to flash and the prerelease tasmota firmware is a must.  no more rubber underwear for anyone!

DonnyBahama, 2019-08-05:
Can someone post a link to details/procedure for replacing the capacitors for 60Hz use? Thanks!

RoboCop0815, 2019-08-08:
I do not get the remote control connected ...
I have the iFan03 (with the fan connected) disconnected via a switch from the mains and hold the III or the wireless link Tate -> turn on iFan03 and count to 15, release the button, but no function of the remote control

tulkas84, 2019-08-08:
> 
> 
> I do not get the remote control connected ...
> I have the iFan03 (with the fan connected) disconnected via a switch from the mains and hold the III or the wireless link Tate -> turn on iFan03 and count to 15, release the button, but no function of the remote control

You need the last precomplied tasmota (mine is 6.6.0.3) and be sure not to use the old profile. 
With that setup, i paired the remote just like you said.

[152 more replies]