arendst/Tasmota #5989: Repeating rule on hold

k4roshi, 2019-06-27:
I'm trying to use a rule to trigger increments in brightness in a light by holding the button.
I was trying to use this rule:
`Rule1 on button1#state=3 do add1 20 endon on var1#state>100 do var1 0 endon on var1#state do publish milight/commands/0x241B/fut089/1 {"level":%var1%} endon`
Followed by a
```
Rule1 1
Rule1 4 
```
To continuously trigger the rule while the button is being kept pressed, but it doesn't work. 
The rule works only once and i have to press the button again to trigger it.
I'm using tasmota 6.5.0. Am I doing something wrong?

ascillato, 2019-06-27:
The action you want to perform is not supported by rules. Rules are triggers: when the condition is met, it just executes once. 

Try scripting. For that you will need to compile by yourself disabling rules and enabling scripting. 

If you need further assistant on that, please find us in the Tasmota support chat. Thanks

closed 2019-06-29 by ascillato2

ferbulous, 2022-11-09:
Hi @k4roshi 
Did you manage to setup the script for this custom action?

k4roshi, 2022-11-10:
No, I'm sorry. I adopted a different approach at the time and now have a different set of dimmers altogether that support this use case natively.