arendst/Tasmota #5990: Display Mode 2 with local sensors	but it still received all tele message from other

wongnam, 2019-06-28:
### BUG DESCRIPTION
- I noted that in **Display Mode 2** with local sensors massages but it still received all tele messages from other in background.
- It should be stop receive tele message in **Display Mode 2** from other devices to save resource and traffics.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : NodeMCU + OLED SSD1366 + PZEM004T
- [x] Tasmota binary firmware version number used : self-compiled
- [x] Development IDE - Compiler / Upload tools used : VSC + PlatformIO IDE
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
17:26:21 CMD: status 0
17:26:21 SRC: WebConsole from 192.168.12.108
17:26:21 RSL: Received Topic /status, Data Size 1, Data 0
17:26:21 RSL: Group 0, Index 1, Command STATUS, Data 0
17:26:21 MQT: stat/pumpmonitor/STATUS = {"Status":{"Module":18,"FriendlyName":["Pump Monitor"],"Topic":"pumpmonitor","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":7,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:26:21 MQT: stat/pumpmonitor/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"","RestartReason":"Software/System restart","Uptime":"0T00:24:35","StartupUTC":"2019-06-28T10:01:46","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":86,"SaveAddress":"F9000"}}
17:26:22 MQT: stat/pumpmonitor/STATUS2 = {"StatusFWR":{"Version":"6.5.0.15(sonoff)","BuildDateTime":"2019-06-08T09:41:12","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
17:26:22 MQT: stat/pumpmonitor/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-1","Wong"],"TelePeriod":10,"Resolution":"55A180C0","SetOption":["00208009","280500000100060000000000000000000000","00000000"]}}
17:26:22 MQT: stat/pumpmonitor/STATUS4 = {"StatusMEM":{"ProgramSize":498,"Free":504,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F28A384","041001A0","000000CC","000002C0"]}}
17:26:22 MQT: stat/pumpmonitor/STATUS5 = {"StatusNET":{"Hostname":"pumpmonitor-0307","IPAddress":"192.168.12.102","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"DC:4F:22:58:41:33","Webserver":2,"WifiConfig":4}}
17:26:22 MQT: stat/pumpmonitor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_584133","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:26:22 MQT: stat/pumpmonitor/STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 28 10:26:22 2019","Local":"Fri Jun 28 17:26:22 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"05:33","Sunset":"18:18"}}
17:26:22 MQT: stat/pumpmonitor/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:26:22 MQT: stat/pumpmonitor/STATUS10 = {"StatusSNS":{"Time":"2019-06-28T17:26:22","ENERGY":{"TotalStartTime":"2019-05-26T07:53:24","Total":23.089,"Yesterday":0.466,"Today":0.414,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":225,"Current":0.000}}}
17:26:22 MQT: stat/pumpmonitor/STATUS11 = {"StatusSTS":{"Time":"2019-06-28T17:26:22","Uptime":"0T00:24:36","SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:04"}}}
17:26:22 SRC: MQTT
17:26:22 RSL: Received Topic tele/energy1/SENSOR, Data Size 236, Data {"Time":"2019-06-28T17:26:22","ENERGY":{"TotalStartTime":"2019-06-24T00:00:00","Total":95.884,"Yesterday":19.467,"Today":14.188,"Period":2,"Power":779,"ApparentPower":858,"ReactivePower":360,"Factor":0.91,"Voltage":228,"Current":3.770}}
17:26:22 DMP: A2 00 00 00 00 00 A2
17:26:22 DMP: A3 01 21 F0 00 00 B5

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
17:26:22 DMP: A2 00 00 00 00 00 A2
17:26:22 DMP: A3 01 21 F0 00 00 B5
17:26:23 DBG: [Current      0.000A  ]
17:26:23 DMP: A0 00 E1 06 00 00 87
17:26:23 UDP: Packet (199)
17:26:23 UDP: Packet (199)
17:26:23 UDP: Packet (199)
17:26:23 DMP: A1 00 00 00 00 00 A1
17:29:38 DMP: A3 01 21 F0 00 00 B5
17:29:39 MQT: tele/pumpmonitor/STATE = {"Time":"2019-06-28T17:29:39","Uptime":"0T00:27:53","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:29:39 MQT: tele/pumpmonitor/SENSOR = {"Time":"2019-06-28T17:29:39","ENERGY":{"TotalStartTime":"2019-05-26T07:53:24","Total":23.089,"Yesterday":0.466,"Today":0.414,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":226,"Current":0.000}}
17:29:39 DBG: [Current      0.000A  ]
17:29:39 SRC: MQTT
17:29:39 RSL: Received Topic tele/pumpmonitor/STATE, Data Size 230, Data {"Time":"2019-06-28T17:29:39","Uptime":"0T00:27:53","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:29:39 SRC: MQTT
17:29:39 RSL: Received Topic tele/pumpmonitor/SENSOR, Data Size 228, Data {"Time":"2019-06-28T17:29:39","ENERGY":{"TotalStartTime":"2019-05-26T07:53:24","Total":23.089,"Yesterday":0.466,"Today":0.414,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":226,"Current":0.000}}
17:29:39 SRC: MQTT
17:29:39 RSL: Received Topic tele/littlesocket/STATE, Data Size 231, Data {"Time":"2019-06-28T17:29:39","Uptime":"1T05:13:05","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":80,"LinkCount":59,"Downtime":"0T00:06:05"}}
17:29:39 SRC: MQTT
17:29:39 RSL: Received Topic tele/littlesocket/SENSOR, Data Size 230, Data {"Time":"2019-06-28T17:29:39","ENERGY":{"TotalStartTime":"2019-06-08T09:11:32","Total":29.679,"Yesterday":2.054,"Today":1.343,"Period":0,"Power":9,"ApparentPower":42,"ReactivePower":41,"Factor":0.21,"Voltage":221,"Current":0.189}}
17:29:39 WIF: Checking connection...
17:29:39 WIF: Connected
17:29:39 SRC: MQTT
17:29:39 RSL: Received Topic tele/groundfloorlight2/STATE, Data Size 230, Data {"Time":"2019-06-28T17:29:39","Uptime":"1T05:13:04","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:53"}}
17:29:39 DMP: A0 00 E2 03 00 00 85
17:29:39 DMP: A1 00 00 00 00 00 A1
17:29:39 SRC: MQTT
17:29:39 RSL: Received Topic tele/groundfloorlight2/SENSOR, Data Size 45, Data {"Time":"2019-06-28T17:29:39","Switch1":"ON"}
17:29:40 DMP: A2 00 00 00 00 00 A2
17:29:40 DMP: A3 01 21 F0 00 00 B5
[... 29 lines cut ...]
17:29:48 UDP: Packet (199)
17:29:48 UDP: Packet (199)
17:29:49 DMP: A0 00 E2 01 00 00 83
17:29:49 DMP: A1 00 00 00 00 00 A1
17:29:50 DMP: A2 00 00 00 00 00 A2
17:29:50 DMP: A3 01 21 F0 00 00 B5
17:29:51 DBG: [Factor       0.00    ]
17:29:51 DMP: A0 00 E2 01 00 00 83
17:29:51 SRC: MQTT

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-07-01:
You're right. Looking at the code I think I see a feature/bug. Working on it...

ascillato, 2019-07-01:
@wongnam 

Hi,

Theo has done a fix. Please, could you test it? Thanks

wongnam, 2019-07-02:
Hi, Tested and confirm it'working fine now.

Thanks.

closed 2019-07-02 by wongnam