arendst/Tasmota #5991: HOSTNAME with serial number as decimal and without MQTT_TOPIC

0n3-70uch, 2019-06-28:
**Have you look for this feature in other issues and in the wiki?**
Yes. Wiki and Discord.

**Is your feature request related to a problem? Please describe.**
No. Only bad charactes in the hostname (like: "_").

**Describe the solution you'd like**
Change the MQTT_TOPIC to "PROJECT "_%06X"" in user_config_override.h. Now the Hostname is like "sonoff_E92J29-2383". Please add an option to set the hostname back to normal. Example:

```
#undef HOSTNAME
#define HOSTNAME     Project "-%04d"
```

ascillato, 2019-06-28:
Hi, 

>Please add an option to set the hostname back to normal.

It is already available the console command `hostname` for changing it.

0n3-70uch, 2019-06-28:
No. Not working if the "MQTT_TOPIC" is changed.

ascillato, 2019-06-28:
Sorrt, I could not reproduce the problem.

I have just tested `hostname` command and works fine in latest version. `Hostname 1` put the default defined in _my_user_config.h_

0n3-70uch, 2019-06-28:
Ok please try this:

Add this in your user_config_override.h
```
#undef MQTT_TOPIC
#define MQTT_TOPIC   PROJECT "_%06X"
```

Then it isn't possible to change the HOSTNAME back to "sonoff-1234".

meingraham, 2019-06-28:
Is the problem that you have a space in the definition of your MQTT_TOPIC? Topics cannot have any whitespace.

0n3-70uch, 2019-06-28:
No there isn't any space. My Topic is asolutly fine with sonoff_A1B2C3. But the hostname is generated by my topic. And so I have this hostname: "sonoff_A1B2C3-1234". It's impossible to remove the unecessary "_A1B2C3" via the user_config_override.h.

agreen, 2019-06-28:
This confused me a bit at first. 

I wanted to change my mqtt topics to match the hostname I was currently seeing, so I changed the mqtt_topic to sonoff-%04d  Now my hostname is actually sonoff-%04d-%04d (sonoff-8001-8001). What I was expecting is to have the mqtt topic and the hostname match exactly.  In @0n3-70uch's case, I think he wants to have his hostname be the default "sonoff-8001" and his MQTT topic be "sonoff_E92J29".

I've even tried setting my hostname to "sonoff-%04d" and I STILL get "sonoff-8001-8001" as my actual hostname.

I would suggest a mode or some other feature to have your hostname and MQTT name match exactly and as well have the hostname NOT contain the %04d automatically as is the case currently.

ascillato, 2019-06-29:
> Ok please try this:
> 
> Add this in your user_config_override.h
> 
> #undef MQTT_TOPIC
> #define MQTT_TOPIC   PROJECT "_%06X"
> Then it isn't possible to change the HOSTNAME back to "sonoff-1234" via the user_config_override.h.
> 
> PS: You can manual rename the hostname. But this is very uncomfortable.

I already try that as my previous comment and the command `hostname` works fine in the console. You can set whatever you want as hostname and as mqtt topic (using `topic` command).

You want your hostname to be equal to your mqtt_topic, right?
Actually, Tasmota config does the other way round. It uses your mqtt topic to make your hostname name.
Hostname is defined in the config as `%s-%04d` in line 78 of the file _sonoff.h_
That means that your hostname will be [TOPIC]_[last 4 digits of the MAC]
So, you are just changing your topic to be SONOFF_MAC, producing a hostname SONOFF_MAC_MAC

If you want to define a MQTT_TOPIC and make your HOSTNAME equals to your MQTT_TOPIC, and you want to make it configurable from _user_override._h or _my_user_config.h_ file, just:

1- Comment or delete line 78 of _sonoff.h_
2- Add in your _user_override.h_ or _my_user_config.h_ file the following line:
`#define WIFI_HOSTNAME "%s"`
(Tasmota will replace the %s with your mqtt topic)

closed 2019-06-29 by ascillato2

0n3-70uch, 2019-06-30:
Hi, thanks for your very helpful answere. I changed my sonoff.h to this:
`//const char WIFI_HOSTNAME[] = "%s-%04d";     // Expands to <MQTT_TOPIC>-<last 4 decimal chars of MAC address>
const char WIFI_HOSTNAME[] = "sonoff-%04d";`

The HOSTNAME is someting like this now: sonoff-1073680949
How I can reduce the HOSTNAME to this: sonoff-1073

Greetings Fabian

ascillato, 2019-06-30:
Which esp8266 core version are you using?

0n3-70uch, 2019-06-30:
I use 2.5.2 (platformio).

ascillato, 2019-06-30:
Please try with core 2.3.0 so as to test. I tested on 2.3.0 and %04d gives me the last 4 digits of the mac correctly

0n3-70uch, 2019-06-30:
Ok I will tyr it later. I got a compiler error with 2.3.0 at the moment.

0n3-70uch, 2019-07-07:
Hi, I compiled it with core 2.3.0. Still the same problem.

ascillato, 2019-07-12:
Hi,

Sorry for the late reply. I have missed your comment. 

Please, erase the flash before flashing (or do a `reset 5` in the console), seems that there is wrong data left on your device.

0n3-70uch, 2019-07-13:
Hi, no problem. Thanks for your help. I erased the devices several times with `reset 2`.
And I reflashed one device (`esptool.py -p /dev/ttyUSB0 erase_flash` and `esptool.py -p /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x00000 'sonoff.bin'`). Still the same problem...

ed-holland, 2020-01-19:
I'm having the same problem. I'm building tasmota.bin using 8.1 with only changes in my_user_config.
`#define WIFI_HOSTNAME          "sonoff-s20-%04d"`
results in "sonoff-s20-1073681064" instead of "sonoff-s20-5014". Resetting does not help.

The resolving of "%04d" does not seem to show the result in the ui. I'm using it for the friendly name, mqtt-topic also. The UI shows it without resolving.

Any update / workaround?

I would like to use this binary for all my S20s.

mgaueca, 2022-03-04:
same problem here, %04d" doesn't works...any workaround?

barbudor, 2022-03-04:
It works as documented. Did you took the time to read the docs?
> If hostname contains % it will be reset to the default instead.

So having % in the name does the same 'hostname 1'
Nowhere it is written that '%04d' would be substituted