arendst/Tasmota #5992: Sichler Ventilator mit TYWE3S Template

Nordlicht77, 2019-06-28:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
Hallo
Kann mir jemand helfen habe diesen Lüfter gekauft 
https://www.pearl.de/mtrkw-10503-wlan-turmventilatoren.shtml
Dort ist ein TYWE3S verbaut
Den konnte ich erfolgreich mit Tuya-Convert flashen
Mit den Template Tuya Dimmer kann ich den Lüfter an und aus schalten
Er hat aber noch andere Schalter siehe Bild
Ich habe es über das Template Generic Module probiert das ich jeden GPIO ein Relay 1-8 zugeordnet habe
Leider keine Reaktion, wenn ich auf die Buttons in Web interface klicke
Kann mir jemand helfen , wie ich weiter vorgehen muss, das alle Funktion von Ventilator gehen 
nicht nur On off auch speed low mid high e.t.c
  grüß
Martin

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [x ] Tasmota binary firmware version number used : 6.5.0.16(6ac1904-basic) and 6.5.0 basic____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
11:52:06 RSL: Group 0, Index 1, Command STATUS, Data 0
11:52:06 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["lüfter","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":3,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:52:06 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020502/sonoff-minimal.bin","RestartReason":"Exception","Uptime":"0T00:11:55","StartupUTC":"2019-06-28T10:40:11","Sleep":50,"CfgHolder":4617,"BootCount":659,"SaveCount":2789,"SaveAddress":"FB000"}}
11:52:06 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.16(6ac1904-basic)","BuildDateTime":"2019-06-21T20:09:37","Boot":4,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
11:52:06 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["FRITZ7560UD2",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["0050000B","280502000100060000000000000000000000","00002810"]}}
11:52:06 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":465,"Free":536,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0F082382","001083A0","20021706","010001C0"]}}
11:52:06 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1015","IPAddress":"192.168.0.118","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"68:C6:3A:E2:23:F7","Webserver":2,"WifiConfig":4}}
11:52:06 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E223F7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:52:06 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 28 10:52:06 2019","Local":"Fri Jun 28 11:52:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:49","Sunset":"20:57"}}
11:52:06 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-06-28T11:52:06"}}
11:52:06 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-06-28T11:52:06","Uptime":"0T00:11:55","Vcc":3.435,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"POWER1":"ON","POWER2":"ON","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
Schalter Power on

12:59:38 TYA: RX Packet: "55aa03070005010100010112"
12:59:38 TYA: RX Device-1 --> MCU State: On Current State:On
12:59:38 TYA: RX Packet: "55aa03070005030400010016"
12:59:38 TYA: RX Device-3 --> MCU State: Off Current State:Off
12:59:38 TYA: RX Packet: "55aa03070005020400010015"
12:59:38 TYA: RX Device-2 --> MCU State: Off Current State:Off
12:59:38 TYA: RX Packet: "55aa03070005080100010018"
12:59:38 TYA: RX Device-8 --> MCU State: Off Current State:Off
12:59:40 TYA: TX Packet: "55aa00000000ff"
12:59:40 TYA: RX Packet: "55aa030000010104"

Schalter Power off

13:00:57 TYA: RX Packet: "55aa03070005010100010011"
13:00:57 TYA: RX Device-1 --> MCU State: Off Current State:On
13:00:57 SRC: Switch
13:00:57 TYA: Send Dim Level skipped due to 0 or already set. Value=0
13:00:57 RSL: RESULT = {"POWER1":"OFF"}
13:00:57 RSL: POWER1 = OFF
13:00:57 TYA: RX Packet: "55aa030700050b040001001e"
13:00:57 TYA: RX Device-11 --> MCU State: Off Current State:Off
13:00:57 TYA: RX Packet: "55aa030700080c0200040000000023"
13:00:57 TYA: RX Dim State=0
13:00:57 TYA: TX Packet: "55aa00000000ff"
13:00:57 TYA: RX Packet: "55aa030000010104"

Schalter Speed low to mid

13:04:03 TYA: RX Packet: "55aa03070005030400010016"
13:04:03 TYA: RX Device-3 --> MCU State: Off Current State:Off
13:04:03 TYA: RX Packet: "55aa03070005020400010116"
13:04:03 TYA: RX Device-2 --> MCU State: On Current State:Off
13:04:03 SRC: Switch
13:04:03 RSL: RESULT = {"POWER2":"ON"}
13:04:03 RSL: POWER2 = ON
13:04:04 TYA: TX Packet: "55aa00000000ff"
13:04:04 TYA: RX Packet: "55aa030000010104"

Schalter Speed mid to high

13:04:47 TYA: RX Packet: "55aa03070005030400010016"
13:04:47 TYA: RX Device-3 --> MCU State: Off Current State:Off
13:04:47 TYA: RX Packet: "55aa03070005020400010217"
13:04:47 TYA: RX Device-2 --> MCU State: On Current State:On
13:04:47 SRC: Switch
13:04:47 RSL: RESULT = {"POWER2":"OFF"}
13:04:47 RSL: POWER2 = OFF
13:04:47 CFG: Saved to flash at F6, Count 2802, Bytes 3584
13:04:48 TYA: TX Packet: "55aa00000000ff"
13:04:48 TYA: RX Packet: "55aa030000010104"

Schalter Speed high to low

Schalter Speed mid to high

13:06:44 TYA: RX Packet: "55aa03070005030400010016"
13:06:44 TYA: RX Device-3 --> MCU State: Off Current State:Off
13:06:44 TYA: RX Packet: "55aa03070005020400010015"
13:06:44 TYA: RX Device-2 --> MCU State: Off Current State:Off

Schalter Swing off to on 

13:07:27 TYA: RX Packet: "55aa03070005080100010119"
13:07:27 TYA: RX Device-8 --> MCU State: On Current State:Off
13:07:27 SRC: Switch
13:07:27 RSL: RESULT = {"POWER1":"ON"}
13:07:27 RSL: POWER1 = ON

Schalter Swing on to off

13:07:54 TYA: RX Packet: "55aa03070005080100010018"
13:07:54 TYA: RX Device-8 --> MCU State: Off Current State:Off
13:07:55 TYA: TX Packet: "55aa00000000ff"
13:07:55 TYA: RX Packet: "55aa030000010104"

```
**(Please, remember to close the issue when the problem has been addressed)**

[image]
[image]
[image]
[image]
[image]

ascillato, 2019-06-28:
Sorry, I don't understand. Can you translate your issue?

Nordlicht77, 2019-06-28:
google translate

Hello
Can someone help me bought this fan
https://www.pearl.de/mtrkw-10503-wlan-turmventilatoren.shtml
There is a TYWE3S installed
I was able to successfully flicker it with Tuya-Convert
With the Template Tuya Dimmer I can switch the fan on and off
But he has other switches see picture
I tried the Template Generic Module which assigned a Relay 1-8 to each GPIO
Unfortunately, no reaction when I click on the buttons in Web interface
Can anyone help me, as I have to proceed, that all function of fan go
not only on off but also speed low mid high e.t.c
greet

meingraham, 2019-06-28:
Likely that many functions on this device are handled by a secondary MCU rather than directly by the ESP GPIO. Unless you reversed engineered the serial communication data exchange, you will not be able to replicate it now that you have flashed Tasmota on your device.

Since you flashed your device with Tuya-Convert, it should have downloaded a copy of the original firmware. You'll need to flash that firmware back on the device. Unfortunately, you will not be able to do that OTA. You will need to connect wires to the device and use the serial flashing method. Once you successfully put the original firmware on the device, you will need to capture the serial codes the device uses to control the functions. Once you have captured those, you can put Tasmota back on the device and then code the necessary rules to replicate the functions.

As this is not a bug nor a feature request, please close this issue. If you need further assistance, please use the Discord Chat (see below). However, please know that most exchanges on the chat will need to be in English. There are only a couple of German speaking volunteer contributors who may or may not be available to assist you.

Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-28 by Nordlicht77