arendst/Tasmota #5993: Tuya not restoring white value on power off/power on

ameel, 2019-06-29:
Have Zemismart Downlight, based on tuya.

Flashed build 6.5.0.14(classic)  - just built, not modified.

Device is configured using AiLights template.

If I turn power switch off, and then back on, white value always resets to maximum value.

For example, I set color using "color 255,175,0,50". However, color on power on becomes: 255,175,0,255. Alternatively, I set color using "color FFAF0032", and color on power on becomes: FFAF00FF.

### BUG DESCRIPTION
White value resetting to maximum after power off/power on

STATUS 0 OUTPUT HERE:
10:16:08 CMD: status 0
10:16:08 SRC: WebConsole from 192.168.0.2
10:16:08 RSL: Received Topic /status, Data Size 1, Data 0
10:16:08 RSL: Group 0, Index 1, Command STATUS, Data 0
10:16:08 RSL: stat/sonoff/STATUS = {"Status":{"Module":27,"FriendlyName":["Dining 1"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:16:08 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"External System","Uptime":"0T00:00:49","StartupUTC":"2019-06-29T09:15:19","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":58,"SaveAddress":"F7000"}}
10:16:08 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.14(classic)","BuildDateTime":"2019-06-03T15:19:06","Boot":4,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
10:16:08 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["The Forbidden Zone",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00028009","280500000100060000000000000000000000","00000000"]}}
10:16:08 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":525,"Free":476,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","6010A390","049403A8","210217C4","010000C0"]}}
10:16:08 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0203","IPAddress":"192.168.0.108","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"CC:50:E3:4F:80:CB","Webserver":2,"WifiConfig":4}}
10:16:08 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4F80CB","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:16:08 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Jun 29 09:16:08 2019","Local":"Sat Jun 29 10:16:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
10:16:08 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-06-29T10:16:08"}}
10:16:08 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-06-29T10:16:08","Uptime":"0T00:00:49","SleepMode":"Dynamic","Sleep":0,"LoadAvg":133,"POWER":"ON","Dimmer":100,"Color":"255,175,10,255","HSBColor":"40,96,100","Channel":[100,69,4,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"The Forbidden Zone","BSSId":"28:80:88:D8:B5:97","Channel":13,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:05"}}}
10:16:10 UDP: Packet (101)
10:16:12 UPP: Hue 3 response packets sent to 192.168.0.3:34842

```

CONSOLE OUTPUT HERE:
BEFORE POWER OFF
10:14:23 CMD: weblog 4
10:14:23 RSL: stat/sonoff/RESULT = {"WebLog":4}
10:14:24 CFG: Saved to flash at F8, Count 57, Bytes 3584
10:14:30 UDP: Packet (101)
10:14:31 WIF: Checking connection...
10:14:31 WIF: Connected
10:14:32 UPP: Hue 3 response packets sent to 192.168.0.3:34842
10:14:40 UDP: Packet (101)
10:14:42 UPP: Hue 3 response packets sent to 192.168.0.3:34842
10:14:42 CMD: color 255,175,10,50
10:14:42 SRC: WebConsole from 192.168.0.2
10:14:42 RSL: Received Topic /color, Data Size 13, Data 255,175,10,50
10:14:42 RSL: Group 0, Index 1, Command COLOR, Data 255,175,10,50
10:14:42 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,175,10,50","HSBColor":"40,96,100","Channel":[100,69,4,20]}

AFTER POWER OFF AND POWER ON:
00:00:00 CFG: Loaded from flash at F8, Count 57
00:00:00 SRC: Restart
00:00:00 Project sonoff Dining 1 Version 6.5.0.14(classic)-2_5_2
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 The Forbidden Zone in mode 11N as sonoff-0203...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-0203 with IP address 192.168.0.108
00:00:05 UPP: Multicast (re)joined
00:00:05 APP: (UTC) Sat Jun 29 09:15:24 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
10:15:29 APP: Boot Count 9
10:15:29 CFG: Saved to flash at F7, Count 58, Bytes 3584
10:15:30 UDP: Packet (101)
10:15:32 UPP: Hue 3 response packets sent to 192.168.0.3:34842
10:15:36 UDP: Packet (174)
10:15:37 CMD: color
10:15:37 SRC: WebConsole from 192.168.0.2
10:15:37 RSL: Received Topic /color, Data Size 0, Data 
10:15:37 RSL: Group 0, Index 1, Command COLOR, Data 
10:15:37 RSL: stat/sonoff/RESULT = {"Color":"255,175,10,255"}

```

### TO REPRODUCE
1. set color in console using command: color 255,175,10,50
2. Turn power off
3. Turn power on

### EXPECTED BEHAVIOUR
Expected: Color remains the same, i.e. 255,175,10,50
Actual: Color changes to:  255,175,10,255

s-hadinger, 2019-07-02:
Good catch. Let me look at it.

s-hadinger, 2019-07-02:
This was tricky, it should be fixed with PR #6014.

closed 2019-07-02 by arendst

ameel, 2019-07-03:
Thanks for quickly looking into this and fixing it. I'll give the update a try and report any issues.

Cheers

PR #6014 (2019-07-02): Fix Not restoring white value on power off/power on (#5993)
## Description:

**Related issue (if applicable):** fixes #5993 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to add to _changelog.ino_ file without updating version)
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works.
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md# [... cut]