arendst/Tasmota #5996: Get Power from USB-Port consumption of Gosund SP112

Ocean-85, 2019-06-29:
Hello is it possible to use the powermeter of the Gosund Sp112 for the Usb ports only? When using this smartplug with tasmota you can only see the consumption of the plug itsself but I would try to have only the USB ones so that I can check the charging of a Smartphone and use the lowest consumption as trigger for turning off the USB relays.

ascillato2, 2019-06-29:
Sorry, the hardware was not designed for that. It is not supported.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-29 by ascillato2

Sonsi79, 2019-07-13:
> Hello is it possible to use the powermeter of the Gosund Sp112 for the Usb ports only? When using this smartplug with tasmota you can only see the consumption of the plug itsself but I would try to [...]

Which type of device did you choose? I also have the SP-112 and can't get it work so switch and measure at same time.
I tried 45 BlitzWolf but it doesn't work for me

ibeerens, 2019-09-06:
@Sonsi79 How did you flash the Gosund SP112 with Tasmoto?

xwox, 2019-10-06:
Hi,
I tried Sp1 (55) and it started sending message like "APP: Button1 multi-press" maybe 10 times per second.
If I set "ButtonDebounce 1000" than the switch starts to open and close like a "ghost" once per second.
To me it looks like that SP1 is also not the correct template.
Did you figure out which template to use?