arendst/Tasmota #6000: [Feature] Pressure Transducer Sensor support

badabing2005, 2019-06-30:
[image]

**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
No, just lack of sensor support

**Describe the solution you'd like**
Would prefer to use Tasmota over ESPHome

**Describe alternatives you've considered**
ESPHome

**Additional context**
I couldn't find any Pressure Transducer Sensor support in Tasmota, is there one that I missed?
If not, and it makes sense for the masses, this ticket can act as the feature request.

Example of such Sensor:
#### Input 5V Output 0.5-4.5V / 0-5V Pressure Transmitter for Water Gas Oil(0-80PSI)
https://www.amazon.ca/gp/product/B07KJHJNFQ

Alternative suggestion is to use ADC sensor in ESPHome
https://community.home-assistant.io/t/how-to-use-pressure-transducer-sensor/115880

Project planned for (but using D1 Mini)
https://www.instructables.com/id/Wireless-Pool-Filter-Pressure-Sensor/

Thanks
BB

badabing2005, 2019-06-30:
Perhaps the support for [ADC](https://github.com/arendst/Sonoff-Tasmota/issues/137) in Tasmota would work for this, I just have to figure out how to use it.
Would that work?

Thanks
BB

Jason2866, 2019-06-30:
Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-30 by ascillato2

badabing2005, 2019-06-30:
@Jason2866 
If that is the case why is there Feature Request option?
Where do those go?

Thanks

BB

ascillato, 2019-06-30:
Hi,

For explanations on how to connect electrically the the ADC pin for your sensor, please, address this to the Tasmota Support Chat.

Tasmota has the ADC pin enabled by default in latest version (dev branch) so, you only have to attach the sensor to the esp8266 and give the proper electrical connection. No extra config needed.

badabing2005, 2019-07-01:
Thank you @ascillato 

That's exactly what I needed, support for ADC, flashed the latest dev and set A0 to Analog and can see it.
Perfect, I just have to setup a voltage divider to handle up to 4.5v.

Thanks a bunch
BB

VR1985, 2020-07-10:
Hi @badabing2005 ,

i just have had the same issue with exact the same sensor... 
just setup like you wrote (Green line to A0) and now i can see "data" incoming
[image]
BUT - this signal is very unstable altough the sensor is right next to me in a cup of water - not moving or anything.
How did you manage to use this signal in a good way like "bar" / "voltage" or similar.
I'm trying to inplement it into my loxone ... thank you for any feedback

badabing2005, 2020-08-10:
@VR1985 
Same here, very unstable, the reading is all over the place.
I'm not sure if this is a deficiency in the sensor or Tasmota.

VR1985, 2020-08-10:
Well in the meantime i crafted a pressure test station (very rudimental) and measured the output voltage from zero to two bar..
Offset ia about 0.46V at zero pressure. 
Then I searched for an adequate formula to convert the output voltage into pressure and then to my whised unit.. Meter water height (mH2O). 
So it's a bit about to calibrate the sensor...

iragm, 2020-10-16:
This is pretty high up in the Google results, so I'll leave some info here.

These sensors are for pressure, just putting them in a glass of water won't do anything.  You need to connect them to system that has pressure.  They fit standard 1/4 NPT fittings, which makes them pretty handy.  They work on both water and gas (I use them to measure CO2 pressure).

They seem to be pretty reliable and don't need calibration.  I have had one fail due to getting wet on the outside and then rusting, but continuous exposure to water where it's supposed to be is not a problem.  I have 2 that are 4 years old now operating in 90% humidity with no issues.

They are meant to work at 5 volts, so you need a simple voltage divider to get them down to the 1 volt needed by the ESP's analog pin.

I process the output back to PSI using the following formula:
```
        var raw = A0; // 0-1024 reading from A0
        var pressure = raw/1025*175;
```
I do this processing in Node-Red but you could probably use a formula in Tasmota for it.

badabing2005, 2020-10-20:
Yeah, I'm aware of all the details, I had it connected to my pool pump and not sitting in a glass of water
The Pressure Transducer Sensor takes 5V and depending on the pressure (0 - 80psi)
It outputs 0.5v - 4.5v
I'm used D1 Mini for this GPIO A0 which takes maximum of 3.3v
At 3.3v the reading in Tasmota was indeed 1024
ADC pin on ESP takes maximum of 1v, which means that there is a voltage divider built into D1 Mini to drop 3.3v to 1v
I needed to add additional resistance on the A0 pin to drop max 4.5V from the transducer to 3.3v of A0
With my calculations (taking into account the internal voltage divider) I had to add 130K Ohm resister

The formula I used is slightly different as when there is no pressure, the output from the transducer is 0.5 v, which would give a reading of 113.77

All that to say, while the pump pressure was steady, the reading of A0 was all over the place.
At the time I used Tasmota 6.5.0.16 from Dev builds as support for it was only available in Dev builds.

Perhaps I had a defective transducer.
I'll try again next spring with the latest Tasmota, I'll also try with ESPHome to determine if there is an issue with the transducer.

Cheers

iragm, 2020-10-20:
I have 3 of these deployed, one controlling a CO2 system and the other 2
connected to a watering system where they tell me if the system is running
or not.  All 3 are extremely consistent.

For all of them, I am using an LM2596 to step 12v down to 5v to power an
ESP12-E.  The pressure transducer is powered from the 5v with the yellow
wire connected to A0 and a voltage divider.  The voltage divider reduces
the voltage from 5v to 1v. I don't remember the exact resistors I used, but
if anyone needs that, give me a mention and I can document this stuff more
completely.

I am extremely happy with the pressure transducers, the readings have been
very consistent and accurate (double checked against an analog gauge) as
far as I can tell.

badabing2005, 2020-10-20:
It would help the community to share the details, what puzzles me with your setup is that you're feeding A0 maximum of 1v
I don't know if it is any different on ESP12-E (I'd have to check) but on d1-mini / wemos A0 voltage of 3.3v reflects 1024 and not a voltage of 1V
1V is actually on the ADC pin of the ESP8266, but we don't connect the transducer to the ESP chip we connect it to A0.

Is my understanding wrong?

iragm, 2020-10-21:
I believe that the ESP's A0 has a max voltage of 1v.  The board I have definitely maxes out at 1 volt.  I found [a thing ](https://www.reddit.com/r/esp8266/comments/97ugbd/adc_max_voltage_is_different_between_versions/)that says some boards might have a built in voltage divider, so that is a possibility.

What I can tell you for sure, is that the following works for me.

I have a couple different voltage dividers on different ESPs, but they all work about the same.
```
0-5v yellow -----\
             R1 (20k)
                 |
                 +----- A0 ESP
                 |
             R2 (5k1)
GND black ------/
```

[Here is a highly technical diagram with labels](https://i.imgur.com/RfrfnHY.jpg). In case it helps, an [unlabeled version](https://i.imgur.com/fokFNvE.jpg) (pardon the bad soldering, I swear I have gotten better since I built this).

[Screenshot of Tasmota setup](https://i.imgur.com/fY5lTqP.png)

This will give you some number between 0 and 1024 which needs to be converted to PSI, which I do in Node Red: `PSI = raw A0 reading / 1025 * 175`

I am pretty sure that these work accurately, I compared them against an old analog gauge I have kicking around.  [Here is a chart of my CO2 pressure.](https://i.imgur.com/qXMJCGE.png)  Green is PSI, yellow is the solenoid that lets CO2 flow.

badabing2005, 2020-10-22:
I think we're both right
D1 mini indeed has a built in voltage divider to take 3.3v and convert it to 1v
https://github.com/arendst/Tasmota/issues/1523#issuecomment-355812473

I'll give it another shot for sure next spring.
I think I did everything correctly and suspect a defective transducer.
Time would tell.