arendst/Tasmota #6003: TSL2561 causing Blocked Loop, possible overflow?

maxxie85, 2019-07-01:
I attached a TSL2561 to a Wemos_D1 mini using i2c. I flashed the latest stable release of Tasmota 6.5.0.

When direct sunlight is hitting the sensor the Wemos crashes. Given in the information report that the last reboot is caused by a Blocked loop. When the Wemos restarts the i2c config is lost and I have te re-enable it again in the configuration. 

My assumption for the crash is that the lx value of the sensor is exceeding 32768 (or possible bigger) and that an overflow is generated. 

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Wemos D1 mini
- [X] Tasmota binary firmware version number used : self-compiled 6.5.0 including TSL2561 support
- [X] Development IDE - Compiler / Upload tools used : Platform IO
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
"Status":{"Module":18,"FriendlyName":["Wemos Woonkamer"],"Topic":"wemos_d1_woonkamer","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Attach a TSL2561 sensor and make sure it shows in the webpage 
2. Make sure the TSL2561 receives direct sunlight

### EXPECTED BEHAVIOUR
A maxxed out sensor reading

### SCREENSHOTS
https://vinch.nl/bin/tasmota.jpg

joba-1, 2019-07-01:
older versions of my tsl2561 lib had this issue. Tasmota should use a newer, fixed version. Can you please doublecheck the version used in your compile? Like verbose build output...

maxxie85, 2019-07-01:
When I download the latest source 6.5.0 from git. I can see that in the lib folder Joba_Tsl2561-2.0.7 is listed.

joba-1, 2019-07-01:
hmm, too old. 
Fix was in 2.0.8 and current (recommended) is 2.0.10

arendst, 2019-07-01:
You'll need the dev branch as that has updated the libs a long time ago. See REAME.md for dev pre-compiled firmares

closed 2019-07-01 by ascillato2