arendst/Tasmota #6006: unresponsive button for touch switch

RomanNagornyi, 2019-07-01:
- [ ] Device used: Sonoff 1T 1CH switch
- [ ] Tasmota binary firmware version number used: Sonoff-Tasmota v6.5.0 pre-compiled (sonoff-RU.bin)

With local MQTT server enabled button is unresponsible from time to time. After disable MQTT everything works without a hitch. Does it relate to slow TCP connection? Can it be fixed?

Thanks in advance.

ascillato2, 2019-07-02:
Hi,

I couldn't reproduce your issue. Sorry. My devices are still responsive if I turn off my mqtt broker.

> After disable MQTT everything works without a hitch. Does it relate to a slow TCP connection which blocks the thread?

Not really. Tasmota doesn't have any blocking (or waiting) code inside. Your issue seems to be another thing.

Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-07-03:
@RomanNagornyi 

Hi,

Any news on this?

RomanNagornyi, 2019-07-04:
@ascillato 

hello mate

> I couldn't reproduce your issue. Sorry. My devices are still responsive if I turn off my MQTT broker.

When you turn off?

ascillato, 2019-07-04:
Yes, if I have my mqtt broker working or not, and in my Tasmota devices the mqtt is enabled, I don't have your issue. So, as I can't reproduce your issue is why I'm asking you for more information. Please, complete the troubleshooting template in order to have more debug information so as to properly help you.

RomanNagornyi, 2019-07-04:
I see. 
I will prepare it in the next few days.

ascillato2, 2019-07-04:
Ok, no problem.

closed 2019-07-04 by ascillato2