arendst/Tasmota #6011: MQTT disconnecting every 30-40 seconds on 6.5.0

pmknowles, 2019-07-02:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
After upgrade to 6.5.0 MQTT disconnects every 30-40 seconds. Downgrade to 6.4.0 stops issue.
`13:36:57 MQT: Attempting connection...
13:37:03 MQT: Connect failed to 192.168.0.183:1883, rc -2. Retry in 10 sec
13:37:14 MQT: Attempting connection...
13:37:14 MQT: Connected
13:37:14 MQT: tele/Sonoff-Pump/LWT = Online (retained)
13:37:14 MQT: cmnd/Sonoff-Pump/POWER = 
13:37:14 MQT: stat/Sonoff-Pump/RESULT = {"POWER":"ON"}
13:37:14 MQT: stat/Sonoff-Pump/POWER = ON
13:38:00 MQT: Attempting connection...
13:38:02 MQT: Connected
13:38:02 MQT: tele/Sonoff-Pump/LWT = Online (retained)
13:38:02 MQT: cmnd/Sonoff-Pump/POWER = 
13:38:02 MQT: stat/Sonoff-Pump/RESULT = {"POWER":"ON"}
13:38:02 MQT: stat/Sonoff-Pump/POWER = ON
13:38:18 MQT: Attempting connection...
13:38:18 MQT: Connected
13:38:18 MQT: tele/Sonoff-Pump/LWT = Online (retained)
13:38:18 MQT: cmnd/Sonoff-Pump/POWER = 
13:38:18 MQT: stat/Sonoff-Pump/RESULT = {"POWER":"ON"}
13:38:18 MQT: stat/Sonoff-Pump/POWER = ON
13:38:58 MQT: Attempting connection...
13:39:02 MQT: Connected
13:39:02 MQT: tele/Sonoff-Pump/LWT = Online (retained)
13:39:02 MQT: cmnd/Sonoff-Pump/POWER = 
13:39:02 MQT: stat/Sonoff-Pump/RESULT = {"POWER":"ON"}
13:39:02 MQT: stat/Sonoff-Pump/POWER = ON
13:39:18 RSL: tele/Sonoff-Pump/STATE = {"Time":"2019-07-02T13:39:17","Uptime":"5T14:09:00","Vcc":3.156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"pkm-24i","BSSId":"0C:80:63:F3:A6:6E","Channel":5,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:39:18 RSL: tele/Sonoff-Pump/SENSOR = {"Time":"2019-07-02T13:39:18","DS18B20-1":{"Id":"0417C15DF4FF","Temperature":19.6},"DS18B20-2":{"Id":"0517C143A6FF","Temperature":19.5},"DS18B20-3":{"Id":"0517C16908FF","Temperature":24.0},"DS18B20-4":{"Id":"0517C172E7FF","Temperature":23.4},"DS18B20-5":{"Id":"0517C17422FF","Temperature":21.6},"TempUnit":"C"}
13:39:18 MQT: Attempting connection...
13:39:20 MQT: Connected`
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ Y] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [Y ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [Y] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [Y] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [Y] Device used (i.e. Sonoff Basic) : TH10 (plus others)
- [Y ] Tasmota binary firmware version number used : 6.5.0 / (pre-compiled)
- [Y] Development IDE - Compiler / Upload tools used : ESPFlasher ____
- [Y] Provide the output of command ``status 0`` :
`3:53:19 MQT: stat/Sonoff-Pump/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff-Pump"],"Topic":"Sonoff-Pump","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:53:19 MQT: stat/Sonoff-Pump/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"5T14:23:02","StartupUTC":"2019-06-26T22:30:17","Sleep":50,"CfgHolder":4617,"BootCount":160,"SaveCount":12581,"SaveAddress":"F5000"}}
13:53:20 MQT: stat/Sonoff-Pump/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:53:20 MQT: stat/Sonoff-Pump/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pkm-24i",""],"TelePeriod":120,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
13:53:20 MQT: stat/Sonoff-Pump/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
13:53:20 MQT: stat/Sonoff-Pump/STATUS5 = {"StatusNET":{"Hostname":"Pump","IPAddress":"192.168.0.120","Gateway":"192.168.0.1","Subnetmask":"255.255.0.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:0D:D0:F2","Webserver":2,"WifiConfig":2}}
13:53:20 MQT: stat/Sonoff-Pump/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.183","MqttPort":1883,"MqttClientMask":"Pump","MqttClient":"Pump","MqttUser":"openHAB","MqttCount":1680,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:53:20 MQT: stat/Sonoff-Pump/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 02 12:53:20 2019","Local":"Tue Jul 02 13:53:20 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"04:51","Sunset":"20:57"}}
13:53:20 MQT: stat/Sonoff-Pump/STATUS10 = {"StatusSNS":{"Time":"2019-07-02T13:53:20","DS18B20-1":{"Id":"0417C15DF4FF","Temperature":19.6},"DS18B20-2":{"Id":"0517C143A6FF","Temperature":19.5},"DS18B20-3":{"Id":"0517C16908FF","Temperature":24.1},"DS18B20-4":{"Id":"0517C172E7FF","Temperature":23.4},"DS18B20-5":{"Id":"0517C17422FF","Temperature":21.6},"TempUnit":"C"}}
[... 35 lines cut ...]
13:39:18 RSL: tele/Sonoff-Pump/SENSOR = {"Time":"2019-07-02T13:39:18","DS18B20-1":{"Id":"0417C15DF4FF","Temperature":19.6},"DS18B20-2":{"Id":"0517C143A6FF","Temperature":19.5},"DS18B20-3":{"Id":"0517C16908FF","Temperature":24.0},"DS18B20-4":{"Id":"0517C172E7FF","Temperature":23.4},"DS18B20-5":{"Id":"0517C17422FF","Temperature":21.6},"TempUnit":"C"}
13:39:18 MQT: Attempting connection...
13:39:20 MQT: Connected`

6.4.0 output
00:00:00 Project sonoff Sonoff-Pump (Topic Sonoff-Pump, Fallback Pump, GroupTopic sonoffs) Version 6.4.0(sensors)-2_4_2
00:00:00 WIF: Connecting to AP1 pkm-24i in mode 11N as Pump...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on Pump.local with IP address 192.168.0.120
13:56:43 MQT: Attempting connection...
13:56:43 MQT: Connected
13:56:43 MQT: tele/Sonoff-Pump/LWT = Online (retained)
13:56:43 MQT: cmnd/Sonoff-Pump/POWER = 
13:56:43 MQT: tele/Sonoff-Pump/INFO1 = {"Module":"Sonoff TH","Version":"6.4.0(sensors)","FallbackTopic":"Pump","GroupTopic":"sonoffs"}
13:56:43 MQT: tele/Sonoff-Pump/INFO2 = {"WebServerMode":"Admin","Hostname":"Pump","IPAddress":"192.168.0.120"}
13:56:43 MQT: tele/Sonoff-Pump/INFO3 = {"RestartReason":"Software/System restart"}
13:56:43 MQT: stat/Sonoff-Pump/RESULT = {"POWER":"ON"}
13:56:43 MQT: stat/Sonoff-Pump/POWER = ON
13:56:43 MQT: stat/Pump/RESULT = {"POWER":"ON"}
13:56:43 MQT: stat/Pump/POWER = ON
13:56:51 MQT: tele/Sonoff-Pump/STATE = {"Time":"2019-07-02T13:56:51","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"pkm-24i","BSSId":"0C:80:63:F3:A6:6E","Channel":5,"RSSI":66}}
13:58:51 MQT: tele/Sonoff-Pump/STATE = {"Time":"2019-07-02T13:58:51","Uptime":"0T00:02:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"pkm-24i","BSSId":"0
```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-07-02:
Hi, 

Seems that you are having a problem in your flash. May be some incorrect data is being left in the memory.

Try to update to latest Tasmota (6.5.0.16) and type in the Tasmota console: `reset 5`

If your issue continues, you will need to erase all flash with esptool.py as explained in the wiki. 

Please check the following links of the wiki for more information:

https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#frequent-mqtt-reconnects

pmknowles, 2019-07-02:
That seems to have fixed it.

Thank you

closed 2019-07-02 by pmknowles