arendst/Tasmota #6012: Compile latest dev with core 2.3.0 fails with VSC (Platformio)

Jason2866, 2019-07-02:
### BUG DESCRIPTION
Compile latest dev fails with core 2.3.0

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : none
- [x] Tasmota binary firmware version number used : 6.5.0.16 / (self-compiled ?)
- [x] Development IDE - Compiler / Upload tools used : VSC / ____
- [x] Provide the output of command ``status 0`` :
```
Not possible compile error

```

### TO REPRODUCE
USE VSC and compile dev with core 2.3.0 enabled

### EXPECTED BEHAVIOUR
Compile without error

### SCREENSHOTS
```
Linking .pioenvs/sonoff/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa@1.40802.0/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libm.a(lib_a-ef_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainderf'
/home/gitpod/.platformio/packages/toolchain-xtensa@1.40802.0/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libm.a(lib_a-ef_fmod.o): In function `__ieee754_fmodf':
ef_fmod.c:(.text+0x36): undefined reference to `__ieee754_remainderf'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff/firmware.elf] Error 1
===================================== [ERROR] Took 85.55 seconds =====================================

============================================= [SUMMARY] =============================================
Environment sonoff              [ERROR]
```

snakuzzo, 2019-07-02:
Just a little info.
Yesterday I tried to build firmware on macOS Mojave + Atom (platformio) ... same error.
Today I tried using Windows x64 + VS Code (platformio) and it worked

Jason2866, 2019-07-02:
Found the problem. Core 2.3.0 uses Xtensa build chain 1.408x 
In this build chain the file `c:\Users\account\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\lib\libm.a ` has not included the needed referenced function.
Replacing this file with file from Xtensa build chain 2.4x does solve the error.

arendst, 2019-07-02:
Agh. 

To solve this for arduino too we have to find the source for function fmodf as used in sonoff_float.ino and add it to this file.

Jason2866, 2019-07-02:
Maybe this `https://github.com/jcmvbkbc/newlib-xtensa/blob/xtensa/newlib/libm/math/ef_fmod.c`

arendst, 2019-07-02:
Won't work either as remainderf also is not in core 2.3.0 lib.

I have another solution and will release soon.

Jason2866, 2019-07-02:
Seems to work :-)

closed 2019-07-02 by Jason2866

snakuzzo, 2019-07-02:
It works!
Thanks ;)