arendst/Tasmota #6013: Sonoff RF with on 6.5.0 no wlan connect after reboot

chrisherzberg, 2019-07-02:
I flashed my sonoff rf with the latest firmware 6.5.0. After reboot I can connect to the internal web server and configure my home wlan.

First reboot ans everythink worked as aspected.
But when I plug my sonoff out an plug it in again it comes up with no wlan connectivity and also after presssing 4 times the switch no internal web server comes up. The LED flashes green or even red after I press 4 times the switch.

I can reproduce this trouble by new flashing the device.

Any ideas?

Thanks
Chris

meingraham, 2019-07-02:
Please try the latest stable development version (http://thehackbox.org/tasmota).

It may be that the device is bootlooping and iteratively performing a rest of setting back to firmware defaults. When you power up the device again after a successful configuration and connection, do you hear any clicks on your device?

In any case, I don't think this is a bug. Nevertheless, what is happening needs to be determined. To do this, please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

chrisherzberg, 2019-07-02:
Hi,
first off all, thank you for your quick response. I used the firmware 6.5.0 with core 2.3.0.
I will test first the 6.5.0 with core 2.4.2 and if it not running I will try the development version.

I don't think that the device is bootlooping because the relais is responding on pressing the button as it should.

Cheers

meingraham, 2019-07-02:
The Tasmota strategy for bootlooping is to iteratively reset settings back to "basics" until it stops bootlooping (see [SetOption36](../wiki/Commands#setoption36) for the five steps taken on bootloop resets). Since the device is "stable" (although not configured as desired) when you initially flash Tasmota, there is a Tasmota settings "state" where it stops bootlooping. Thus, when it reaches that state, you will be able to use the device via its physical button as expected. In other words, the bootloop resets eventually reach a point where the device no longer reboots and thus you are able to control it manually.

Development binaries:
[2.3.0](http://thehackbox.org/tasmota/020300) (tends to be the go to Core for stability)
[2.5.2](http://thehackbox.org/tasmota/020502) (the latest and seems to be pretty reliable)
[2.4.2](http://thehackbox.org/tasmota/) (Probably not worth trying if 2.5.2 doesn't work)

chrisherzberg, 2019-07-02:
can be closed. I'am sorry. It was not the firmware, it was a smal piece of solder did a shortcut on R21.

closed 2019-07-02 by ascillato2