arendst/Tasmota #6015: Sdm 120 Does not send the correct power value

vmfs1968, 2019-07-03:
Sdm120 
Does not send the correct power value to domoticz
It should send the value x1000 and it does not send

ascillato, 2019-07-03:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

vmfs1968, 2019-07-03:
Hello
thank you for answering
This is the sdm120 logo.

18:13:45 MQT: domoticz/in = {"idx":45,"nvalue":0,"svalue":"373;5361.967","Battery":200,"RSSI":4}
18:13:45 MQT: tele/Eletro/SENSOR = {"Time":"2019-07-03T18:13:45","ENERGY":{"Total":5361.967,"ActivePower":373,"ApparentPower":374,"ReactivePower":-25,"Frequency":50,"Factor":1.00,"Voltage":248,"Current":1.639}}
18:18:45 MQT: tele/Eletro/STATE = {"Time":"2019-07-03T18:18:45","Uptime":"0T09:20:15","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"domoticz","BSSId":"CE:2D:E0:C2:85:31","Channel":1,"RSSI":46,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:18:45 MQT: domoticz/in = {"idx":45,"nvalue":0,"svalue":"397;5362.000","Battery":200,"RSSI":4}
18:18:45 MQT: tele/Eletro/SENSOR = {"Time":"2019-07-03T18:18:45","ENERGY":{"Total":5362.000,"ActivePower":397,"ApparentPower":398,"ReactivePower":-23,"Frequency":50,"Factor":1.00,"Voltage":249,"Current":1.476}}
---------------------------------------------------------------------------------------------

and this is the pzem-004t logo
this sends the data correctly to the domoticz

18:32:54 MQT: domoticz/in = {"idx":43,"nvalue":0,"svalue":"195;49488.0","Battery":200,"RSSI":8}
18:32:54 MQT: domoticz/in = {"idx":44,"nvalue":0,"svalue":"248.4","Battery":200,"RSSI":9}
18:32:54 MQT: tele/Quadro-Luz-Casa/SENSOR = {"Time":"2019-07-03T18:32:54","ENERGY":{"TotalStartTime":"2019-06-25T08:06:15","Total":49.488,"Yesterday":4.461,"Today":2.931,"Period":1,"Power":195,"ApparentPower":273,"ReactivePower":191,"Factor":0.71,"Voltage":248.4,"Current":1.100}}

ascillato, 2019-07-03:
Please, could you be so kind on completing the troubleshooting template in order to have more information so as to properly help you? There is a lot of information missing like status 0, Tasmota version, etc etc.

vmfs1968, 2019-07-03:
00:00:00 CFG: Loaded from flash at 3FA, Count 18
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.5.0.16(sonoff)-2_5_2
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 domoticz in mode 11N as Quadro...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on Quadro with IP address 192.168.1.121
00:00:04 HTP: Main Menu
19:50:51 APP: (UTC) Wed Jul 03 18:50:51 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
19:50:52 MQT: Attempting connection...
19:50:52 MQT: Connected
19:50:52 MQT: tele/Eletro/LWT = Online (retained)
19:50:52 MQT: cmnd/Eletro/POWER = 
19:50:52 MQT: Subscribe to cmnd/Eletro/#
19:50:52 MQT: Subscribe to cmnd/sonoffs/#
19:50:52 MQT: Subscribe to cmnd/DVES_D543E6_fb/#
19:50:52 MQT: tele/Eletro/INFO1 = {"Module":"Generic","Version":"6.5.0.16(sonoff)","FallbackTopic":"cmnd/DVES_D543E6_fb/","GroupTopic":"sonoffs"}
19:50:52 MQT: tele/Eletro/INFO2 = {"WebServerMode":"Admin","Hostname":"Quadro","IPAddress":"192.168.1.121"}
19:50:52 MQT: tele/Eletro/INFO3 = {"RestartReason":"Software/System restart"}
19:50:52 HTP: Console
19:50:56 APP: Boot Count 7
19:50:56 CFG: Saved to flash at 3F9, Count 19, Bytes 3584
19:51:00 MQT: tele/Eletro/STATE = {"Time":"2019-07-03T19:51:00","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"domoticz","BSSId":"CE:2D:E0:C2:85:31","Channel":1,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:51:00 MQT: domoticz/in = {"idx":45,"nvalue":0,"svalue":"402;5362.601","Battery":200,"RSSI":5}
19:51:00 MQT: tele/Eletro/SENSOR = {"Time":"2019-07-03T19:51:00","ENERGY":{"Total":5362.601,"ActivePower":402,"ApparentPower":404,"ReactivePower":-32,"Frequency":50,"Factor":1.00,"Voltage":247,"Current":1.628}}
19:51:10 WIF: Checking connection...
19:51:10 WIF: Connected

vmfs1968, 2019-07-03:
Program Version | 6.5.0.16(sonoff)
-- | --
2019-07-01T22:48:22
2_5_2/2.2.2-dev(c0eb301)
0T00:02:27
19 at 0x3F9000
7
Software/System restart
 
domoticz (52%)
Quadro
192.168.1.121
192.168.1.254
255.255.255.0
192.168.1.254
18:FE:34:D5:43:E6
 
192.168.1.99
1883
DVES_USER
DVES_D543E6
Eletro
sonoffs
cmnd/Eletro/
cmnd/DVES_D543E6_fb/
 
None
Disabled
 
13976550
0x1740EF
8192kB
4096kB
563kB
2508kB
27kB

ascillato2, 2019-07-03:
@vmfs1968 

Thanks for reporting. :+1: 

@arendst 

This issue can be solved only adding a *1000 to sdm120_active_power at line:
https://github.com/arendst/Sonoff-Tasmota/blob/db05d920cfc3bc492ebcbf033d00124041e88874/sonoff/xsns_23_sdm120.ino#L352

but the question is why the SDM120 is not managed as an energy sensor as the PZEM. The PZEM driver puts all the data into variables and then calls `EnergyUpdateToday()` of _xdrv_03_energy.ino_

https://github.com/arendst/Sonoff-Tasmota/blob/db05d920cfc3bc492ebcbf033d00124041e88874/sonoff/xnrg_03_pzem004t.ino#L189

The `EnergyUpdateToday()` makes *1000 for the power value before sending to domoticz. This is not used in SDM for something special?

arendst, 2019-07-03:
@ascillato  correct. SDM support was written by Genarro.

Adding your fix to both sdm drivers will solve this issue.

vmfs1968, 2019-07-03:
gives an error while compiling

[image]

arendst, 2019-07-03:
Pls see the solution in xdrv_energy. There is more to it. You cannot multiply a string.

vmfs1968, 2019-07-03:
I'm not getting
can give one more help
What do I have to change to work?

ascillato, 2019-07-03:
Later I will make the Pull Request with this fix added. Please, give us time. Thanks

vmfs1968, 2019-07-03:
OK
  thank you very much
I really liked to have my sdm working well.
wait

ascillato, 2019-07-03:
Pull Request #6017 for fixing this.

ascillato, 2019-07-03:
If you want, you can test the changes now, just modifying your code as in https://github.com/arendst/Sonoff-Tasmota/pull/6017/files

PR #6017 (2019-07-04): Fix SDM120 and SDM630 Energy Values to Domoticz
## Description:

Fix SDM120 and SDM630 Energy Values to Domoticz

**Related issue (if applicable):** fixes #6015

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to add to _changelog.ino_ file without updating version)
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works.
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmo [... cut]
closed 2019-07-04 by arendst