arendst/Tasmota #6016: Relay goes on by itself

seppelicous, 2019-07-03:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I use a Sonoff 4CH Pro R2  with TASMOTA 6.5.0 Firmware. The Sonoff is configured without any timer, and there is an automatic cutoff after 30min on, for safety reasons. I use the 4 relays to controle 4 sprinkler valves.
The relays are triggered by OpenSprinkler and http commands. I do not use MQTT and not any of the buttons on the sonoff.

But sometimes a relay opens by itself, fortunately it switches off after 30min (Tasmota setting). 
Mostly it is Relay 4, sometimes Relay. 
I checked the log and it seems, that a Button was pushed, but nobody was at home.
This is the log, filtered by "Knopf" (German for Button) of the last 4 weeks, while nobody was at home.
[image]
Every Button actions is followed by these actions:
[image]

What happens here? I guess this is a hardware issue, but I wanted to double check before I contact the seller.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4) 
--> nobody replied
- [ x] Device used (i.e. Sonoff Basic) : 4ch pro r2
- [ ] Tasmota binary firmware version number used : 6.5.0 / (pre-compiled)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

22:30:45 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:30:45 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Power on","Uptime":"39T13:39:49","StartupUTC":"2019-05-25T06:50:56","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":1056,"SaveAddress":"F4000"}}
22:30:45 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019.03.19 12:30:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:30:45 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":3,"LogHost":"192.168.178.5","LogPort":514,"SSId":["Accio Internet!",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
22:30:45 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":508,"Free":492,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","0FDAE394","000783A0","23B617CE","00003BC0"]}}
22:30:45 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5078","IPAddress":"192.168.178.127","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"C4:4F:33:8C:93:D6","Webserver":2,"WifiConfig":4}}
22:30:45 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.63","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8C93D6","MqttUser":"DVES_USER","MqttCount":61,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:30:45 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Jul 03 20:30:45 2019","Local":"Wed Jul 03 22:30:45 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:52","Sunset":"21:56"}}
22:30:45 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-07-03T22:30:45"}}
22:30:45 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-07-03T22:30:45","Uptime":"39T13:39:49","Vcc":3.134,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Accio Internet!","BSSId":"1A:E8:29:61:64:48","Channel":6,"RSSI":70,"LinkCount":3,"Downtime":"0T00:03:31"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-07-03:
Have you attempted any of the recommendations in the [FAQ](../wiki/FAQ#D4)?

seppelicous, 2019-07-03:
> Have you attempted any of the recommendations in the [FAQ](../wiki/FAQ#D4)?

I did and I already followed the steps regarding the retaining message issue but that is about 6 weeks ago, so it seemed not to be the root of the issue.

meingraham, 2019-07-03:
So, could there be a random signal on the button input. Do you have a low pass filter on the button input?

seppelicous, 2019-07-03:
I did not any hardware upgrades to the Sonoff. When there is a random signal on the button, then that must be a hardware failure of the sonoff. Do we speak about the same buttons? 
When log says "button4", it means that the button R4 was pressed?

meingraham, 2019-07-03:
OK, I see.

I still believe that you are getting a random electronic glitch on that button... which indicates a minor defect with the device hardware itself.

I have a Sonoff 4Ch Pro (R1). No issues with ghost switching. Other users are not reporting ghost issues with the device. See if you can use [ButtonDebounce](../wiki/Commands#ButtonDebounce) to eliminate these ghost button presses.

Mike

ascillato2, 2019-07-04:
@seppelicous

Hi,

Could you solve your issue as Mike has explained?

seppelicous, 2019-07-04:
> OK, I see.
>
> I still believe that you are getting a random electronic glitch on that button... which indicates a minor defect with the device hardware itself.
>
> I have a Sonoff 4Ch Pro (R1). No issues [...]

Mike, thank you very much for the tip with the debounce setting. Set should prevent open the valves when a ghost is visiting my home 😉
Now I am running in another issue: I set ButtonDebounce to 1000ms. But right after the Relay goes on and I remove my finger it goes off again.
Is there any other function on the button, when I press them >1000ms?

meingraham, 2019-07-04:
Try lowering the debounce setting a bit. I'm not sure you need it set that high.

Also, see if working with SetOption13 and SetOption32 affects the behavior.

Alveran81, 2019-07-05:
Hello, i have three Sonoff 4ch pro. All Work fine but after three month i get this Error. So i replaced it with a new one but again after some month same Error.

ascillato, 2019-07-05:
@seppelicous

Hi,

Could you solve your issue as Mike has explained?

Alveran81, 2019-07-06:
I did not test i now. I already put these in the cupboard half a year ago and switched to wago. If I can find the next weeks time I can test it with pleasure times.

seppelicous, 2019-07-06:
I made a video about the behavior when I press the button longer then debounce time. See the link attached. 

The Relay toggles from off to on or on to off when its pressed a little bit longer. The optional you mentioned allow me to set HOLD ups to 1s (?). Is it possible to completeöy disable the button hold Option. 

I think then my initial issue is solved 😊 

https://share.icloud.com/photos/0mK728BeO-JRppUViojUdj_dg#Zuhause

ascillato, 2019-07-06:
For disabling multipress, go to the Tasmota console and type:

Setoption1 1
Setoption13 1

ascillato, 2019-07-06:
Also, as Mike stated, debounce at 1000 is too high. You should use 250 and test it.

This issue has the label duplicated because this has been addressed before and the best result for other users, as explained in other issues, was disabling multipress and setting debounce to 250.

seppelicous, 2019-07-06:
Thank you very much for your Support!

closed 2019-07-06 by seppelicous

ascillato, 2019-07-06:
We want your device to work properly, so, if you have any other issue, please don't hesitate on asking here or in the Tasmota support.

Please, try those settings for few days and tell us if these solved your issue. Thanks