arendst/Tasmota #6018: H801 - Unable to access main menu after flashing

kingchad5, 2019-07-03:
I am trying to set up my H801.  I have tried a few different sonoff.bin versions and I can only get Version 6.4.1(minimal)-2_4_2 (pre-compiled) to "successfully" flash. I used NodeMCU-PyFlasher-4.0 to do the flashing.  All other versions I tried never broadcast a wifi network.  When I do connect to the 6.4.1 minimal version the main menu will not load.  The only thing that loads is the console page 192.168.4.1/cs.  In the console, everything looks ok

00:00:00 Project sonoff Sonoff Version 6.4.1(minimal)-2_4_2
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-5782 with IP address 192.168.4.1

I confirmed my H801 has the 1mb of memory not 512k.  I have tried to upgrade to the full version via serial by flashing again without wiping the flash, but that produces the same results as a normal flash when I wipe the flash.  From what I understand with the minimal version the only option I have is to upgrade it.  I am unable to save any configurations via the console.  Any guidance would be helpful.
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (i.e. Sonoff Basic) : _____
- [X ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [X ] Development IDE - Compiler / Upload tools used : ____ / ____
- [X ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

00:01:42 CMD: status 0
00:01:42 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:01:42 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:42","StartupUTC":"","Sleep":50,"BootCount":2,"SaveCount":3,"SaveAddress":"F9000"}}
00:01:42 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(minimal)","BuildDateTime":"2018-12-24T14:42:33","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:01:42 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
00:01:42 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":365,"Free":636,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","00082580","000003A2","21021702","000000C0"]}}
00:01:42 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5782","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:4B:D6:96","Webserver":2,"WifiConfig":4}}
00:01:42 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4BD696","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:42 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:42 1970","Local":"Thu Jan 01 00:01:42 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00"}}
00:01:42 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:42"}}
00:01:42 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:42","Uptime":"0T00:01:42","Vcc":3.370,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

Logs when I try to go to the main menu.... the webpage times out
00:02:07 HTP: Redirected to captive portal
00:02:07 HTP: Redirected to captive portal
00:02:35 HTP: Console
00:02:37 HTP: Redirected to captive portal
00:02:37 HTP: Redirected to captive portal
00:02:47 HTP: Main Menu
00:02:49 HTP: Main Menu
00:02:57 HTP: Redirected to captive portal
00:02:57 HTP: Main Menu
00:03:07 HTP: Redirected to captive portal
00:03:07 HTP: Redirected to captive portal
00:03:17 HTP: Main Menu

```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-07-03:
Hi,

Version 6.4.1 is very old.

Please, flash latest Tasmota 6.5.0.16 from http://thehackbox.org/tasmota/020502/sonoff.bin

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-03 by ascillato2

kingchad5, 2019-07-03:
I have tried the latest version and a few in-between.  The only one I could get to work was this version.

ascillato, 2019-07-03:
Please, try erasing all flash and flashing again but with esptool.py as explained in the wiki. Esptool.py is the recommended for Tasmota