arendst/Tasmota #6021: Tasmota schedule timer reliability

robin-s-007, 2019-07-04:
Hello,

I have been a happy Tasmota user for the past 6 months.
And now am experiencing some strange issues with the timer scheduling.
I configured one daily timer task via the web interface, in the following way:
Position 1:  output 1 -> ON @ 8:00 and checked: Time, active, repeat and every weekday.
Position 2:  output 1 -> OFF @ 17:00 and checked: Time, active, repeat and every weekday.
 
I also configured the time zone and daylight saving via MQTT.

Most of the time the unit switches at the correct times, but at certain moments the unit misses at random either the ON and OFF switch moments.
If I log onto the console the timestamps are running on the correct time.

I could not find an similar issue report online.

Could it be that the unit does a reboot or time sync just around the switching times, so the unit misses the exact switch moments ?
Can the timer cope with missing the exact switching moment ?  
Can I configure the unit in a different way so that its more reliable ?
Or do I need to forgot about this feature, and create a remote schedule ?  Did not do this as this seemed more reliable.

Thanks in advance,

Best Regards,

Robin

meingraham, 2019-07-04:
@robin-s-007 

I suggest that you set up a syslog server so that you may trap all the log output from this device. This way, when the issue occurs you'll have the log of exactly what and when event occurred. THis will give you more insight was to whether it is a coincidental reboot timing or whether there is a deeper issue which we need to follow up with.

Regards.

Mike

ascillato, 2019-07-05:
Hi, 

I use Tasmota on several lights that turns on and off using the timers. Since I installed Tasmota on them from more than a year I never had a miss on turning on/off. So, may be your device is rebooting and missing the trigger but that is just a guess since the troubleshooting template was not completed. You can add a rule to check the time at boot and turn on/off if your device was powered after the time you set in your timer. There are some examples of that in the wiki at the rules cookbook. 

Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

arendst, 2019-07-05:
Some background info:
- The esp8266 software clock is synced every hour at xx:yy:zz where xx is every hour and yy = 01 minutes and zz = (Last nibble of MAC address *3) +3 seconds.
- Timer is checked every minute
- If the software clock drifts less than 58 seconds per sync moment the timer will still be handled correctly
- If the software clock drifts more than 58 seconds per sync moment you will probably miss a timer 
- Missing a timer is probably close to the next sync moment
- You can sync more often using command ``timezone 99`` or ``ntpserver3 <your ntpserver3 setting>`` every 30 minutes. A rule like ``rule1 on time#minute|30 do timezone 99 endon`` should fix larger drifts

Having more logging info might reveal if drift is the issue here.

robin-s-007, 2019-07-06:
Thanks for all your quick responses : )
@meingraham:  
Last night I installed a syslog server and set the plugs syslog info to level 2 (info)
Since this is the same debug level as the serial and console port, this seemed sufficient for now as I expect that the disconnects will be visible at this log level. 
If not please let me know.

@ascillato:
I will skip your workaround for now as I think its best to try to discover the root cause first.

@arendst:
First off all; Thanks for greating such great and comprehensive software .
Thanks for the insight in how the code works.
Based on your info; The timesync moment and the drift, I would say that as simple work around could be to shift my on/off times by 5 mins. This will eliminate the error caused by the (possible) drift and time sync moment.

I will keep tabs on the log output and report back if I found something.
This could take sometime as the errors do not happen daily.

Thanks for your feedback sofar.

meingraham, 2019-07-06:
@robin-s-007 yes, level 2 ought to be sufficient to trap what was going on at the time that the event was "missed".

Please post your findings as you begin to understand the root cause better.

It will be helpful to get a complete picture if you could post the output of a `Status 0` command.

Mike

ascillato2, 2019-07-06:
Putting this issue on hold until you get the logs.

closed 2019-07-06 by ascillato2

robin-s-007, 2019-07-09:
I have an update...
Yesterday and today the switching problem returned again.
Examining the logs I discovered that the unit had quite some connects and disconnects over the day.
They also happened around the switching times.

In both cases (yesterday afternoon, and this morning), the log showed that the unit disconnected briefly on or 5 min after the switching time and then was issues an retained MQTT message from Node-red.
This due to the manual control action done after the automatic switching did (seem) not work.
The retained status is a bit of a blessing and a curse as it has caused me a headache before...

I was not aware of the retained flag as I copied and pasted a piece of Node-red flow and renamed parts.
In the logs of this morning I could clearly see that at 8:00 the unit switched on, and disconnected from the network approx 5 min later, at which time the unit got a retained off message.
Yesterday evening at 17:00 the network disconnect happened approx 10 seconds after 17:00, so the logs revealed no timer based switch.

I think that I have captured the issue but will keep monitoring logging for the next 2 weeks, to try and capture what happens on the network disconnect at switching time.

arendst, 2019-07-09:
Thx for the update.

robin-s-007, 2019-08-11:
I have an other update:
After quite a long time of running the timers without problems., they suddenly started appearing.
In the meantime I changed the unit from DHCP to fixed IP, which I noticed for some units increased their uptime.
But since the last couple of days the unit is in a state that it reconnects just at the switching time (OFF moment).
See image:
[image]

The unit unfortunately does not apply the timer after it has restarted, although its within the 58 seconds of drift period that @arendst described, and should switch off.

I now raised the syslog level to 3 (debug) and hope that I will get some more info.

robin-s-007, 2019-08-13:
I now have a log with the debug level3:
[image]

It mentions "reset reseason: External system"
What does this mean ?
In the log I also see not mention of the timer interval being checked, just that the unit is reverting to its last known state.

meingraham, 2019-08-13:
@robin-s-007 Go ahead and post the new information.

arendst, 2019-08-13:
6.4.1 ....

Upgrade pls and report back when issue reappears.

robin-s-007, 2019-09-13:
Please reopen:
Issue still appears with version 6.6.0

robin-s-007, 2019-09-13:
Hi,

In the meantime I have upgraded to the latest version (6.6.0) and ran it for a couple of weeks.
Since 2 days the problem now has re-appeared.
As I this time was logging debug level, I think that there might be more usefull info:

[image]

As far as I can tell, the unit switches off when it should, but then the wifi reconnects (most likely because it detected a broken link when sending data to broker).
When the unit has reconnected to wifi, and it connects to the broker, it thinks that it needs to switch on again, but did not receive an retained cmnd. Just a retained old status (a sign that it was not able to send the new switch off state to the broker).

To me it almost looks like Tasmota is using the retain state to set its output state, but for this it should use the cmnd, and actually should update its MQTT state to reflect its proper output state.

What is your opinion ?

--

meingraham, 2019-09-13:
I'm not sure if we've been down this path yet... What is `PowerRetain` set to? `PowerOnState`?

Even though you now are on 6.6.0, I think we should try the latest which is a couple of months since 6.6.0 was released.

`Backlog OTAURL http://thehackbox.org/tasmota/json/sonoff.bin; Upgrade 1`

Mike

robin-s-007, 2019-09-13:
@meingraham:
PowerRetain is set to ON: on = enable MQTT power retain on status update 
PowerOnState is set to 3: 3 = switch relay(s) to their last saved state (default) 

So it should report every update of the output state on MQTT (they should be in sync, what they are not)
After a Power on the device should revert its latest state.

The PowerRetain function failed to update the MQTT Power state, due to a network reconnect.
As a result the device power state and the MQTT retained power state differed. While the unit did not restart not power cycled, the power state seems to have changed after receiving the wifi connected again, and received the retained /stat/POWER at which time the output of the device seems to have switched on.

Switching to another newest Tasmota build will be an sort of race against the clock, as when I update I will never see this in that specific newest version as the fault, for some reason seems to take a couple of weeks of nonstop runtime before this problem reproduces.
It is also an issue that has exhibited itself in several releases now.

To tackle this problem I think that diagnosing this results seems the most logical, otherwise their will never be an opportunity as the failed version is always outdated.

In my analysis it looks like Tasmota seems to use the /stat/POWER topic to sync its output state  when it reconnects. And for some reason after some time it disconnects just before it can sent its current state and seems to fall inbetween the reconnect cycle.

meingraham, 2019-09-14:
@robin-s-007 

I just think something else is causing your issues. Otherwise I think we'd be hearing from a lot more users. There were some know issues with 6.6.0... although not related to power state, I don't believe. Enough where we're really want folks to try a newer version. Installing 6.6.0.12 will eliminate 6.6.0 as a variable and put this troubleshooting effort on a known foundation.

I think you need `SetOption0 ON`, `PowerRetain OFF`, `PowerOnState 3`.

Also, for more regarding unexpected switching or state restoration, see the information in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#My-device-randomly-switches-on-and-off-Do-I-have-ghosts-in-my-house).

robin-s-007, 2019-09-22:
@meingraham 
A bit late response, but I have just updated my unit to 6.6.0.12 git 763601c 
I have also set PowerRetain to OFF as you suggested, the other 2 items already had the same settings.
I will look at the FAQ in more detail, but a quick look at it and the PowerRetain description have now raised some questions on this features.  
Imo the PowerRetain should create a persistant state flag of the last known device switch state, for other MQTT users to use it even when the device is offline.
A retained cmd/Power should set the devices switch state back to the retained value everytime the devices reconnects to the MQTT server. (Have run into this issue in the past).
So at this moment I am a bit confused...
I will report back when the unit has exhibited a fault again, usually it takes a couple of weeks.

robin-s-007, 2019-09-29:
Please reopen:
Issue still appears with version 6.6.0.12 git 763601c and @meingraham recommend setting.

robin-s-007, 2019-09-29:
The problem has reoccurred (a bit faster than expected), with  version 6.6.0.12 git 763601c and @meingraham recommend settings.
In the logs I see:
16:59:59 => unit sends last tele and state update  (Power state = ON)
17:00:01 => Timer is activated  ( should switch to Power OFF)
17:00:10 => Wifi connected (connection appearantly got lost at the timer switch) (Power state = ON)
17:00:11 => Wifi reconnect and connected 
17:00:19 => First telemetry info, with epoch time stamp    (Power state = ON)
17:02:19 => NTP has finally synced and Telemetry info is send with good timestamp  (Power state = ON)
Power state remains ON, until its manually deactivated.
I have checked the FAQ and the retain settings for this unit.
As can be seen in the screenshot, now only the Tele messages are retained.
The unit is a Blitzwolf SHP-6 and has no external connections.

For some reason the device seems to be disconnected from wifi at the moment the timer should switch the relay.  
Due to the delay of the NTP sync, the time out of sync mechanism described some posts earlier does not work, and the time scheduled switch point is passed.
I do however not understand why the state did not change at the first timer moment.
Looking at the logs I see only wifi dis/reconnects and no reboots.
I also do not know why the NTP sync took so long, as this should be almost immediate. 

Any other suggestions, ideas or tips ?
[image]
[image]
[image]