arendst/Tasmota #6022: Any news about the Sonoff mini

freaktech, 2019-07-05:
is the new mini compatible with tasmota?
I ordered it 4 weeks ago and it was shipped 2 weeks ago
should be here every day and i need a pulse function

so anyone already tested it and provide a info how to flash?

https://www.banggood.com/de/SONOFF-Mini-Two-Way-Smart-Switch-For-Smart-Home-Can-Work-With-An-External-Switch-p-1484758.html?gmcCountry=AT&currency=EUR&cur_warehouse=CN&createTmp=1&ID=49554&utm_source=googleshopping&utm_medium=cpc_union&utm_content=2zou&utm_campaign=ssc-at-all-0701&ad_id=359090513510&gclid=CjwKCAjwx_boBRA9EiwA4kIELmrr7yDSKTk-fLEe5Oi9TisqSvpBDqbkE9v75IkQp-p6khvkgfhnfRoCOfsQAvD_BwE

Jason2866, 2019-07-05:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

arendst, 2019-07-15:
Have one and working on it.

arendst, 2019-07-15:
The mini makes a DIY mode available which might be used to OTA update Tasmota one day but it currently won't work.

So the good old serial flashing is needed.

As the mini does NOT contain a transformer AND the documentation alerts about high voltage on the Antenna I expect the whole unit to be on AC voltage so do not make a serial connection when AC is connected too; YOU WILL BLOW YOUR COMPUTER, like the Pow.

Luckily you can power the unit from the USB port during flashing. I used small wires connected to the testpoints as shown

[image]

Once flashed the below template makes the switch input available on GPIO04.
```
{"NAME":"Sonoff Mini","GPIO":[17,0,0,0,9,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":1}
```

ascillato2, 2019-07-15:
Closing this issue as support for this device has been added to Tasmota by Theo. Thanks

closed 2019-07-15 by ascillato2

meingraham, 2019-07-31:
@arendst 

From 16 days ago `The mini makes a DIY mode available which might be used to OTA update Tasmota one day but it currently won't work.`

DIY OTA now works; yes?
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY

arendst, 2019-07-31:
Yes. DIY OTA now works

tifrei, 2019-07-31:
Sorry for the dumb question but how do i use the template theo provided? Paste it in to the console?

meingraham, 2019-07-31:
https://github.com/arendst/Sonoff-Tasmota/wiki/Templates#importing-templates

On Wed, Jul 31, 2019 at 2:37 PM tifrei wrote:

> Sorry for the dumb question but how do i use the template theo provided?
> Paste it in to the console?
>

mattcony, 2019-08-03:
How do I go about exposing the Serial In & Serial Out pads using the template to allow for reassignment. They appear to be on GPIO01& GPIO03. Is it possible with the ESP8285 ?

meingraham, 2019-08-03:
@mattcony,

Getting to the GPIO pins on the Sonoff mini is pretty difficult. GPIO1 and
GPIO3 are not broken out on the PCB. You will have to solder into very
tight quarters. Having said that, if you can do so successfully, you just
assign a component to the second (01) and fourth GPIO (03) fields in the
template.

andraz, 2019-08-13:
I just successfully OTA flashed **sonoff-basic.bin** to Sonoff Mini and everything seems to work.

**Trick is to first upgrade OTA the firmware to 3.3.0 with the standard ewelink software.**

This enables you to use the DIY tools, which can flash OTA any firmware.
https://github.com/itead/Sonoff_Devices_DIY_Tools

I can't find Sonoff Mini in the Configuration->Module->"Module type" dropdown of Tasmota.
Which of the existing module types listed would be the most appropriate?

arendst, 2019-08-13:
See https://blakadder.github.io/templates/sonoff_mini.html

andraz, 2019-08-13:
Cool, this works. Thanks.

btw, Mini has 2 input ports, one is the physical pushbutton (it by defaults works like `power toggle` command) and the last two connectors (s1 & s2) next to power terminals, which can be connected together.

I assume GPIO16 is the s1-s2 connection. Is there a way to access the pushbutton gpio?

arendst, 2019-08-13:
Open the Template menu. This will show what GPIO does what. Your assumptions are wrong.

mmisnan, 2019-08-16:
How do we activate the S1/S2 switch? It doesn't work on mine currently.

rediculum, 2019-09-02:
Just for addition, this is my shocky wire soldering on the mini because DIY did not work
[image]
But succeeded flashing with USB TTY and NodeMCU PyFlasher

Rxich, 2019-09-05:
My shaky old hands just did the same thing. It really helps if you have some solder paste and can apply extremely tiny amounts. Dam those pads are very close together, as is the resistor, so be careful.

altimmons, 2019-11-16:
Using 30 AWG "wire wrapping wire" thats copper cored and pre-tinned (avail on amazon for sub $10) is the way to go.  

You can pop off the super thin insulation with your teeth. or just melt it with the iron as you go, and you get this super thin and stiff wire that will go right where you want it.

Pretty much like the first image, but thats probably 24-26 AWG

https://www.amazon.com/gp/product/B07WCTKKZL/ref=ppx_yo_dt_b_asin_title_o03_s01?ie=UTF8&psc=1

https://www.amazon.com/gp/product/B006C4A9TA/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

nagyrobi, 2019-11-27:
Guys, which pin is GPIO16 on the board? Is it the jumper?

meingraham, 2019-11-27:
Yes

qtronik, 2020-01-03:
Is one of the available GPIO pins can do one wire?
I'd like to add an DS18b20 on it but the template don't let me add it...
Is GPIO 16 or 02 could handle one wire ?

ErnieFR, 2020-02-15:
I learned that wireless flashing of the mini workst best if the hotspot has internet access too (mobile data=on). Only managed to flash from eWeLink software to Tasmota v6.6 by using my normal internetrouter and temporarily switch to the bizar network of Sonoff Mini. 
Anyways, I added the template as mentioned above in the 'configure other' menu, and the external switch came alive BUT then I upgraded to v8.1.0 and now I can't get the external switch to work. 
Am I confused? Yes, but can't find any other reason why else the external switch suddenly doesn't seem to work anymore. Tried 'reset config' and only put in wifi/template settings. Nothing. HELP

SzymonZy, 2020-03-04:
I flashed mini with 8.1.0 and the issue is that physical switch does not work. Tested with multimeter. Template imported but still no go.

SzymonZy, 2020-03-04:
What a shame ... that was the issue.

ErnieFR, 2020-03-05:
It happened to me too, and I guess lots of others alos, cause that little marker is made nearly invisible being on or off. Bad design, I hope next version will do something about it.

nagyrobi, 2020-03-05:
Just a quick observation: 
In case you connect long wires to the physical switch connectors (like for example to establish a 2-way switchning circuit) it may happen that Sonoff Mini won't detect the physical switching. 
On Sonoff forum ITEAD support suggested to add a 100nF capacitor to the switch input connectors like this:
[image]
I can confirm that this fixes the issue.

phrfpeixoto, 2020-04-08:
Hi @nagyrobi
Could you please let me know the specs of that cap?

nagyrobi, 2020-04-09:
It's a capacitor of 100nF, 40V (or more) rated. Usually it has marked "104" on.
I had a couple of 100nF 100V rated but those were too big to fit in the casing, I applied them outside to the screws. But 40V is more than enough (the voltage at those pins is only 3.3V).

phrfpeixoto, 2020-04-11:
@nagyrobi Thanks!

Jacksoft, 2020-04-17:
Hi, I measured the voltage between Earth and S1/S2 and it results as 120V. There's some schematic diagram of the Mini?
(I have the inverse problem. When I switch on the light it works for a while and then turn off like the switch was switched again. The switch was already replaced with a new one)
Thanks

BBBits, 2020-05-21:
Hi,
S1 is 0v esp8285
S2 has a 1k resistor between it &  GPIO4
GPIO4 has a 10k pullup to 3v
OTA is directly connected to GPIO16
GPIO16 has a 10k pullup to 3v
Onboard Button is directly connected to GPIO0
GPIO0 has a 10k pullup to 3v
the 0.1uF cap is not needed if the switch wires are only a few inches long.
The BIG confusion here is that the default configuration of GPIO0 is Button1 (17) .
This makes NO sense ! It  should be Button1i because it is Pulled low when Pressed.
is someone trying to reinvent the meaning of "inverted" or "not" ?

qtronik, 2020-05-22:
Marveillous gig for flashing with acupuncture needle!

[image]

SeanPaulE, 2020-10-09:
I bought a 3 pack from Amazon - my first one worked without a hitch on a 2way switch. Due to the wiring of the switch I had to re-wire a few things and released the magic smoke on two of them...

Now the 2nd/3rd won't work with the switch at all.

I've installed the 104 cap at the beginning of the week, this didn't help so I've just installed a new switch. Still no joy!

Anyone got any more suggestions at all? (It's the downstairs light so shorter cable run)

nagyrobi, 2020-10-12:
I had to put capacitors on both switches too, for my 2way setup, so that made 3 capacitors required to make it work stable. It all depends on how dense is your RF environment, if you have many electronics in the house, LED bulbs etc these all produce interference which can be picked up by the wires behaving like an antenna.

phrfpeixoto, 2020-10-14:
@nagyrobi Can you please post a pic of the cap on the switches?

Jason2866, 2020-10-14:
The picture shows 104-> 100nF

phrfpeixoto, 2020-10-14:
What I meant is where in the switch did he connect them.

nagyrobi, 2020-10-15:
[image]

pabloalcantara, 2020-12-22:
> Marveillous gig for flashing with acupuncture needle!
> 
> [image]

Hi,
Where do you bought this?

phrfpeixoto, 2020-12-22:
I guess he printed it. Still, I don't have a 3d printer, and would like to buy a jig like that

qtronik, 2020-12-24:
On thingiverse for .stl
Here the new version that I didn't tried...

https://www.thingiverse.com/thing:4618099

qtronik, 2020-12-24:
> I guess he printed it. Still, I don't have a 3d printer, and would like to buy a jig like that

In that time I didn't had a 3d printer but this worth all the effort for finding a friend who had and would 3dprint it!
I will NEVER regret to get this tool of mine!
And Chinese have super cheap 200 pack of copper acupuncture needles for sell on aliexpress or banggood!

qtronik, 2020-12-24:
> > Marveillous gig for flashing with acupuncture needle!
> > [image]
> 
> Hi,
> Where do you bought this?

I think you could still buy it from the original designer:
https://www.thingiverse.com/thing:3615910