arendst/Tasmota #6023: Tasmota on KS-601 Light Switch

martinst87, 2019-07-05:
Hi all, 

I successfully flashed some KS-601 wall light switches. 
They should be identical to Sonoff Touch. 
With 1 channel wall switches everything works fine, configured as Sonoff Touch. 

But with the 2 channel switches I´m having issues. 
According to this page it should have ESP8285 chip.  
https://www.letscontrolit.com/wiki/index.php?title=ESP_Hardware 
But mine has an ESP8266. 

If I configure it to be an Sonoff Touch 2CH relay 1 goes on and off all the time. 
Looks like Relay1 is connected to GPIO13 and not the LED. 
So I used "Generic" and played around until I found a configuration that works. 

LED = GPIO2
Button1 = GPIO16
Relay1 = GPIO13
Button2 = GPIO5
Relay2 = GPIO12

This setup works, but now it happens that the ESP is constantly restarting without any reason. Sometime I can turn on and off the lights several times, but somethimes it happens after every button press. 

Using Tasmota 6.5.0 and flashed sonoff-de.bin with NodeMCU-PyFlasher 4.0. 
I also have configured MQTT to send to my ioBroker installation. 

Any idea how to fix constant reboots? 

Thanks in advance

BR
Martin

Jason2866, 2019-07-05:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

martinst87, 2019-07-05:
Thanks for that. Posted my question there. Lets see how long it takes to get an response.

Jason2866, 2019-07-05:
Button1 = GPIO16  isnt possible since this GPIO is pulled to low!

closed 2019-07-05 by ascillato2

nils-hondong-imprivata, 2023-10-03:
Hi @martinst87, 

i have the same Kesen 2CH old and new one. 
For the new one i created a template with this config:

[image]