arendst/Tasmota #6026: Translation error on Russian language

ilya89vet, 2019-07-06:
In version 6.5.0 Fw sonoff-RU a part of the text is cut off
[image]

Here an example of correct translation in version 6.4.1
[image]

arendst, 2019-07-06:
As Russian (and others) use UTF-16 character set they need the double amount of char byte size. For next release it was raised to 64 but I notice that it is even too small. So I will raise it to 100.

arendst, 2019-07-06:
This looks better:

[image]

ascillato2, 2019-07-06:
Closing this issue as it has been solved. Thanks for reporting :+1: 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-06 by ascillato2