arendst/Tasmota #6027: IR Receiver On Tx or Rx pin

Adityakush, 2019-07-06:
I tried using IR receiver on the Rx pin and also on the tx pin which did not work even after disabling serial log. 
However these pins can be used as IR receiver as I can confirm using ESPHOME (just to confirm that its not something about the hardware itself). 
Maybe someone can point me in the right direction if I am wrong somewhere. 
Thanks

Jason2866, 2019-07-06:
Yes, IR Receive does not work on GPIO1 and GPIO3 in Tasmota.

Adityakush, 2019-07-06:
So that means its not gonna be added in the future also?

arendst, 2019-07-06:
Correct

ascillato2, 2019-07-06:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-06 by ascillato2

Adityakush, 2019-07-06:
Sorry for asking again but why cant these be used?