arendst/Tasmota #6028: Arduino Json fails

atuxnull, 2019-07-06:
Hi. Trying to complile v6.5.0 and i am getting the following error
`StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6`
I would like to install tasmota in 5 sonoff devices that i have, but it fails to compile. Any ideas on how to overcome this issue please?
Arduinojson.org/upgrade leads to editing the code, but even that failed in my case and still got the same error.

arendst, 2019-07-06:
ArduinoJson 6 is not supported.

All Tasmota supported libraries are provided in the source download.

ascillato2, 2019-07-06:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-06 by ascillato2

aherzer69, 2020-04-28:
I had the same problem. Downgrade Arduino sketch "ArduinoJson" to version 5.x