arendst/Tasmota #6029: Shelly 2.5 ignores SwitchMode 1 (continues to operate in SwitchMode 0)

bg1000, 2019-07-06:
Hardware: Shelly 2.5
Tasmota version: 6.5.0
Template used: {"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
From console: SwitchMode 1 ON - reply confirms that SwitchMode 1 is set

Results: both switches continue to operate in SwitchMode 0 (toggle mode) 

Note: Have previously used SwitchMode 1 with Shelly 1 switches so I am fairly confident I am setting it correctly.

Jason2866, 2019-07-06:
Please try latest release version. V.6.6.0

ascillato, 2019-07-06:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ptath, 2019-07-07:
Same problem

ascillato, 2019-07-07:
Without more information we can't help.

ascillato2, 2019-07-07:
When you have time, please complete the troubleshooting template in order to have more debug information so as to properly help you. Thanks

closed 2019-07-07 by ascillato2

ptath, 2019-07-07:
Problem gone on last 6.6.0 dev, exists on 6.5.0.11 and later. No need to reopen

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>