arendst/Tasmota #6030: DHT22 on GPIO14 causes device to crash and bootloop when relay turned on

DoctorOctagonapus, 2019-07-06:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
With DHT22 connected to GPIO14 and configured on the device as an AM2302, turning on the relay through any means causes the device to lock up completely and apparently start boot looping. Relay starts flicking on and off randomly, web interface is inaccessible, and device stops answering to pings or MQTT messages.

Device is still unstable even after removing sensor. Managed to get it to stay on long enough to get a status 0 output, but it dropped offline soon after. Unusable in its current state.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [x] Tasmota binary firmware version number used : 6.6.0 sonoff-sensor.bin / (pre-compiled or self-compiled ?)
- [x] Development IDE - Compiler / Upload tools used : ____ / esptool.py
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
22:53:09 CMD: status 0
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"bedroom","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Exception","Uptime":"0T00:00:23","StartupUTC":"2019-07-06T21:52:46","Sleep":50,"CfgHolder":4617,"BootCount":26,"SaveCount":44,"SaveAddress":"F8000"}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sensors)","BuildDateTime":"2019-07-06T13:12:39","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["_",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS4 = {"StatusMEM":{"ProgramSize":545,"Free":456,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDEE396","0017A3A4","B7FFBFCD","01DABBC0"]}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS5 = {"StatusNET":{"Hostname":"bedroom-0349","IPAddress":"_","Gateway":"_","Subnetmask":"_","DNSServer":"_","Mac":"_","Webserver":2,"WifiConfig":4}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS6 = {"StatusMQT":{"MqttHost":"_","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37A15D","MqttUser":"_","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 06 21:53:09 2019","Local":"Sat Jul 06 22:53:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:55"}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS10 = {"StatusSNS":{"Time":"2019-07-06T22:53:09"}}
22:53:09 MQT: home/bedroom/bedsidelamp/STATUS11 = {"StatusSTS":{"Time":"2019-07-06T22:53:09","Uptime":"0T00:00:23","Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"_","BSSId":"_","Channel":1,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
N/A: Web interface is inaccessible during issue so cannot see any output

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect DHT22 to GPIO14, configure as AM2302 as DHT is not an option, attempt to turn on relay through any method

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Basic function, relay to turn on and stay on, network to remain stable

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
DHT22 I know has never been officially supported, another issue elsewhere said to configure it as an AM2302 instead. This works fine provided you never turn the relay on.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-07-06:
Hi,

Sorry, I couldn't reproduce your issue, I have several devices with DTH22 and relays and never have that issue.

Have you erased all the flash before flashing with esptool.py?

Have you flashed as explained in the wiki using DOUT?

ascillato2, 2019-07-07:
@DoctorOctagonapus 

Hi, 

Any news on this? 

Have you erased all the flash before flashing with esptool.py?

Have you flashed as explained in the wiki using DOUT?

DoctorOctagonapus, 2019-07-07:
Hey, yes I did a flash_erase before and copied the program command from the wiki. I've reflashed it again without the sensor and it's fine, but adding the sensor was what caused it to break before.

ascillato, 2019-07-07:
Are you connecting the DHT correctly?Seems that it is making a short circuit or something.

ascillato, 2019-07-07:
Remember that a sonoff basic doesn't have enough power in its regulator for powering any sensor plus itself. You need to use another power source in parallel and have gnd connected.

ascillato2, 2019-07-07:
Closing this issue as it is a hardware issue and can't be solved in software, sorry. Please, add another power source for your sensor or use a nodemcu instead.

If you need further assistant, please do not hesitate on asking here or in the Tasmota support chat. Thanks

closed 2019-07-07 by ascillato2