arendst/Tasmota #6031: magichome Scheme4 效果的时候无法更改亮度，只有最低的亮度

qq24021438, 2019-07-07:
magichome Scheme4 效果的时候无法更改亮度，只有最低的亮度，控制台将亮度调为100%也无效

0n3-70uch, 2019-07-07:
Hello, can you write it in english please.

Jason2866, 2019-07-07:
without complete filled out issue template no one can help

cyclemat, 2019-07-07:
Here the translation 

he brightness cannot be changed when the effect is in effect. Only the lowest brightness, the console will adjust the brightness to 100%.

ascillato2, 2019-07-07:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-07 by ascillato2