arendst/Tasmota #6035: Hot to connect ADS1115, LM393 sensors with a WeMos D1

Ingo1977, 2019-07-07:
Hello, 
I Need your help in connectiong the ADS1115, LM393 sensors to a WeMos D1 chip correctly.
I connected it and for AnalogA0, I receive a value ~8500. The other three A1-A3 remain at ~3100.
If this correct and how can I Interpret the values?
I want to connect a rain sensor and a moisture sensor, each of them uses a LM383 chip in between.
Thanks Ingo

ascillato2, 2019-07-07:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-07 by ascillato2

Ingo1977, 2019-07-07:
I did but no answer so far. That's why I raised the Question here. Would be nice if somebody could help… Thanks in Advance.

lalo-uy, 2019-07-08:
The output is LM383 or LM393?  They are very different.
A0 only allow  0~1V input. Check your sensors swing and add the necessary divisor if needed.