arendst/Tasmota #6036: ENERGY_OVERTEMP as SetOption

cinemarene, 2019-07-07:
**Have you look for this feature in other issues and in the wiki?**
Yes, it seems as if some people need this function as well.
https://github.com/arendst/Sonoff-Tasmota/issues/5968#issuecomment-506975001

**Is your feature request related to a problem? Please describe.**
My Shelly1PM for my washing machine already switches off at 73 degrees. But there is a statement from Shelly that says that the first one switches off at 95 degrees.
https://m.facebook.com/groups/1686781668087857?view=permalink&id=2054834997949187

**Describe the solution you'd like**
ENERGY_OVERTEMP in xdrv_03_energy as SetOption

arendst, 2019-07-08:
Default changed from 73C to 90C and add command SetOption42 to change it.

cinemarene, 2019-07-08:
Thank you for the quick implementation.

closed 2019-07-08 by cinemarene