arendst/Tasmota #6037: RFBridge should pause between transmissions

abrileroux, 2019-07-07:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
When two requests to transmit a 433Mhz message are received by the RFBridge simultaneously, the RFBridge should queue the messages and transmit them one at a time. 

If you have multiple 433 devices (eg. lights), and one for example tell Home Assistant to "Turn on all lights", usually one one of the lights would respond. If you manually turn all the lights on, one by one, it works fine.

My suspicion is that if commands are received rapidly on the cmnd/rfbridge/rfcode MQTT, they get "trampled" on by each other somehow, and only one makes it out.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x Device used (i.e. Sonoff Basic) : Sonoff RF Bridge
- [x] Tasmota binary firmware version number used : 6.6.0 / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : N/A
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
16:26:53 MQT: rfbridge/stat/STATUS = {"Status":{"Module":25,"FriendlyName":["RFBridge"],"Topic":"rfbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:26:53 MQT: rfbridge/stat/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/020502/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:54","StartupUTC":"2019-07-07T15:15:59","Sleep":50,"CfgHolder":4617,"BootCount":38,"SaveCount":71,"SaveAddress":"FA000"}}
16:26:53 MQT: rfbridge/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-07-06T14:26:16","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
16:26:53 MQT: rfbridge/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IgorsRoost",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","280500000100060000000000000000000000","00000200"]}}
16:26:53 MQT: rfbridge/stat/STATUS4 = {"StatusMEM":{"ProgramSize":557,"Free":444,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
16:26:53 MQT: rfbridge/stat/STATUS5 = {"StatusNET":{"Hostname":"rfbridge-0565","IPAddress":"192.168.1.103","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.4.4","Mac":"60:01:94:CC:C2:35","Webserver":2,"WifiConfig":4}}
16:26:53 MQT: rfbridge/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CCC235","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:26:53 MQT: rfbridge/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 07 15:26:53 2019","Local":"Sun Jul 07 16:26:53 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:55","Sunset":"20:55"}}
16:26:53 MQT: rfbridge/stat/STATUS10 = {"StatusSNS":{"Time":"2019-07-07T16:26:53"}}
16:26:53 MQT: rfbridge/stat/STATUS11 = {"StatusSTS":{"Time":"2019-07-07T16:26:53","Uptime":"0T00:10:54","Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"IgorsRoost","BSSId":"0C:80:63:CF:88:93","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_

CONSOLE OUTPUT HERE:
Output for turning on two lights simultaneously

```
16:28:04 CMD: weblog 4
16:28:04 MQT: rfbridge/stat/RESULT = {"WebLog":4}
16:28:04 CFG: Saved to flash at F9, Count 72, Bytes 3584
16:28:09 WIF: Checking connection...
16:28:09 WIF: Connected
16:28:10 SRC: MQTT
16:28:10 RSL: Received Topic rfbridge/cmnd/rfcode, Data Size 7, Data #AAAAA3
16:28:10 RSL: Group 0, Index 1, Command RFCODE, Data #AAAAA3
16:28:10 MQT: rfbridge/stat/RESULT = {"RfCode":"#AAAAA3"}
16:28:10 SRC: MQTT
16:28:10 RSL: Received Topic rfbridge/cmnd/rfcode, Data Size 7, Data #AAAAA1
16:28:10 RSL: Group 0, Index 1, Command RFCODE, Data #AAAAA1
16:28:10 MQT: rfbridge/stat/RESULT = {"RfCode":"#AAAAA1"}
16:28:10 DMP: A0 55
```

### TO REPRODUCE

Trigger group.all_lights to turn on on Home Assistant.

### EXPECTED BEHAVIOUR
Each 433Mhz Message should be transmitted in turn, perhaps with a slight delay between each one, to ensure that each device has a chance to receive it.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Devices used:
RFBridge with Tasmota 6.6.0
Sesoo Touch panel wall light switches

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-07-07:
Already there. Use backlog with command delay to achieve this.
No need for changing something in Tasmota

abrileroux, 2019-07-07:
Thank you for the reply.
There could be potential situations where multiple transmission requests are received by the RF Bridge by coincidence. In this case the situation will still persist, and there would be no way to hard-code a backlog into it from client side. 

The client would expect to be able to write a transmission request, and have it transmitted. If two or more are sent at the same time, it should be handled by the RFBridge.

If I misunderstood you, please forgive me.

Jason2866, 2019-07-07:
If you send something like: 
`Backlog RfSync 11320; RfLow 360; RfHigh 1090; RfCode #151544; Delay 2`
all received commands will be handeled correct

abrileroux, 2019-07-07:
Alright, so just to confirm, and to help future users with this issue, Can you please confirm, that instead of configuring the MQTT client to send a message direct to cmnd/yoursonoff/rfcode, that clients should instead be configured to send messages as per the following example follows:

mosquitto_pub -t 'cmnd/yoursonoff/Backlog' -m ''RfSync 11320; RfLow 360; RfHigh 1090; RfCode #151544; Delay 2"

Thanks

abrileroux, 2019-07-07:
Ok, great. It just work. Thank you for your help. I really appreciate it. 
This is how they need to be set up in Hassio configuration.yaml for any other users who are curios:

`light:
  - platform: mqtt
    name: "Kitchen Ceiling Light"
    command_topic: rfbridge/cmnd/Backlog
    availability_topic: "rfbridge/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    payload_on: "RfSync 11320; RfLow 360; RfHigh 1090; RfCode #AAAAA1; Delay 2"
    payload_off: "RfSync 11320; RfLow 360; RfHigh 1090; RfCode #AAAAA2; Delay 2"
    optimistic: true    
  - platform: mqtt
    name: "Bathroom Light"
    command_topic: rfbridge/cmnd/Backlog
    availability_topic: "rfbridge/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    payload_on: "RfSync 11320; RfLow 360; RfHigh 1090; RfCode #AAAAA3; Delay 2"
    payload_off: "RfSync 11320; RfLow 360; RfHigh 1090; RfCode #AAAAA4; Delay 2"
    optimistic: true    
    `

closed 2019-07-07 by abrileroux

Digl, 2024-08-19:
I have a similiar issue but when receiving multiple rf singals from the bridge simultaneously.
The first received signal is detected, but all others are completely ignored during the transmission of the first one.
Can the Sonoff RF Bridge not receive multiple signals at once?
Thanks