arendst/Tasmota #6038: Effect 4 has the lowest brightness and cannot be changed

qq24021438, 2019-07-07:
I use !magic when I type in the console! Scheme 4 selects the fourth effect, the brightness of the RGB light effect is only 10%, and the brightness can not be controlled by the brightness bar, only the lowest brightness?

ascillato2, 2019-07-07:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-07-07:
@qq24021438 

Thanks for reporting. 

@arendst 

Hi, I can confirm that scheme 4 is not working anymore in latest version 6.6.0. In 6.4.x was working fine.

s-hadinger, 2019-07-08:
I'll take a look. Probably due to new light class.

arendst, 2019-07-08:
@s-hadinger when commenting out the below line it seems to work fine. I cannot see the consequences though.
```
void LightRandomColor(void)
{
  uint8_t light_update = 0;
  for (uint32_t i = 0; i < LST_RGB; i++) {
    if (light_new_color[i] != light_current_color[i]) {
      light_update = 1;
    }
  }
  if (!light_update) {
    light_wheel = random(255);
    LightWheel(light_wheel);
    memcpy(light_current_color, light_entry_color, sizeof(light_current_color));
//    light_controller.changeChannels(light_current_color);
  }
  LightFade();
}
```

s-hadinger, 2019-07-08:
@arendst You nailed it!

Indeed you can remove the commented line, it should not have been there in the first place.

ascillato2, 2019-07-08:
Fixed in ff2d235

Thanks :+1:

closed 2019-07-08 by ascillato2