arendst/Tasmota #6039: Tuya Module (54) on 6.6.0 gives 6 power switches

john5788, 2019-07-07:
Tuya dimmer (KS-7011) works fine on 6.5.0, but upgrading to 6.6.0 converts the device into something that has 6 power channels and the dimmer slider is gone. See attached images.

[image]

[image]

The device reports back to HomeAssistant as multiple switches instead of a light with the update. Toggling the leftmost bit in the web UI toggles the light switch on and off.

Jason2866, 2019-07-07:
@s-hadinger can you take a look into? Just selecting module 54 (Tuya) results in this.
Thx!

s-hadinger, 2019-07-07:
This part of Tasmota is still obsure to me. I will try to see when the regression happened.

arendst, 2019-07-07:
I'll investigate tomorrow.

s-hadinger, 2019-07-07:
I'm exploring commits. Up to https://github.com/arendst/Sonoff-Tasmota/commit/ab9baf2ac074280ea3fb3b008d8878cae66b547d (6.5.0.13) it still works.

arendst, 2019-07-07:
I did some changes in tuya in one of the latest commits ...

s-hadinger, 2019-07-07:
And broken in https://github.com/arendst/Sonoff-Tasmota/commit/12da2fd6f9b1e4ef4c2eb078aaeea425b695a8aa

arendst, 2019-07-08:
Caused by commit aaacb20 (thirug010) which uses uninitialised, undocumented and messy param setting (SetOption38) to simulate amount of switches.

As the use of param SetOption38 was not documented the same parameter is used and initialised with 6 by IRRemote support resulting in 6 toggles pop up.

Performing command ``SetOption38 1`` could be used as a workaround if IRReceive was enabled in the binary. This is not always the case making this issue even more problematic.

I will need to dig into this further and would request PR suppliers to more thoroughly test their submitted PR's.

To be continued.

arendst, 2019-07-08:
Use SetOption41 to increment number of relays/switches. Default set to zero to supply legacy support.

thirug010, 2019-07-08:
@arendst, My apologies for not using with the proper document Setting parameter Index

arendst, 2019-07-08:
@thirug010 accepted ;-)

ascillato2, 2019-07-08:
Closing as this issue has been fixed. Thanks :+1:

closed 2019-07-08 by ascillato2

sonybansi, 2019-09-02:
WiFi dimmer worked fine on full brightness and make a flikring at low brightness on Tasmota v6.5.0 , but after upgrading to Tasmota 6.6.0 show 6 switchs and  dimmer not working more. I tried to setoption 38 and 41 but no changes. See attached images.
[image]

[image]

[image]

[image]

ascillato, 2019-09-02:
Upgrade to latest 6.6.0.9. It is fixed in latest version

sonybansi, 2019-09-02:
Thanks! but dimmer not working just flashing a + led.

[image]

[image]

[image]
[image]