arendst/Tasmota #6042: gbridge.io + new version FW 6.6

ilya89vet, 2019-07-08:
I compiled new version 6.6 with having uncommented both options 
#define USE_MQTT_TLS and #define USE_MQTT_TLS_CA_CERT
After I tried to be connected to mqtt gbridge.io, BUT the device gives an error message "MQT: TLS connection error: 0" 
What do I not so do?

And still, if I at assembly of the binary file activated TLS I cannot use usual unprotected mqtt connection any more?

s-hadinger, 2019-07-08:
When using Letsencrypt, the cipher BR_TLS_RSA_WITH_AES_128_GCM_SHA256 is rejected by the server because considered not secure enough.

You can check with the following command:

`openssl s_client -showcerts -connect mqtt.gbridge.io:8883 -tls1_2 -cipher DHE-RSA-AES128-GCM-SHA256`

What you need is an hybrid between the current version and the AWS IoT version that uses BR_TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256. It should work as shown with the following command:

`openssl s_client -showcerts -connect mqtt.gbridge.io:8883 -tls1_2 -cipher ECDHE-RSA-AES128-GCM-SHA256`

I'll add it as an option, but I'm lacking time right now.

ilya89vet, 2019-07-08:
Problem as to me seems that I incorrectly collect Binary file! Except a raskomentirovaniye of two lines in my_user_config it is necessary to do something still?

ilya89vet, 2019-07-08:
> openssl s_client -showcerts -connect mqtt.gbridge.io:8883 -tls1_2 -cipher ECDHE-RSA-AES128-GCM-SHA256
I did not understand where I need to write this command?

s-hadinger, 2019-07-08:
You can run this command on any machine to check the server certificates. But you don't need to.

Can you try to compile from this commit ? https://github.com/s-hadinger/Sonoff-Tasmota/commit/edc898f3a360d3bfc72044c3843976955d08ab8e

ilya89vet, 2019-07-08:
> Можете ли вы попытаться скомпилировать из этой фиксации ? [s-hadinger@edc898f](https://github.com/s-hadinger/Sonoff-Tasmota/commit/edc898f3a360d3bfc72044c3843976955d08ab8e)
> 
> И раскомментировать:`#define USE_MQTT_TLS_FORCE_EC_CIPHER`

I already made it, But there is no positive result!

ilya89vet, 2019-07-08:
I work in Windows 10 and VSCode PlatformIO
I entered team in a command line
`openssl s_client - showcerts - connect mqtt.gbridge.io: 8883-ls1_2 - cipher DHE-RSA-AES128-GCM-SHA256` 
and it says that openssl is not found...
I need to establish OpenSSL still?

ilya89vet, 2019-07-08:
Perhaps a problem only at me? Maybe who else will check connection with **gbridge.io**

s-hadinger, 2019-07-08:
I doubled check, and I was mistaken. It is not a TLS problem. Actually the TLS connection completes normally. It is when MQTT connection is started that there is an issue. Maybe a user/password issue or Retained messages unsupported.

`TLS connection error: 0` means no error at the TLS layer.

Just forget everything I said above, and check your MQTT parameters.

ilya89vet, 2019-07-08:
Most likely the problem is in my gbridge.io account

ilya89vet, 2019-07-09:
The problem as to me seems in the certificate
Here a screenshot from my account gbridge
[image]

On android the client (Virtuino MQTT) I managed successfully it was connected to gbridge.io only to a psla of the indication of the file of certificates downloaded at this **[link](https://letsencrypt.org/certs/lets-encrypt-x3-cross-signed.pem.txt)**

timbru31, 2019-07-09:
I've had to enable the private key support, too: https://github.com/arendst/Sonoff-Tasmota/compare/development...s-hadinger:tls_force_ec#diff-abb73cfd1f384bb284b47c0788a61e8eR857 @ilya89vet. For me it connects to my mqtt broker successfully.

s-hadinger, 2019-07-09:
I tried again, and in both cases I saw the TLS connection working: with or without forcing EC (Elliptic Curve). @ilya89vet your problem is in the MQTT layer, not the TLS layer.

@timbru31 what mqtt broker are you using? Is it useful to merge "Force EC" into development? I'm still not sure this option is any useful.

timbru31, 2019-07-09:
@s-hadinger I'm using mosquitto on my Raspberry Pi. Since it's in my local network only I'm also my own CA. For me it's useful :) - and with 6.6 it was the only way to connect, with fingerprint and no CA verification I had a lot of crashes (like https://github.com/esp8266/Arduino/issues/6232)

s-hadinger, 2019-07-09:
@timbru31 You don't have crash anymore with this version? s-hadinger@edc898f

I've had devices connecting to AWS IoT (TLC EC with Private Key) and did not observe a single crash in days.

timbru31, 2019-07-09:
Correct. (I've made the changes locally myself with the addition of the EC Private Key enabled, too, as stated above. But I'm not 100% if this was needed). My test device is running >2 days now with no restart/crash.

PR #6077 (2019-07-12): Fix Force Elliptic Curve for Letsencrypt TLS #6042
## Description:

Add a compilation option to use EC ciphers with Letsencrypt.
No code change, just different #ifdef combinations. Adding `#define USE_MQTT_TLS_FORCE_EC_CIPHER`

I'm not sure it really fixes #6042, but anyways this might be useful in the future when MQTT brokers raise the requirements for TLS ciphers.

**Related issue (if applicable):** fixes #6042 (supposedly)

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  -  [... cut]
closed 2019-07-12 by arendst