arendst/Tasmota #6045: Compiling for Scripts in 6.6.0 fails with compilation error

davidgurr, 2019-07-08:
In user_config_override.h I have:
```
#undef USE_RULES
#define USE_SCRIPT
```

On compilation I get the following error:
```
C:/Users/emeai/Documents/Sonoff-Tasmota-6.6.0/sonoff/xdrv_10_scripter.ino: In function 'uint16_t GetStack()':
C:/Users/emeai/Documents/Sonoff-Tasmota-6.6.0/sonoff/xdrv_10_scripter.ino:1616:21: error: 'g_pcont' was not declared in this scope
return (4 * (sp - g_pcont.stack));
```

Also tried with "#undef USE_SCRIPT_FATFS" as well, but same result.

6.6.0, VSC with PlatformIO.

arendst, 2019-07-08:
What is your core version? Pls fill in the template to better help you.

davidgurr, 2019-07-08:
Hi Theo. 

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [x] Tasmota binary firmware version number used : 6.6.0, self-compiled
- [x] Development IDE - Compiler / Upload tools used : VSC with PlatformIO

The rest of the standard template is N/A for compile-time errors.

6.6.0 release notes say that core 2.3.0 is used in this release ... is that what you mean?

arendst, 2019-07-08:
As you compile yourself there is no means to know what core version you use. Providing the output of status 0 or at least the information within it makes helping you a lot easier.

Anyway, just found out that core 2.3.0 only causes this error. As a workaround use core 2.4.2 or higher for your use-case.

meingraham, 2019-07-08:
@gemu2015 Can you look into this to see why Core 2.3.0 fails to compile with 6.6.0? For cases where 2.4.2 or 2.5.2 is not an option due to Wi-Fi connectivity incompatibility, having 2.3.0 as an option for the scripting capability would be critical.

Thanks.

Mike

davidgurr, 2019-07-08:
@arendst I just downloaded the full source from https://github.com/arendst/Sonoff-Tasmota/archive/v6.6.0.zip

Where can I find instructions on using a different Core?

arendst, 2019-07-08:
PlatformIO handbook.

In short: edit platformio.ini by uncommenting core 2.4.2 and commenting core 2.3.0

```
[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_3_0.platform}
;build_flags               = ${core_2_3_0.build_flags}

platform                  = ${core_2_4_2.platform}
build_flags               = ${core_2_4_2.build_flags}

;platform                  = ${core_2_5_2.platform}
;build_flags               = ${core_2_5_2.build_flags}

;platform                  = ${core_stage.platform}
;build_flags               = ${core_stage.build_flags}
```

davidgurr, 2019-07-08:
Ah - beyond my current experience. Many thanks for the shortcut!

arendst, 2019-07-08:
Try the current dev branch. It contains solutions to your core 2.3.0 compile error(s)

gemu2015, 2019-07-08:
ok, located the incompatibilities in scripter with 2.3 but

also a freshly download dev version without scripter enabled does not compile with 2.3

section `.text' will not fit in region `iram1_0_seg'

arendst, 2019-07-08:
- Just downloaded dev
- Edit plarformio.ini by uncommenting env_default = sonoff
- Change core from 2.5.2 to 2.3.0
```
[core_active]
; Select one core set for platform and build_flags
platform                  = ${core_2_3_0.platform}
build_flags               = ${core_2_3_0.build_flags}
;platform                  = ${core_2_4_2.platform}
;build_flags               = ${core_2_4_2.build_flags}
;platform                  = ${core_2_5_2.platform}
;build_flags               = ${core_2_5_2.build_flags}
```
- Compiles just fine
[image]
- Edit my_user_config.h by commenting USE_RULES and uncommenting USE_SCRIPT
- Compiles just fine:
[image]

With latest changes to scripter.ino compilation errors are gone.

gemu2015, 2019-07-08:
thats exactly what i did  (one again)

there are no compile errors but a linker error  (platfomio) 

any advice ???

Jason2866, 2019-07-08:
@gemu2015  This happens to me if i use 
```
[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_3_0.platform}
;build_flags               = ${core_2_3_0.build_flags}
;platform                  = ${core_2_4_2.platform}
;build_flags               = ${core_2_4_2.build_flags}
;platform                  = ${core_2_5_2.platform}
;build_flags               = ${core_2_5_2.build_flags}
platform                  = ${core_stage.platform}
;uild_flags               = ${core_stage.build_flags}
```
feature/stage is broken since 2? month.

arendst, 2019-07-08:
Stage works just fine too.

I suggest you do a pio update

Jason2866, 2019-07-08:
Nope!
```
PS D:\Sonoff-Tasmota-development.6.6.0.1> platformio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.190418    [Up-to-date]
Updating contrib-piohome                 @ 2.2.0          [Up-to-date]
Updating tool-pioplus                    @ 2.5.2          [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.2.2          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.190218 [Up-to-date]

```

Jason2866, 2019-07-08:
@arendst  Try with gitpod. You will get the linker error

ascillato, 2019-07-08:
Hi,

In my environment (latest Platformio in a PC), latest Tasmota (6.6.0.1) is compiling fine with core 2.3.0 with and without `#define USE_SCRIPT`

For the Stage to work **_without_** `#define USE_SCRIPT`, you have to download ALL the files manually. Now there are some symlinks in the Arduino ESP8266 repository that are not automatically downloaded with **git** like for example the library of the following picture.

[image]

If you download all the core with the **download zip** function of github, it works fine. If you do a git with Github Desktop Application, it will not download all the required files making the stage core to fail compiling. Also, the toolchain has to be up to date. So may be this is what is happening to gitpod ?

If you use `#define USE_SCRIPT` with the STAGE core it will **not work**. Script uses the SD FAT library but in the stage core that library has been replaced, so the _xdrv_10_scripter.ino_ must be updated to latest library (for stage core) in order to work:

```
Fat\src/SdFat.h:33,
from C:\Users\Usuario\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\libraries\SDFS\src/
SDFS.h:36,
from C:\Users\Usuario\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\libraries\SD\src/SD
.h:25,
from C:/Users/Usuario/Desktop/Sonoff-Tasmota-development/sonoff/xdrv_10_scripter.ino:129:
C:\Users\Usuario\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\libraries\ESP8266SdFat\s

rc/FatLib/FatStructs.h:799:24: note:   'sdfat::FAT_TIME'
static inline uint16_t FAT_TIME(uint8_t hour, uint8_t minute, uint8_t second) {
^
C:/Users/Usuario/Desktop/Sonoff-Tasmota-development/sonoff/xdrv_10_scripter.ino: In function 'bool Xdrv10(uint8_t)':
C:/Users/Usuario/Desktop/Sonoff-Tasmota-development/sonoff/xdrv_10_scripter.ino:6017:9: error: 'SdFile' has not been declared
*** [.pioenvs\sonoff-ES_4MB\src\sonoff.ino.cpp.o] Error 1
```

ascillato2, 2019-07-08:
The SD FAT library was changed in core 2.5.2, so scripting is not working under core 2.5.2 and Stage.

You need to use core 2.3.0, 2.4.2 or 2.5.0 for scripting to compile until a fix to xdrv_10_scripter.ino is done.

gemu2015, 2019-07-08:
The sd Card Support is optional. Most users wont need or want it
So simple comment it out

ascillato, 2019-07-08:
Confirmed.

commenting `//  #define USE_SCRIPT_FATFS 4` in _my_user_config.h_ file makes the scripter to compile in all cores and also in STAGE.

ascillato2, 2019-07-09:
Closing this issue as now Scripter is working in all cores and stage.

The optional USE_SCRIPT_FATFS works only for cores 2.3.0, 2.4.2 and 2.5.0 until a fix for the new SDFAT library of core 2.5.2 and Stage is done.

Thanks everyone for working on this.

closed 2019-07-09 by ascillato2

gemu2015, 2019-07-09:
ok, i checked the differences in FAT support  for 2.52 

USE_LONG_FILE_NAMES is hardcoded in 2.52  as i use only short names for memory reasons
this can only be corrected by editing the arduino core in SdFatConfig.h
also  the macros FAT_DATE and FAT_TIME are not included and the callback for setting the file creation time is not defined.

regarding core 2.3  deleting .platformio solved the linker issue
but i can confirm @Jason2866 that stage core  still throws a linker error.

Jason2866, 2019-07-09:
@ascillato  Platformio (running with VSC) DOES download the symlink files. Everything is in the folders.
I use rules, so no use of scripter engine. Compiling fails as i wrote before.

ascillato, 2019-07-09:
Just for the record on the scripting issue:

With PR #6048 from @gemu2015, now the scripter works on all cores and stage also with the sdfat option enabled.

Thanks everyone for working on this :+1:

ascillato2, 2019-07-09:
@Jason2866

You are right. I hadn't that linker issue due to I had Emulation disabled so my builds used less iram.

Anyone having this linker error with stage core, please follow the issue #6047