arendst/Tasmota #6047: Linker error with stage core

Jason2866, 2019-07-09:
### BUG DESCRIPTION
Linker error with core feature/stage

```
Indexing .pioenvs/sonoff/libFrameworkArduino.a
Linking .pioenvs/sonoff/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff/firmware.elf] Error 1
============================================= [ERROR] Took 124.28 seconds =============================================
```

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : none, compiler/linker error
- [x] Tasmota binary firmware version number used : latest dev 6.6.01 / (pre-compiled)
- [X] Development IDE - Compiler / VSC with platformIO 
- [ ] Provide the output of command ``status 0`` :

### TO REPRODUCE
_Steps to reproduce the behavior:_

Compiling with in platformio.ini

```
env_default = sonoff

platform                  = ${core_stage.platform}
build_flags               = ${core_stage.build_flags}

```

### EXPECTED BEHAVIOUR
Successfull linking

See https://github.com/arendst/Sonoff-Tasmota/issues/6045#issuecomment-509298834 too

arendst, 2019-07-09:
I see the same "issue".

This is not really a Tasmota problem. As the message states the Tasmota .text1 does not fit in iram region simply because the stage core uses more iram than core 2.5.2

As core people are currently working on moving code between heap and flash this is kind of expected. I suggest you monitor Arduino esp8266 github issues for future solutions.

Current Tasmota workaround would be to disable some features using iram (interrupt routine using options) to make it compile again.

Jason2866, 2019-07-09:
Will open a issue in Arduino esp8266.
I think we should leave this issue open.

arendst, 2019-07-09:
Agree

arendst, 2019-07-09:
Background info gained with my config override:
```
**** Core 2.5.2

Linking .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Retrieving maximum program size .pioenvs\sonoff\firmware.elf
Checking size .pioenvs\sonoff\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  55.0% (used 45040 bytes from 81920 bytes)
PROGRAM: [=====     ]  48.9% (used 500348 bytes from 1023984 bytes)
Creating BIN file ".pioenvs\sonoff\firmware.bin" using ".pioenvs\sonoff\firmware.elf"

readelf -l firmware.elf:

Elf file type is EXEC (Executable file)
Entry point 0x4010065c
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x0000d4 0x00000000 0x00000000 0x00000 0x00000     0x1000
  LOAD           0x001000 0x3ffe8000 0x3ffe8000 0x01424 0x01424 RW  0x1000
  LOAD           0x002428 0x3ffe9428 0x3ffe9428 0x00000 0x09bd8 RW  0x1000
  LOAD           0x003000 0x40100000 0x40100000 0x07838 **0x07838** R E 0x1000
  LOAD           0x00b010 0x40201010 0x40201010 0x7162c 0x7162c R E 0x1000

 Section to Segment mapping:
  Segment Sections...
   00
   01     .data .noinit .rodata
   02     .bss
   03     .text .text1
   04     .irom0.text

**** Core Stage (20190709)

Linking .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Retrieving maximum program size .pioenvs\sonoff\firmware.elf
Checking size .pioenvs\sonoff\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  55.1% (used 45172 bytes from 81920 bytes)
PROGRAM: [=====     ]  48.5% (used 496768 bytes from 1023984 bytes)
Creating BIN file ".pioenvs\sonoff\firmware.bin" using ".pioenvs\sonoff\firmware.elf"

readelf -l firmware.elf:

Elf file type is EXEC (Executable file)
Entry point 0x401000b8
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x0000d4 0x00000000 0x00000000 0x00000 0x00000     0x1000
  LOAD           0x001000 0x3ffe8000 0x3ffe8000 0x01420 0x01420 RW  0x1000
  LOAD           0x002420 0x3ffe9420 0x3ffe9420 0x00000 0x09c60 RW  0x1000
  LOAD           0x003000 0x40100000 0x40100000 0x07c70 **0x07c70** R E 0x1000
  LOAD           0x00b010 0x40201010 0x40201010 0x703fc 0x703fc R E 0x1000

 Section to Segment mapping:
  Segment Sections...
   00
   01     .data .noinit .rodata
   02     .bss
   03     .text .text1
   04     .irom0.text
```
From the above info section 03 contains the iram usage. The default linker allows for max 0x08000 bytes iram. As can be seen Stage uses a lot more iram (0x7c70) than 2.5.2 (0x7838). Enabling more features would grow iram usage beyond 0x8000 and fail linking.

arendst, 2019-07-09:
In addition a core 2.5.2 default image of sonoff.bin will use:
```
Elf file type is EXEC (Executable file)
Entry point 0x40100b0c
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x0000d4 0x00000000 0x00000000 0x00000 0x00000     0x1000
  LOAD           0x001000 0x3ffe8000 0x3ffe8000 0x016e8 0x016e8 RW  0x1000
  LOAD           0x0026e8 0x3ffe96e8 0x3ffe96e8 0x00000 0x09ee0 RW  0x1000
  LOAD           0x003000 0x40100000 0x40100000 0x07ce8 **0x07ce8** R E 0x1000
  LOAD           0x00b010 0x40201010 0x40201010 0x805dc 0x805dc R E 0x1000

 Section to Segment mapping:
  Segment Sections...
   00
   01     .data .noinit .rodata
   02     .bss
   03     .text .text1
   04     .irom0.text
```
0x7ce8 bytes of iram. Adding the Stage difference of 0x7c70 - 0x7838 = 0x438 bytes will tip the scale by 0x120 bytes above 0x8000 and fail linking.

Jason2866, 2019-07-10:
iram space will be left as it is for stage and upcoming core 2.6.0
Explained here https://github.com/esp8266/Arduino/issues/6275#issuecomment-510070822

arendst, 2019-07-10:
Great. So we have to remove features from binaries to make them fit.

Jason2866, 2019-07-10:
It seems I2C needs a lot. Which is a problem, because most used sensors are I2C

ascillato, 2019-07-10:
Just disabling `USE_RC_SWITCH` in _my_user_config.h_ file, Tasmota compiles with default settings on Stage core without issue.

```
//#define USE_RC_SWITCH                            // Add support for RF transceiver using library RcSwitch (+2k7 code, 460 iram)
```

ascillato, 2019-07-10:
Made the PR #6060 for review so as to disable this driver (USE_RC_SWITCH).

It is not needed anymore due those protocols are now supported using portish firmware in the RF chip of the sonoff bridge.

closed 2019-07-10 by arendst

ascillato, 2019-07-10:
@arendst Sorry, my bad. With further testings we did with @Jason2866, disabling USE_RC_SWITCH is not enough.

We have to disable also `USE_TX20_WIND_SENSOR` in order to free enough iram for STAGE to compile.

Do you agree to disable also this sensor from the defaults?

Jason2866, 2019-07-10:
Disabled TX20 wind sensor (PR #6062)

Jason2866, 2019-07-10:
[image]

PR #6060 (2019-07-10): Disable USE_RC_SWITCH by default ...
## Description:

Disable USE_RC_SWITCH by default in order to compile with latest Stage Core.
The USE_RC_SWITCH driver uses too much iram for the latest core.

**Related issue (if applicable):** fixes #6047 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CL [... cut]