arendst/Tasmota #6050: Pzem AC, sometimes all value show = 0

ninhvanthang, 2019-07-09:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION

i'm using pzem-004T V3 with sonoff basic to measure power,
i've just upgrade from 6.5 to 6.6 and got bug of sensor, sometime all values show 0 on tasmota

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : sonoff basic
- [x] Tasmota binary firmware version number used : ____ / pre-compiled
- [x] Development IDE - Compiler / Upload tools used : ____ / pyflasher
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

23:22:53 CMD: status 0
23:22:53 SRC: WebConsole from 192.168.1.121
23:22:53 RSL: Received Topic /status, Data Size 1, Data 0
23:22:53 RSL: Group 0, Index 1, Command STATUS, Data 0
23:22:53 MQT: stat/powermeter/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"powermeter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:22:53 MQT: stat/powermeter/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:26:40","StartupUTC":"2019-07-09T14:56:13","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":59,"SaveAddress":"F8000"}}
23:22:53 MQT: stat/powermeter/STATUS2 = {"StatusFWR":{"Version":"6.6.0.1(sonoff)","BuildDateTime":"2019-07-09T21:54:47","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
23:22:53 MQT: stat/powermeter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["THANGA",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A00000000000000","00000000"]}}
23:22:53 MQT: stat/powermeter/STATUS4 = {"StatusMEM":{"ProgramSize":420,"Free":580,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","00802182","04008000","0000001D","000002C0"]}}
23:22:53 MQT: stat/powermeter/STATUS5 = {"StatusNET":{"Hostname":"powermeter-7470","IPAddress":"192.168.1.129","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:AF:5D:2E","Webserver":2,"WifiConfig":4}}
23:22:53 MQT: stat/powermeter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.138","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF5D2E","MqttUser":"hassio","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:22:53 MQT: stat/powermeter/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 09 16:22:53 2019","Local":"Tue Jul 09 23:22:53 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00"}}
23:22:53 MQT: stat/powermeter/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:22:53 MQT: stat/powermeter/STATUS10 = {"StatusSNS":{"Time":"2019-07-09T23:22:53","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":11.852,"Yesterday":0.000,"Today":11.852,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
23:22:53 MQT: stat/powermeter/STATUS11 = {"StatusSTS":{"Time":"2019-07-09T23:22:53","Uptime":"0T01:26:40","Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":11,"RSSI":46,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

23:14:00 WIF: Connected
23:14:05 DMP: 01 04 14 08 B7 08 9F 00 00 12 B8 00 00 FE D4 00 01 01 F4 00 61 00 00 F5 98
23:14:18 DMP: 01 04 14 08 B0 07 85 00 00 10 1D 00 00 FE D6 00 01 01 F4 00 60 00 00 9F 5A
23:14:19 DMP: 01 04 14 08 B0 07 85 00 00 10 1D 00 00 FE D6 00 01 01 F3 00 60 00 00 2A 9A
23:14:20 DMP: 01 04 14 08 B1 07 86 00 00 10 20 00 00 FE D6 00 01 01 F4 00 60 00 00 00 EB
23:14:20 WIF: Checking connection...
23:14:20 WIF: Connected
23:14:21 DMP: 01 04 14 08 B2 07 96 00 00 10 3C 00 00 FE D6 00 01 01 F4 00 60 00 00 66 36
23:14:22 DMP: 01 04 14 08 B3 07 A4 00 00 10 69 00 00 FE D6 00 01 01 F3 00 60 00 00 97 30
23:14:30 MQT: tele/powermeter/STATE = {"Time":"2019-07-09T23:14:30","Uptime":"0T01:18:17","Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":11,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:14:30 MQT: tele/powermeter/SENSOR = {"Time":"2019-07-09T23:14:30","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":11.827,"Yesterday":0.000,"Today":11.827,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
23:14:35 DMP: 01 04 14 08 AF 07 80 00 00 10 0D 00 00 FE D8 00 01 01 F3 00 60 00 00 D8 ED
23:14:40 WIF: Checking connection...
23:14:40 WIF: Connected
23:14:42 DMP: 01 04 14 08 B3 07 7E 00 00 10 12 00 00 FE D9 00 01 01 F3 00 60 00 00 08 20
[... 52 lines cut ...]
23:16:35 DMP: 01 04 14 08 AC 07 12 00 00 0F 0E 00 00 FE E5 00 01 01 F4 00 60 00 00 F3 A9
23:16:40 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

open tasmota web interface

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

sometimes, all sensors value random show: 0

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-07-09:
Your log shows you are missing some data during several seconds. The latest release expect the sensor disconnected after, I thought 8 seconds of no data received. It then returns only zeros..

You have several debug options.
- try another core version and verify regular data reception
- chk you are using hardware serial
- go back to previous version and verify regular data reception

ninhvanthang, 2019-07-10:
I've just built from core 2.3.0 and this bug still exist
Console Log:

`00:00:00 CFG: Loaded from flash at FB, Count 70
00:00:00 Project sonoff Sonoff Version 6.6.0.1(sonoff)-2_3_0
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 THANGA in mode 11N as powermeter-7470...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on powermeter-7470 with IP address 192.168.1.129
00:00:05 MQT: Attempting connection...
08:09:37 MQT: Connected
08:09:37 MQT: tele/powermeter/LWT = Online (retained)
08:09:37 MQT: cmnd/powermeter/POWER = 
08:09:37 MQT: tele/powermeter/INFO1 = {"Module":"Sonoff Basic","Version":"6.6.0.1(sonoff)","FallbackTopic":"cmnd/DVES_AF5D2E_fb/","GroupTopic":"sonoffs"}
08:09:37 MQT: tele/powermeter/INFO2 = {"WebServerMode":"Admin","Hostname":"powermeter-7470","IPAddress":"192.168.1.129"}
08:09:37 MQT: tele/powermeter/INFO3 = {"RestartReason":"Software/System restart"}
08:09:37 MQT: stat/powermeter/RESULT = {"POWER":"OFF"}
08:09:37 MQT: stat/powermeter/POWER = OFF
08:09:45 MQT: tele/powermeter/STATE = {"Time":"2019-07-10T08:09:45","Uptime":"0T00:00:15","Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":1,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:04"}}
08:09:45 MQT: tele/powermeter/SENSOR = {"Time":"2019-07-10T08:09:45","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":24.228,"Yesterday":11.990,"Today":12.238,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
08:10:46 MQT: tele/powermeter/STATE = {"Time":"2019-07-10T08:10:46","Uptime":"0T00:01:16","Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":1,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:04"}}
08:10:46 MQT: tele/powermeter/SENSOR = {"Time":"2019-07-10T08:10:46","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":24.228,"Yesterday":11.990,"Today":12.238,"Period":0,"Power":303,"ApparentPower":313,"ReactivePower":79,"Factor":0.97,"Frequency":50,"Voltage":220,"Current":1.422}}
08:10:48 LOG: SerialLog 0, WebLog 4, SysLog 2, LogHost , LogPort 514, TelePeriod 60
08:10:48 HTP: Configuration
08:10:49 CFG: Saved to flash at F9, Count 73, Bytes 3584
08:10:53 DMP: 01 04 14 08 77 05 91 00 00 0B A6 00 00 0C 85 00 02 01 F5 00 61 00 00 54 D8
08:10:54 DMP: 01 04 14 08 83 05 9D 00 00 0B D8 00 00 0C 86 00 02 01 F5 00 61 00 00 AE B0
08:10:55 DMP: 01 04 14 08 83 05 9D 00 00 0B D8 00 00 0C 86 00 02 01 F5 00 61 00 00 AE B0
08:10:57 WIF: Checking connection...
08:10:57 WIF: Connected
08:11:02 DMP: 01 04 14 08 87 05 89 00 00 0B AA 00 00 0C 86 00 02 01 F5 00 61 00 00 A8 F6
08:11:03 DMP: 01 04 14 08 90 05 8D 00 00 0B BC 00 00 0C 86 00 02 01 F5 00 60 00 00 FD 16
08:11:04 DMP: 01 04 14 08 90 05 8D 00 00 0B BC 00 00 0C 86 00 02 01 F5 00 60 00 00 FD 16
08:11:11 DMP: 01 04 14 08 8F 05 95 00 00 0B D2 00 00 0C 87 00 02 01 F5 00 61 00 00 2F B9
08:11:12 DMP: 01 04 14 08 8F 05 95 00 00 0B D2 00 00 0C 87 00 02 01 F4 00 61 00 00 12 79
08:11:13 DMP: 01 04 14 08 90 05 8C 00 00 0B BD 00 00 0C 87 00 02 01 F5 00 61 00 00 F2 57
08:11:14 DMP: 01 04 14 08 8F 05 9E 00 00 0B E7 00 00 0C 87 00 02 01 F4 00 61 00 00 28 BE
08:11:15 DMP: 01 04 14 08 8E 05 8C 00 00 0B B9 00 00 0C 87 00 02 01 F4 00 61 00 00 6F 39
08:11:17 WIF: Checking connection...
08:11:17 WIF: Connected
08:11:22 DMP: 01 04 14 08 91 05 8A 00 00 0B B9 00 00 0C 88 00 02 01 F4 00 61 00 00 5E BE
08:11:28 HTP: Main Menu
08:11:32 HTP: Main Menu
08:11:35 DMP: 01 04 14 08 92 05 A9 00 00 0C 0A 00 00 0C 89 00 02 01 F4 00 61 00 00 43 65
08:11:36 DMP: 01 04 14 08 8A 05 8B 00 00 0B B0 00 00 0C 89 00 02 01 F5 00 60 00 00 11 91
08:11:36 HTP: Main Menu
08:11:37 DMP: 01 04 14 08 7D 05 93 00 00 0B BA 00 00 0C 89 00 02 01 F4 00 61 00 00 29 FD
08:11:38 WIF: Checking connection...
08:11:38 WIF: Connected
08:11:48 MQT: tele/powermeter/STATE = {"Time":"2019-07-10T08:11:47","Uptime":"0T00:02:17","Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":1,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:04"}}
08:11:48 MQT: tele/powermeter/SENSOR = {"Time":"2019-07-10T08:11:48","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":24.229,"Yesterday":11.990,"Today":12.239,"Period":1,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
08:11:50 DMP: 01 04 14 08 88 05 8E 00 00 0B B8 00 00 0C 8A 00 02 01 F5 00 61 00 00 21 00
08:11:51 DMP: 01 04 14 08 8B 05 93 00 00 0B CB 00 00 0C 8A 00 02 01 F5 00 61 00 00 46 E1
08:11:52 DMP: 01 04 14 08 88 05 87 00 00 0B AC 00 00 0C 8A 00 02 01 F5 00 61 00 00 C7 46
08:11:58 WIF: Checking connection...
08:11:58 WIF: Connected
08:12:05 DMP: 01 04 14 08 86 04 C0 00 00 09 CA 00 00 0C 8B 00 02 01 F5 00 5E 00 00 C9 8E
08:12:06 DMP: 01 04 14 08 8C 04 C0 00 00 09 D0 00 00 0C 8C 00 02 01 F5 00 5E 00 00 52 FD
08:12:07 DMP: 01 04 14 08 8C 04 C0 00 00 09 D0 00 00 0C 8C 00 02 01 F4 00 5E 00 00 6F 3D
08:12:18 WIF: Checking connection...
08:12:18 WIF: Connected
08:12:20 DMP: 01 04 14 08 91 04 C0 00 00 09 D5 00 00 0C 8D 00 02 01 F5 00 5E 00 00 B9 06
08:12:21 DMP: 01 04 14 08 91 04 C0 00 00 09 D5 00 00 0C 8D 00 02 01 F5 00 5E 00 00 B9 06
08:12:22 DMP: 01 04 14 08 94 04 C2 00 00 09 DC 00 00 0C 8D 00 02 01 F5 00 5E 00 00 C9 63
08:12:23 DMP: 01 04 14 08 89 04 BA 00 00 09 C2 00 00 0C 8D 00 02 01 F5 00 5E 00 00 A9 0A
[... cut ...]
08:06:13 MQT: tele/powermeter/STATE = {"Time":"2019-07-10T08:06:13","Uptime":"0T00:40:30","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":1,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:38"}}
08:06:13 MQT: tele/powermeter/SENSOR = {"Time":"2019-07-10T08:06:13","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":24.212,"Yesterday":11.990,"Today":12.222,"Period":5,"Power":313,"ApparentPower":322,"ReactivePower":75,"Factor":0.97,"Frequency":50,"Voltage":218,"Current":1.473}}`

ninhvanthang, 2019-07-10:
I tried change time to retry to 10 `send_retry = 10;` but bug still exist

arendst, 2019-07-10:
When using 6.5 also enable more logging with weblog 4 and report if you receive data every second.

As a workaround for 6.6 you might want to increase define ENERGY_WATCHDOG in xdrv_03_energy.ino from 4 to 40 or even higher to stop resetting measured values.

ninhvanthang, 2019-07-10:
Please check log for 6.5
I'll try edit xdrv_03_energy and report to you today.

```
16:43:29 DMP: 01 04 14 08 91 08 07 00 00 11 29 00 00 1C A1 00 02 01 F6 00 61 00 00 68 25
16:43:30 DMP: 01 04 14 08 92 08 61 00 00 11 F8 00 00 1C A1 00 02 01 F6 00 62 00 00 2A CE
16:43:31 DMP: 01 04 14 08 92 08 61 00 00 11 F8 00 00 1C A2 00 02 01 F6 00 62 00 00 3E 3E
16:43:32 WIF: Checking connection...
16:43:32 WIF: Connected
16:43:32 DMP: 01 04 14 08 7F 08 4F 00 00 11 AF 00 00 1C A2 00 02 01 F6 00 62 00 00 11 FB
16:43:33 MQT: tele/powermeter/STATE = {"Time":"2019-07-10T16:43:33","Uptime":"0T07:52:27","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":1,"RSSI":52,"LinkCount":74,"Downtime":"0T00:04:02"}}
16:43:33 MQT: tele/powermeter/SENSOR = {"Time":"2019-07-10T16:43:33","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":28.149,"Yesterday":11.990,"Today":16.159,"Period":9,"Power":453,"ApparentPower":463,"ReactivePower":95,"Factor":0.98,"Frequency":50,"Voltage":218,"Current":2.127}}
16:43:33 DMP: 01 04 14 08 7C 08 2D 00 00 11 5B 00 00 1C A2 00 02 01 F6 00 62 00 00 92 B4
16:43:34 DMP: 01 04 14 08 8A 08 73 00 00 12 19 00 00 1C A2 00 02 01 F6 00 62 00 00 CB DA
16:43:35 DMP: 01 04 14 08 8A 08 73 00 00 12 19 00 00 1C A2 00 02 01 F6 00 62 00 00 CB DA
16:43:42 DMP: 01 04 14 08 8C 08 B8 00 00 12 A0 00 00 1C A3 00 02 01 F6 00 62 00 00 CA 34
16:43:43 DMP: 01 04 14 08 91 08 AA 00 00 12 9C 00 00 1C A3 00 02 01 F6 00 62 00 00 DC C3
16:43:50 DMP: 01 04 14 08 98 09 03 00 00 13 72 00 00 1C A4 00 02 01 F6 00 62 00 00 B2 CC
16:43:51 DMP: 01 04 14 08 99 08 FF 00 00 13 6A 00 00 1C A4 00 02 01 F6 00 62 00 00 7E 24
16:43:52 WIF: Checking connection...
16:43:52 WIF: Connected
16:43:52 DMP: 01 04 14 08 99 08 F4 00 00 13 56 00 00 1C A4 00 02 01 F6 00 62 00 00 59 2A
16:43:59 DMP: 01 04 14 08 99 08 F2 00 00 13 4B 00 00 1C A5 00 02 01 F6 00 62 00 00 9A C6
16:44:12 WIF: Checking connection...
16:44:12 WIF: Connected
16:44:12 MQT: Attempting connection...
16:44:17 MQT: Connected
16:44:17 MQT: tele/powermeter/LWT = Online (retained)
16:44:17 MQT: cmnd/powermeter/POWER = 
16:44:17 MQT: Subscribe to cmnd/powermeter/#
16:44:17 MQT: Subscribe to cmnd/sonoffs/#
16:44:17 MQT: Subscribe to cmnd/DVES_AF5D2E_fb/#
16:44:18 DMP: 01 04 14 08 89 08 FF 00 00 13 41 00 00 1C A8 00 02 01 F5 00 62 00 00 CE 0C
16:44:19 DMP: 01 04 14 08 81 09 15 00 00 13 78 00 00 1C A8 00 02 01 F6 00 62 00 00 0A FC
16:44:26 DMP: 01 04 14 08 7D 09 A3 00 00 14 98 00 00 1C A9 00 02 01 F5 00 62 00 00 CD C8
16:44:27 DMP: 01 04 14 08 7B 09 9E 00 00 14 9D 00 00 1C A9 00 02 01 F5 00 63 00 00 98 19
16:44:28 HTP: Console
16:44:34 DMP: 01 04 14 08 7B 08 FF 00 00 13 2B 00 00 1C AA 00 02 01 F5 00 62 00 00 44 6B
16:44:36 WIF: Checking connection...
16:44:36 WIF: Connected
16:44:38 MQT: tele/powermeter/STATE = {"Time":"2019-07-10T16:44:38","Uptime":"0T07:53:32","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"THANGA","BSSId":"C4:71:54:D2:37:46","Channel":1,"RSSI":56,"LinkCount":74,"Downtime":"0T00:04:02"}}
16:44:38 MQT: tele/powermeter/SENSOR = {"Time":"2019-07-10T16:44:38","ENERGY":{"TotalStartTime":"2019-07-08T23:56:49","Total":28.157,"Yesterday":11.990,"Today":16.167,"Period":8,"Power":491,"ApparentPower":500,"ReactivePower":96,"Factor":0.98,"Frequency":50,"Voltage":217,"Current":2.303}}
16:44:41 DMP: 01 04 14 08 7C 08 F2 00 00 13 0A 00 00 1C AB 00 02 01 F6 00 62 00 00 51 FD
16:44:42 DMP: 01 04 14 08 7D 08 F9 00 00 13 23 00 00 1C AB 00 02 01 F5 00 62 00 00 3D 9A
16:44:43 DMP: 01 04 14 08 7D 08 F9 00 00 13 23 00 00 1C AB 00 02 01 F5 00 62 00 00 3D 9A
16:44:44 DMP: 01 04 14 08 7A 09 09 00 00 13 43 00 00 1C AC 00 02 01 F5 00 62 00 00 3A B1
16:44:48 HTP: Main Menu
16:44:51 HTP: Configuration
16:44:56 WIF: Checking connection...
16:44:56 WIF: Connected
16:44:58 HTP: Configure Logging
16:45:02 HTP: Configuration
16:45:03 HTP: Main Menu
16:45:04 HTP: Console
16:45:15 DMP: 01 04 14 08 8E 08 EE 00 00 13 33 00 00 1C B0 00 02 01 F6 00 62 00 00 71 5F
16:45:16 DMP: 01 04 14 08 8E 08 EE 00 00 13 33 00 00 1C B0 00 02 01 F6 00 62 00 00 71 5F
16:45:16 WIF: Checking connection...
16:45:16 WIF: Connected
16:45:17 DMP: 01 04 14 08 91 09 13 00 00 13 89 00 00 1C B0 00 02 01 F6 00 62 00 00 C5 EE
16:45:18 DMP: 01 04 14 08 8E 09 4E 00 00 13 FC 00 00 1C B0 00 02 01 F6 00 62 00 00 4D 8F
16:45:19 DMP: 01 04 14 08 90 09 1C 00 00 13 AC 00 00 1C B0 00 02 01 F5 00 63 00 00 AB 44
16:45:20 DMP: 01 04 14 08 8F 09 11 00 00 13 81 00 00 1C B1 00 02 01 F6 00 62 00 00 D8 7C
16:45:27 DMP: 01 04 14 08 78 09 0A 00 00 13 45 00 00 1C B1 00 02 01 F6 00 62 00 00 6B 6F
16:45:28 DMP: 01 04 14 08 56 08 ED 00 00 12 B0 00 00 1C B2 00 02 01 F6 00 62 00 00 78 BF
16:45:29 DMP: 01 04 14 08 56 08 ED 00 00 12 B0 00 00 1C B2 00 02 01 F6 00 62 00 00 78 BF
16:45:30 DMP: 01 04 14 08 7B 09 29 00 00 13 7F 00 00 1C B2 00 02 01 F6 00 62 00 00 8B 1B
16:45:31 DMP: 01 04 14 08 85 09 19 00 00 13 88 00 00 1C B2 00 02 01 F6 00 62 00 00 CD 5E
16:45:32 DMP: 01 04 14 08 83 09 5D 00 00 14 14 00 00 1C B2 00 02 01 F6 00 62 00 00 E0 99
16:45:36 WIF: Checking connection...
16:45:36 WIF: Connected
[... 410 lines cut ...]
16:56:53 DMP: 01 04 14 08 84 09 05 00 00 13 4C 00 00 1D 23 00 02 01 F6 00 62 00 00 C7 B2
16:56:56 WIF: Checking connection...
16:56:56 WIF: Connected
16:57:00 DMP: 01 04 14 08 82 08 F2 00 00 13 1D 00 00 1D 24 00 02 01 F6 00 62 00 00 0C 25
16:57:01 DMP: 01 04 14 08 7D 09 23 00 00 13 86 00 00 1D 24 00 02 01 F6 00 62 00 00 AE 6A
[... cut ...]
16:57:19 DMP: 01 04 14 08 64 09 0D 00 00 13 18 00 00 1D 27 00 02 01 F6 00 62 00 00 0B B2
16:57:20 DMP: 01 04 14 08 62 08 EE 00 00 12 CC 00 00 1D 27 00 02 01 F5 00 62 00 00 1E 14
16:57:21 DMP: 01 04 14 08 6A 09 08 00 00 13 1B 00 00 1D 27 00 02 01 F5 00 62 00 00 33 C9
16:57:22 DMP: 01 04 14 08 61 09 0A 00 00 13 0E 00 00 1D 27 00 02 01 F5 00 62 00 00 06 19
16:57:23 DMP: 01 04 14 08 61 09 0A 00 00 13 0E 00 00 1D 27 00 02 01 F5 00 62 00 00 06 19
16:57:30 DMP: 01 04 14 08 5D 09 08 00 00 13 02 00 00 1D 28 00 02 01 F5 00 62 00 00 AB 10
```

arendst, 2019-07-10:
From your 6.5 log you see you miss data too resulting in the error you encounter in 6.6

Just checked my PZEM016:
```
11:52:35 DMP: 01 04 14 08 C5 00 6C 00 00 00 F2 00 00 18 3D 00 00 01 F4 00 64 00 00 53 8E
11:52:36 DMP: 01 04 14 08 C5 00 6C 00 00 00 F2 00 00 18 3D 00 00 01 F4 00 64 00 00 53 8E
11:52:37 DMP: 01 04 14 08 C5 00 6C 00 00 00 F2 00 00 18 3D 00 00 01 F4 00 64 00 00 53 8E
11:52:38 DMP: 01 04 14 08 C3 00 6C 00 00 00 F2 00 00 18 3D 00 00 01 F4 00 64 00 00 DB 06
11:52:39 DMP: 01 04 14 08 C4 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 63 00 00 DA 70
11:52:40 MQT: tele/rf1/STATE = {"Time":"2019-07-10T11:52:40","Uptime":"0T00:00:17","Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:52:40 MQT: tele/rf1/SENSOR = {"Time":"2019-07-10T11:52:40","ENERGY":{"TotalStartTime":"2018-11-23T15:33:47","Total":0.009,"Yesterday":0.000,"Today":0.009,"Period":0.00,"Power":24.10,"ApparentPower":24.24,"ReactivePower":0.00,"Factor":0.99,"Frequency":50.0,"Voltage":224.40,"Current":0.108}}
11:52:40 DMP: 01 04 14 08 C4 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 63 00 00 DA 70
11:52:41 DMP: 01 04 14 08 C7 00 6C 00 00 00 F2 00 00 18 3D 00 00 01 F4 00 64 00 00 2A 36
11:52:42 DMP: 01 04 14 08 C5 00 6C 00 00 00 F2 00 00 18 3D 00 00 01 F4 00 64 00 00 53 8E
11:52:43 DMP: 01 04 14 08 C4 00 6C 00 00 00 F2 00 00 18 3D 00 00 01 F4 00 64 00 00 6E 72
11:52:44 DMP: 01 04 14 08 C1 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 64 00 00 A7 7D
11:52:45 DMP: 01 04 14 08 C1 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 64 00 00 A7 7D
11:52:46 DMP: 01 04 14 08 C2 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 64 00 00 E3 39
11:52:47 DMP: 01 04 14 08 C2 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 64 00 00 E3 39
11:52:48 DMP: 01 04 14 08 C3 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 63 00 00 6F 04
11:52:49 DMP: 01 04 14 08 C3 00 6C 00 00 00 F1 00 00 18 3D 00 00 01 F4 00 63 00 00 6F 04
```
No missed data while configured this way:
[image]

and using a RS485 converter like below
[image]

I guess you will have to consider your serial connection between PZEM and ESP8266.

ninhvanthang, 2019-07-10:
I'm using pzem-004t v3, and follow this image to connect it to sonoff,
[image]

that module show tx,rx, not A,B like on pzem016,
does i need rs 485 converter?

arendst, 2019-07-10:
Ah, it seems the 004T V3 does already contain a RS232 to RS485 converter.

In that case I think you are blowing the serial interface of the esp8266 using 5V. You might want to try the fix as used for the old PZEM004 to make it's interface work on 3V3 inline with the ESP8266 serial interface protocol.

ninhvanthang, 2019-07-10:
Thank @arendst , finally it's worked,

problem is i used 5V for serial,
this new version is different as v1, they swapped position of 2 optocouplers,
i soldered 1K resistor as photo and it's work

[image]

closed 2019-07-10 by ninhvanthang

sheminasalam, 2020-09-17:
> Thank @arendst , finally it's worked,
> 
> problem is i used 5V for serial,
> this new version is different as v1, they swapped position of 2 optocouplers,
> i soldered 1K resistor as photo and it's work
> 
> [image]

Is this setup still working, i mean changing the 1k resistor optocoupler locations.? I am also having this problem with one of my pzem.  Can you confirm that the device is working in 3v serial with this setup?

ninhvanthang, 2020-09-17:
> Is this setup still working, i mean changing the 1k resistor optocoupler locations.? I am also having this problem with one of my pzem. Can you confirm that the device is working in 3v serial with [...]

yes, i's working in 3V of sonoff basic after more than one year

sheminasalam, 2020-09-17:
> yes, i's working in 3V of sonoff basic after more than one year

 I have tried this setup now but found that the device was not transmitting. when i disconnected the resistor, it was working fine.

hacksics, 2021-01-28:
Just FYI if ENERGY_WATCHDOG is set to higher value, then tasmota stops reading from the pzem (it goes to some kind of the dead lock) - I need to investigate this more.

I have had this problem in 9.2.0 and setting the ENERGY_WATCHDOG = 60. Sometimes 4 is not enough as it will show 0 voltage time to time. 

To check if my PZEM was the problem, I checked the DC voltage coming to PZEM and it's 5.1V. I've plugged in the PZEM to windows machine and used the Peacefair application to check and it was working fine without voltage dropouts. So my guess is something causing tasmota not to send the PzemAcModbus->Send() command. 

I will investigate more.

Amruth-Smaragdine, 2025-02-18:
Awesome Information. After I reviewed the content in this post and many articles, I successfully implemented the sensor using ESP32, and PZEM004TV3. I have written an article in two parts. Here are the links:

https://www.instructables.com/member/AmruthKumarSala/instructables/
I also have a few YouTube Videos in this playlist: https://www.youtube.com/watch?v=3L-hY_tTx24&list=PL0eiemFx0ClTO9qRJ8L540a3W-12NzTdY
Comments & constructive criticism to improve the content are appreciated.