arendst/Tasmota #6053: TLS Support for Hue Emulation?

smitty124, 2019-07-10:
I know there are only a handful of people who use tasmota with smarthings.  One of the essential components for this to work is the hue emulation ability, for more complex automations.   Smartthings recently updated their firmware on June 27th, 2019, and set their Hue protocol handler to use TLS over port 443 for the v2 Hue bridges.   For some reason, this caused HA hue emulation, as well as Tasmota Hue Emulation to stop working.   While the issue with the HA bridge is (According to ST staff) that it identifies as  a v2 hub, which does not support TLS,  I wonder if the Tasmota hue emulation is doing the same thing, as all my tasmota devices (various) using hue emulation no longer show up in ST.   ST staff have been manually been changing from port 443 to 80 for some HA users.  For Tasmota users, is there a way to activate TLS for hue emulation in the console, or potentially add an option to do so in future releases? 
  
(For more information about the change issue, although not directly tasmota related.)
https://community.smartthings.com/t/hue-integration-stopped-working-27-june-2019/166962/64
https://github.com/bwssytems/ha-bridge/issues/1107

n17ikh, 2019-07-10:
Ah, this would explain why my Tasmota devices quit working with Smartthings for me while I was on vacation. I've been poking around with old versions of Tasmota thinking it was something in 6.5.0. I was hoping the changes in #5849 would help, but had no luck with a development version of 6.6.0. I tried the official release of 6.6.0 this evening with no success there either. I was about to open an issue of my own.

Smartthings does print a message saying "please press the button on the hue bridge" but doesn't progress any further.

Looking at some packet captures, the ST hub performs a GET on /description.xml, Tasmota responds with a 200 OK and then its text/xml response. The ST hub then closes the connection and never tries to do anything else. Repeat about every 5 seconds while on the "add a thing" screen in Smartthings.

s-hadinger, 2019-07-10:
This is not good news. Tasmota announces `Philips hue bridge 2012` which should be recognized as V1.

Implementing a TLS server on 8266 is a challenge. TLS eats up a lot of flash and memory, and calculating the TLS handshake on the server side takes a lot of compute (understand several seconds).

Alos, if you need a valid server certificate, it's a burden. For ex Letsencrypt needs to be re-issued every 3 months.

Wouldn't it be easier to have a local proxy an a Raspberry PI listening on 443?

Jason2866, 2019-07-10:
@s-hadinger  explained why there will be no TLS support as a standard in Tasmota.
So the answer to your question is: There will be no TLS support for Hue Emulation (in precompiled binaries)

closed 2019-07-10 by ascillato2