arendst/Tasmota #6054: TLS no longer works

avrhack, 2019-07-10:
I have been running v6.4.1 for some time, using the Arduino TLS library to encrypt the MQTT sessions. 

In general it's pretty reliable with only the occasional (every day or two) reconnection.

However v6.5.0 is terrible - drops out several times an hour but strangely this is also dependent on how far the device is from the WiFi AP; for example a device right next to it with strong signal has many fewer drop outs. NOTE THAT THE EXACT SAME DEVICE / DISTANCE was fine in 6.4.1.

And v6.6.0 makes things even worse. No TLS working at all with that one. I've enabled BearSSL debugging and the error is "BSSL:_wait_for_handshake: failed"; looking at their forum and bug reports it seems the only answer is to use a 160MHz esp! WTF?!

I appreciate Theo you may say this is third party library issue, but I would say otherwise. This was working in v6.4.1 but it seems impossible to use later versions of Tasmota with the versions of third party libraries that actually work, specifically that TLS is no longer supported with SDK 2.3.0 which frankly seems far better than any of the junk that's been produced since.

Sorry to be blunt but this is incredibly annoying - things that work are getting broken, and the ability to use the libraries that use to work are being closed off. The only solution left will be to move to another firmware than Tasmota and I really don't want to go that route.

Any comments anyone?

### REQUESTED INFORMATION

Occurs on a Sonoff S20, Blitzwolf BHP-4 and NodeMCU v2

Fails with self-built firmware v6.5.0 and v6.6.0 with 2.4.2 and 2.5.2, works with self-built firmware v6.4.1 with same defines.

Cannot provide output of status 0 as cannot connect to MQTT.

### TO REPRODUCE

Any MQTT connection over TLS will fail with v6.5.0 or v.6.6.0

### EXPECTED BEHAVIOUR

A working MQTT connection, as used to work with v6.4.1

ascillato, 2019-07-10:
Hi,

Please, complete the troubleshooting template and provide logs in order to check your issue.

Complaints in a free software is totally useless since there is nothing to work on. Please, just provide the technical information so as to solve your issue.

avrhack, 2019-07-10:
This isn't a complaint. I followed the template and provided the serial log output, specifically enabling the appropriate additional SSL options. What more "technical information" are you expecting?

ascillato, 2019-07-10:
Sorry, could you please complete the troubleshooting template with status 0, console output, mqtt broker logs, etc etc. (you can access Tasmota logs from the serial interface) remember to use seriallog 4 for more debug information

ascillato2, 2019-07-10:
From my experience in this issues, no one in the community want to read a walltext of complaints.

Please, let's start over with a fresh new issue with the troubleshooting template completed. Keep it just technical. Logs, steps you did, etc.

closed 2019-07-10 by ascillato2