arendst/Tasmota #6056: New kinetic tuya device - Model ERC309

h4nc, 2019-07-10:
I'd like to add the following Tuya device to tasmota. It's a device that is also control-able with a kinetic switch which doen't need a battery.

https://www.amazon.co.uk/Thinkbee-Wireless-Required-Self-powered-Controlled/dp/B06VY9KWDD

I already flashed the device. I had to desolder the ESP Board to be able to flash it, else it wasn't possible. See picture:

[image]

When I select the device tuya dimmer I'm able to control it with the first of the 6 switches (seems like there is a currently an [issue](https://github.com/arendst/Sonoff-Tasmota/issues/6039) with that.)
The device is not dimmable, so this will not be the right setting anyway.

So when I click that first switch the relay turns on (the led to), but it turns off after some seconds. When I click the phyiscal button it also gets reportet in tasmota. 

I already tried using "generic modul" and setting TUYA RX and TUYA TX but no button shows up to control it on the main page. 

What am I missing? How can I make it work? Thanks

ascillato2, 2019-07-10:
Hi,

Please try latest Tasmota from the dev branch (http://thehackbox.org/tasmota/020300/sonoff.bin)

h4nc, 2019-07-10:
I tried 6.6.0.1. unless something changed in the last 3-4 hours it doesn’t work

ascillato, 2019-07-10:
Can you post the output of the console when your relay turn off ?
Remember to use weblog set to 4 for more debug information

h4nc, 2019-07-10:
Yes of course, here it is:

Few seconds after I pressed the switch it automatically turned off again.

```
19:59:57 CMD: WebLog 4
19:59:57 RSL: stat/sonoff/RESULT = {"WebLog":4}
19:59:58 CFG: in Flash gespeichert am F4, zählen 176, Bytes 3584
19:59:59 TYA: Send "55aa00000000ff"
19:59:59 TYA: RX Packet: "55aa010000010102"
19:59:59 TYA: Heartbeat
20:00:02 TYA: RX Packet: "55aa01070005010100010110"
20:00:02 TYA: RX Device-1 --> MCU State: On Current State:Off
20:00:02 SRC: Switch
20:00:02 TYA: Send dim value=26 (id=103)
20:00:02 TYA: Send "55aa00060008670200040000001a94"
20:00:02 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
20:00:02 RSL: stat/sonoff/POWER = ON
20:00:02 TYA: RX Packet: "55aa01070008670200040000001a96"
20:00:02 TYA: RX Dim State=26
20:00:03 CFG: in Flash gespeichert am FB, zählen 177, Bytes 3584
20:00:09 TYA: RX Packet: "55aa0107000867020004000000128e"
20:00:09 TYA: RX Dim State=18
20:00:09 SRC: Switch
20:00:09 RSL: empfangenes topic /Dimmer, Datengröße 1, Daten 7
20:00:09 RSL: Gruppe 0, Index 1, Befehl DIMMER, Daten 7
20:00:09 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":7}
20:00:09 TYA: Send dim skipped value=18
20:00:10 CFG: in Flash gespeichert am FA, zählen 178, Bytes 3584
20:00:10 TYA: Send "55aa00000000ff"
20:00:10 TYA: RX Packet: "55aa010000010102"
20:00:10 TYA: Heartbeat
20:00:12 WIF: Prüfe Verbindung...
20:00:12 WIF: verbunden
20:00:21 TYA: Send "55aa00000000ff"
20:00:21 TYA: RX Packet: "55aa010000010102"
20:00:21 TYA: Heartbeat
20:00:28 TYA: RX Packet: "55aa0107000501010001000f"
20:00:28 TYA: RX Device-1 --> MCU State: Off Current State:On
20:00:28 SRC: Switch
20:00:28 TYA: Send dim skipped value=0
20:00:28 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
20:00:28 RSL: stat/sonoff/POWER = OFF
20:00:29 CFG: in Flash gespeichert am F9, zählen 179, Bytes 3584
20:00:29 TYA: RX Packet: "55aa0107000867020004000000007c"
20:00:29 TYA: RX Dim State=0
20:00:32 WIF: Prüfe Verbindung...
20:00:32 WIF: verbunden
20:00:32 TYA: Send "55aa00000000ff"
20:00:32 TYA: RX Packet: "55aa010000010102"
20:00:32 TYA: Heartbeat
```

Jason2866, 2019-07-11:
@chaosmaster  or @thirug010 can you take a look into?

h4nc, 2019-07-11:
After turning the Relay on from the tasmotaUI it takes arround 26 senconds before it automatically turns off again.

This also happens when I turn it on with the hardware button. 

And of course there is the Dimmer Slider which isn't needed for that relay.

thirug010, 2019-07-11:
@h4nc,

can you provide the link for the device and Debug log with Level 4 

Thanks 
Thiru

chaosmaster, 2019-07-11:
It looks like it's receiving an off-command from the MCU:
```
20:00:28 TYA: RX Packet: "55aa0107000501010001000f"
20:00:28 TYA: RX Device-1 --> MCU State: Off Current State:On
```

BTW, for flashing the device you probably don't need to desolder. it should be enough to keep the MCU in reset while flashing.
Can you take a picture of the MCU?

thirug010, 2019-07-11:
I will check and update you, BTW use setoption65 --> 1 ti disable the dimmer and try gain 

Thanks 
Thiru

h4nc, 2019-07-11:
> Can you take a picture of the MCU? Is that ok?

[image]
[image]
[image]

After setting setoption65 1 it works. No slider AND no turning back to OFF.

That's awesome!

How to spread that information for other people?

1) Set it as Tuya Dimmer
2) Console: `setoption65 1`

chaosmaster, 2019-07-11:
Can you try to take a better picture of the markings on the square chip on your first picture please?

ascillato2, 2019-07-11:
> After setting setoption65 1 it works.

Great to know that now it is working for you. Thanks. Closing this issue.

closed 2019-07-11 by ascillato2

h4nc, 2019-07-11:
@chaosmaster do you still want that picture.

chaosmaster, 2019-07-11:
@h4nc yes, that would be great, since it could help identifying the MCUs reset signal.
Having that most likely allows flashing this device without desoldering the ESP.

h4nc, 2019-07-11:
This one should be better.
[image]

chaosmaster, 2019-07-11:
Thanks, that looks good.
Can't really find anything for the left chip.
The right one seems to be this one: https://www.renesas.com/us/en/doc/products/mpumcu/doc/rl78/r01ds0193ej0220_rl78g12.pdf
So looks like that is the MCU.
According to the datasheet pin 5 is the reset line, so connecting it to ground while you try to flash the ESP should let you flash without desoldering.
Would be great if you could test and document in the wiki.

[image]

h4nc, 2019-07-11:
I will test this tomorrow. I want to flash more of those and would be great if it was possible without desoldering.

Thanks.

h4nc, 2019-07-12:
I was able to flash the device without desoldering by connection the Reset pin to ground, as you mentioned. 

That's great. However it will be hard for people that don't solder that often (I do) and don't have the right tools, because the Reset pin is tiny and also it's a little hard to connect the pins of the ESP board without desoldering it (you will need a small soldering iron).

So probably not the easiest device to be flashed.

chaosmaster, 2019-07-12:
Great to hear that it worked.
It still is good to have the option I guess.
Would you mind documenting both options in the wiki, as well as the configuration "SetOption65 1"?

h4nc, 2019-07-12:
Can you lead me to the right section for this?

Also I don't know how to call that device.

chaosmaster, 2019-07-12:
The general Tuya Dimmer wiki is [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer).
Allthough your device isn't a Dimmer.
@thirug010 Is there a already a wiki-section for non-dimmer tuya devices?

If the device doesn't have a proper name or model then describe it and add pictures and a link where it can be bought.

Thank you

lalo-uy, 2019-07-12:
The other chip is Amiccom A7129 RF tranceiver
http://amiccom.weebly.com/uploads/3/9/5/9/3959395/a7129_datasheet_v0.5_preliminary.pdf

h4nc, 2019-07-12:
There are no traces pointing away from
The reset pin so I’m afraid it’s not broken out.

chaosmaster, 2019-07-12:
@lalo-uy interesting
@h4nc funny, it should at least be pulled high with a resistor, a floating reset pin could lead to spontaneous resets of the MCU.
Are you sure there aren't any traces below the MCU or going through a via to the other side of the board?

h4nc, 2019-07-12:
> Are you sure there aren't any traces below the MCU or going through a via to the other side of the board?

I’m quite sure about the other side, but yeah could (probably will) be under the MCU if you say that’s important.

h4nc, 2019-07-12:
Link to the device:

https://s.click.aliexpress.com/e/54JDwLm

lalo-uy, 2019-07-12:
The pullup may be inside the chip.

chaosmaster, 2019-07-12:
@lalo-uy, seems like you're right, didn't think of that on the reset pin:
[image]

h4nc, 2019-07-12:
👌 so I don’t have to search any more 

I will but I still don’t know where

chaosmaster, 2019-07-12:
@h4nc just poke around with a multimeter int continuity mode :joy:

h4nc, 2019-07-12:
Picture of the device:
[image]

h4nc, 2019-07-12:
Ok so I updated the wiki, hope it's ok like this.

Feel free to add additional information.

https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer-and-Relay

chaosmaster, 2019-07-12:
Looks good, thank you.
I have also added the product link you provided.

meingraham, 2019-07-12:
@h4nc @chaosmaster 

Should that device not have its own wiki entry (i.e., a separate page)?

chaosmaster, 2019-07-12:
@meingraham I think the "Tuya Dimmer" section is fine, since the Tasmota-Module that needs to be configured is called "Tuya Dimmer" as well.
Might make sense to rename the module (and wiki-page) now that it also supports non-dimmer devices.

meingraham, 2019-07-12:
I think it would be better to have a page for each device and then in the configuration information it selects the same module. Likewise, the Template repository can indicate the same template spec. This makes searching for the device easier for users to find. Also, the filtering on the template repository will find the dimmer and not include the switch.

chaosmaster, 2019-07-12:
I don't think that is supported with the way templating is currently implemented.
What differentiates this device (and other non-dimmer tuya-devices from a tuya-dimmer are configuration options defined via SetOptionX, AFAIK these cannot be predefined in a template.

meingraham, 2019-07-12:
Is this Module (BASE) 54? I couldn't find it in the module list.

If so, then a template like this? https://blakadder.github.io/templates/moes_DS01.html

chaosmaster, 2019-07-12:
Yes base 54 is correct, but you cannot define a non-dimmer tuya device like this (you can, but it won't work correctly), since it's not the GPIO-configuration that is different but the SetOption parameters.

meingraham, 2019-07-12:
You can create a template for each device. Base it on 54, set the necessary GPIO and in the guide, specify the necessary SetOptions. Lots of other devices with Templates, Rules, SetOptions. This makes each entry according to device, model, function, etc. Better for classification.

chaosmaster, 2019-07-12:
Feel free to do that, I don't see much of a point defining a template that doesn't change any of the GPIOs.

meingraham, 2019-07-12:
OK. Just better for the filter mechanism on the templates site.

meingraham, 2019-07-12:
I'll separate the pages - one for the dimmer and one for the switch based on the 54 BASE.

meingraham, 2019-07-13:
Templates Repository - Awaiting PR merge for addition of flashing guides (i.e., information in Tasmota GitHub wiki pages linked below)
- https://blakadder.github.io/templates/moes_DS01.html
- https://blakadder.github.io/templates/ERC309.html

Wiki
- https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer
- https://github.com/arendst/Sonoff-Tasmota/wiki/ERC309-Kinetic-Switch

chaosmaster, 2019-07-14:
@meingraham your descriptions don't seem quite right:

>     GPIO13 - Tuya Rx (108)
>     GPIO15 - Tuya Tx (107)

Tuya devices typically use GPIO01 and GPIO03 for TX/RX GPIO13 and GPIO15 are only used on some tuya devices.

Also 

> SetOption34 - Set the Tuya Dimmer device ID
> 
> If power on/off works, but dimming doesn’t, you probably haven’t set the correct ID

This doesn't make any sense for the Kinetic switch as it isn't a dimmer (I thought that was the whole point of separating them)

> Setoption65 1 - Enable slider control

This is also wrong, as _SetOption65 1_ actually **disables** slider control (Yes, it's a bit counter-intuitive, but is that way for backwards-compatibility)

meingraham, 2019-07-15:
@chaosmaster 

GPIO information - I took this information off of the photo in the Tuya Dimmer article - https://user-images.githubusercontent.com/34905120/47967431-4a537880-e02b-11e8-955b-ee9843bd222d.jpg

Also, the Template for the Moes DS01 used the same GPIO.

Yes, it's not a dimmer. Once `54` is selected for this device, does it need to set a Tuya "device ID" at all? Or does selecting the module suffice?

Regarding SetOption65 - From the wiki

Command		 | Parameters
:---		 | :---
SetOption65|`0` = disable dimmer slider control<BR>`1` = enable dimmer slider control *(default)*

Does the wiki need to be corrected?

chaosmaster, 2019-07-15:
@meingraham 
The screenshot was under the section for the section for the "Charging Essentials" Dimmer as can be seen here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer/d6b0e6eb1fb644713b0a322c8588db527734ae7a#charging-essentials-from-costco.
AFAIK that's the only tuya-device that uses this GPIO-configuration I am not sure however @thirug010 @digiblur ?

I'm pretty confident the Oittm/Moes uses GPIO01/GPIO03, so the template for Moes is probably also incorrect.

Setting device ID shouldn't be necessary for a non-dimmer device.
And even for a dimmer-device it should be detected automatically.

Yes I think the documentation about SetOption65 is wrong @thirug010 ?

digiblur, 2019-07-15:
The new Moes dimmers use the same GPIO 13 and 15 pins.

meingraham, 2019-07-15:
OK, I'll remove the Dimmer ID for the ERC309. I can clarify the GPIO assignment once we pin this down. The CE & Moes use the GPIO as currently shown in the article. The Template for the Moes is also OK as is. Let me know what the GPIO for the other Dimmers are and also what the GPIO for the ERC309 are.

I'll wait to make these flashing guide changes all at once so that @blakadder only has one PR to merge for each template (Moes & ERC309).

@thirug010 - Let me know about the `SetOption65` change in the Commands wiki article.

chaosmaster, 2019-07-15:
Most of them use the default hardware-serial on GPIO1/GPIO3 but that should be selected as fallback anyway if none are defined at all.
The old Moes used GPIO1/3 as well if i understood @digiblur correctly.

meingraham, 2019-07-15:
Since users "today" will likely get the newer models of the CE & Moes, they'll likely get the 13/15 version. I'd recommend staying with it as is.

I can update the article to say these dimmers use "hardware serial GPIO 01/03 or software serial, typically on GPIO 13/15."

Does the ERC309 use 1/3 or 13/15?

thirug010, 2019-07-15:
> @chaosmaster
> 
> GPIO information - I took this information off of the photo in the Tuya Dimmer article - https://user-images.githubusercontent.com/34905120/47967431-4a537880-e02b-11e8-955b-ee9843bd222d.jpg
> 
> Also, the Template for the Moes DS01 used the same GPIO.
> 
> Yes, it's not a dimmer. Once `54` is selected for this device, does it need to set a Tuya "device ID" at all? Or does selecting the module suffice?
> 
> Regarding SetOption65 - From the wiki
> 
> Command	Parameters
> SetOption65	`0` = disable dimmer slider control
> `1` = enable dimmer slider control _(default)_
> Does the wiki need to be corrected?

Please update SetOption66 also, as some Tuya dimmer has the dimmer range from 1- 100 and 1 -255 

Command | Parameters
-- | --
SetOption66 | `0` = dimmer Range 1 - 100 _(default)_   and   `1` = dimmer Range 1 -255

chaosmaster, 2019-07-15:
> Since users "today" will likely get the newer models of the CE & Moes, they'll likely get the 13/15 version. I'd recommend staying with it as is.
>
> I can update the article to say these dimmers use [...]

GPIO 13/15 also uses hardware-serial and default should be GPIO 1/3 (since that is also the fallback if nothing is selected)

meingraham, 2019-07-15:
In Tasmota, GPIO 01/03 is implemented as hardware serial. Tx/Rx on other pins is implemented as serial protocol emulation (i.e., software based). That's my understanding, at least. @arendst? Also, for software serial, I believe this requires a Core >= 2.4.2 due to RAM requirements if software serial.

@thirug010 - is `SetOption65` documented properly in the wiki?

I added `SetOption66` to the wiki

chaosmaster, 2019-07-15:
13/15 also support hardware-serial alternatively to 1/3 any other pins use software-serial (allthough gpio2 could actually be used for hardware-serial TX only)
Hardware-support for 13/15 was introduced in commit #d6983d268b9a0c173d1e54a0e26d69c601d01dbe

meingraham, 2019-07-15:
OK then "These dimmers typically use hardware serial either on GPIO01 and GPIO03, or GPIO13 and GPIO15. Different devices/models vary in which GPIO pair is used."

Which dimmers require `SetOption66 1`?

h4nc, 2019-07-18:
Hi again,

I'd like to add a temperature sensor (DS18B20) to that device. Is that possible? If yes witch pin shoudl I take?

Jason2866, 2019-07-18:
@h4nc I see no transformer (from a switching power supply). So you can have AC mains on GPIO pins!
So dont connect any kind of sensors. It is very dangerous. You will be electrocuted!

h4nc, 2019-07-18:
I did try it in the meantime. I used GPIO14 for it and the measurement works.

After reading your comment I measured the pins I broke out and I did not measure AC mains. 

So it should be fine right?

Zorgino, 2019-08-08:
Did anybody think about making custom use of the kinetic switch buttons?
As far as I understood, MCU which is in charge of radio lives separately from main TYWE1S and just sends signals to it to change the light state. 
Also, user manual says that it's possible to form a mesh out of receivers, where the signal from switch will be transferred to the target it's bound to. 
Any ideas on how to listen to all of the signals in the broadcast?

[2 more replies]