arendst/Tasmota #6058: Tuya Light Switch - LED Blinking at "LED1" and controlable with "Relay2"

h4nc, 2019-07-10:
I got a new Tuya Light switch. I was able to flash it. 

I already found the right pins for the Button, Relay and the LED.

Generic Module
GPIO 3 goes to Button1(17)
GPIO13 goes to Relay1(21)

And GPIO14 is connected to a LED. But I don't know yet how to set it correctly.

I want:
LED off = Relay on
LED on = Relay off

This should be possible when I set GPIO14 to LED1i, but it doesn't matter which LED setting I use the LED blinks all of the time , when it's not connected to mqtt. And when mqtt is set the LED is turned on all of the time. 
I also tried LEDlink.

When I set GPIO14 to Relay2 it's possible to control this LED, but ON = LEDoff and OFF = LEDon. This tells me that GPIO14 is the right pin for this blue LED. 

I don't get whats the issue here?

[image]
[image]

ascillato, 2019-07-10:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-07-10:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-10 by ascillato2

whc2001, 2019-09-20:
Led1 is the status LED in default settings (only one LED is set). When there is an error it will blink constantly. Use [SetOption31 1](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#setoption-overview) to disable it. As for the inverted indication, simply choose Led1i, since "i" means invert.