arendst/Tasmota #6064: DS1818x20 sensor not working after update to 660

mvveelen, 2019-07-10:
Been using a Sonoff TH16 with a DS18x20 temperature sensor for a few years now and after updating Tasmota from 650 to 660 I don't get any readings. After downgrading to 650 it's working again. I don't see anything that might be wrong with the configuration and didn't change anything too.

Couldn't find anything about having to change something or other users with the same issue.

GPIO14 sensor: DS18x20(4)

Jason2866, 2019-07-10:
Please fill out complete issue template. Provide `status 0` output

ascillato, 2019-07-10:
Have you updated to Sonoff-sensors.bin?

mvveelen, 2019-07-10:
(Trying to get the output)

No, I updated to sonoff-minimal (OTA) and after that to sonoff.bin (OTA) like I always do.

ascillato2, 2019-07-10:
Please, update to sonoff-sensors.bin

mvveelen, 2019-07-10:
Shoot me. After getting the output from 650 and updating to sonoff-minimal.bin and sonoff.bin (660) to get the output from that version, the sensor suddenly works....?

Case closed and I'm feeling ashamed.....

ascillato2, 2019-07-10:
Great to know that now it is working for you :+1:

Please, next time use the troubleshooting template. It gaves us a lot of information so as to properly help you.

Also remember that you can find us also in the Tasmota Support Chat.

closed 2019-07-10 by ascillato2