arendst/Tasmota #6066: Tuya Dimmer Dimmer Value not equal to Rx Dim State

Hollako, 2019-07-10:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
I recently bought and Tuya dimmer US type and flash Tasmota Firmware. Everything goes smoothly and i started to test it then i noticed that the commands sent from the WEB or MQTT are not "accepted" but the dimmer MCU of the Tuya switch. 
After some troubleshoots i did the Dim UP and Dim Down commands from the keypad to see what are the commands received from the ESP i noticed that the RX Dim State is correct and the Dimmer Value in the stat/Result is wrong and not synchronized. And when i send the commands to the MCU, it will reply by another value or it doesn't respond at all.

Noting that the ON/OFF commands are working properly. And i am using the 6.5.0 Version.

HERE IS BELOW THE CONSOLE OUTPUT HERE, And 

Garden_Dim1/stat/RESULT = {"POWER1":"ON","Dimmer":_**28**_}
22:33:40 TYA: RX Packet: "55aa03070008020200040000004861"
22:33:40 TYA: RX Dim State=**_72_**
MQT: Garden_Dim1/stat/RESULT = {"POWER1":"ON","Dimmer":**_20_**}
22:44:38 TYA: RX Packet: "55aa0307000802020004000000334c"
22:44:38 TYA: RX Dim State=**_51_**

### EXPECTED BEHAVIOUR
RX Dim State = to Dimmer value

Thank you in advance, and i am ready fir any further troubleshoot when advised.

Regards,

thirug010, 2019-07-11:
@Hollako, 
I will check and update you on this

thirug010, 2019-07-12:
PR submitted for this issue

Hollako, 2019-07-12:
Hello thirug010,

Firs of all, thank you for the support.

And is there something to do or test from my side? 

regards,

thirug010, 2019-07-12:
@Hollako,

Please take the code from the cloned branch : https://github.com/thirug010/Sonoff-Tasmota
[default setoption66-->0 will use dimmer range 1 - 100, setOption66--> 1 will use dimmer range 1- 255] 

Thanks
Thiru

closed 2019-07-12 by arendst

Hollako, 2019-07-13:
The link is down !!

ascillato, 2019-07-13:
@Hollako 

It has been fixed in Tasmota. Please, try latest version from dev branch.

Hollako, 2019-07-15:
Tested and Done !!

Thank you.

PR #6076 (2019-07-12): Updated Code for Issue #6066- Dimmer Value not equal to Rx Dim State
Change Summary

1. Added Option to select Dimmer Range (1-100 and 1-255) through SetOption66 (default is 1-100 for legacy device support

2. Updated Dimmer code to set range based on the option66 100 / 225

## Description:

**Related issue :** fixes  #6066 Tuya Dimmer Dimmer Value not equal to Rx Dim State

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0 [... cut]