arendst/Tasmota #6067: BH1750 Illuminance erratic values and creates rogue sensors

xbmcnut, 2019-07-11:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
I've just added a BH1750 sensor to a Sonoff S20 on serial pins 1 and 3 (SCL and SDA). While it reports sensor information, the lux level only fluctuates between three values; 640, 3200 and13440. It also creates six other sensors shown below with rogue values. Tasmota version is 6.6.0.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
03:05:13 MQT: stat/hall_lux/STATUS = {"Status":{"Module":0,"FriendlyName":["Hall_Lux"],"Topic":"hall_lux","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:05:13 MQT: stat/hall_lux/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:07","StartupUTC":"2019-07-11T02:03:06","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":62,"SaveAddress":"F6000"}}
03:05:13 MQT: stat/hall_lux/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
03:05:13 MQT: stat/hall_lux/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
03:05:13 MQT: stat/hall_lux/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":2,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
03:05:13 MQT: stat/hall_lux/STATUS5 = {"StatusNET":{"Hostname":"hall_lux-4795","IPAddress":"10.0.1.85","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.200","Mac":"5C:CF:7F:3B:52:BB","Webserver":2,"WifiConfig":4}}
03:05:13 MQT: stat/hall_lux/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B52BB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
03:05:13 MQT: stat/hall_lux/STATUS7 = {"StatusTIM":{"UTC":"Thu Jul 11 02:05:13 2019","Local":"Thu Jul 11 03:05:13 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:52"}}
03:05:13 MQT: stat/hall_lux/STATUS10 = {"StatusSNS":{"Time":"2019-07-11T03:05:13","BH1750":{"Illuminance":13440},"SHT3X-0x44":{"Temperature":-42.9,"Humidity":0.0},"SHT3X-0x45":{"Temperature":-40.2,"Humidity":0.0},"SHTC3-0x70":{"Temperature":-42.9,"Humidity":0.0},"TempUnit":"C"}}
03:05:13 MQT: stat/hall_lux/STATUS11 = {"StatusSTS":{"Time":"2019-07-11T03:05:13","Uptime":"0T00:02:07","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"FE:EC:DA:38:36:E9","Channel":11,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:11"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Lux sensor only

### SCREENSHOTS
[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-07-11:
You have seriallog enabled. Please, disable seriallog by typing in the Tasmota console: seriallog 0
Then reboot your device and test again.

xbmcnut, 2019-07-11:
Thanks. Tried that. Still the same.

```
03:52:00 CMD: seriallog 0
03:52:00 SRC: WebConsole from 10.0.1.63
03:52:00 RSL: Received Topic /seriallog, Data Size 1, Data 0
03:52:00 RSL: Group 0, Index 1, Command SERIALLOG, Data 0
03:52:00 MQT: stat/hall_lux/RESULT = {"SerialLog":"0 (Active 0)"}
03:52:08 MQT: tele/hall_lux/STATE = {"Time":"2019-07-11T03:52:08","Uptime":"0T00:01:59","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"7A:8A:20:DD:A8:E4","Channel":1,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:09"}}
03:52:08 MQT: tele/hall_lux/SENSOR = {"Time":"2019-07-11T03:52:08","BH1750":{"Illuminance":13440},"SHT3X-0x44":{"Temperature":-40.2,"Humidity":0.0},"SHT3X-0x45":{"Temperature":-40.2,"Humidity":0.0},"SHTC3-0x70":{"Temperature":-42.9,"Humidity":0.0},"TempUnit":"C"}
```

ascillato, 2019-07-11:
Can you post the output of the console when your device boots? Please set weblog to 4. Thanks

xbmcnut, 2019-07-11:
```
00:00:00 SRC: Restart
00:00:00 Project sonoff Hall_Lux Version 6.6.0(release-sonoff)-2_3_0
00:00:00 SHT: Sensor did not ACK command
00:00:00 I2C: BH1750 found at 0x23
00:00:00 I2C: SHT3X found at 0x44
00:00:00 I2C: SHT3X found at 0x45
00:00:00 I2C: SHTC3 found at 0x70
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 IoT in mode 11N as hall_lux-4795...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on hall_lux-4795 with IP address 10.0.1.85
00:00:05 HTP: Console
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/hall_lux/LWT = Online (retained)
00:00:06 MQT: cmnd/hall_lux/POWER = 
00:00:06 MQT: Subscribe to cmnd/hall_lux/#
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/DVES_3B52BB_fb/#
00:00:06 MQT: tele/hall_lux/INFO1 = {"Module":"Sonoff S20","Version":"6.6.0(release-sonoff)","FallbackTopic":"cmnd/DVES_3B52BB_fb/","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/hall_lux/INFO2 = {"WebServerMode":"Admin","Hostname":"hall_lux-4795","IPAddress":"10.0.1.85"}
00:00:06 MQT: tele/hall_lux/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/hall_lux/RESULT = {"POWER":"OFF"}
00:00:06 MQT: stat/hall_lux/POWER = OFF
03:58:00 NTP: Drift 0, (UTC) Thu Jul 11 02:58:00 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
03:58:02 APP: Boot Count 20
03:58:02 CFG: Saved to flash at FA, Count 66, Bytes 3584
03:58:07 MQT: tele/hall_lux/STATE = {"Time":"2019-07-11T03:58:07","Uptime":"0T00:00:15","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"7A:8A:20:DD:A8:E4","Channel":1,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:05"}}
03:58:07 MQT: tele/hall_lux/SENSOR = {"Time":"2019-07-11T03:58:07","BH1750":{"Illuminance":13440},"SHT3X-0x44":{"Temperature":-40.2,"Humidity":0.0},"SHT3X-0x45":{"Temperature":-40.2,"Humidity":0.0},"SHTC3-0x70":{"Temperature":-42.9,"Humidity":0.0},"TempUnit":"C"}
03:58:17 MQT: tele/hall_lux/STATE = {"Time":"2019-07-11T03:58:17","Uptime":"0T00:00:25","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"7A:8A:20:DD:A8:E4","Channel":1,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:05"}}
03:58:17 MQT: tele/hall_lux/SENSOR = {"Time":"2019-07-11T03:58:17","BH1750":{"Illuminance":13440},"SHT3X-0x44":{"Temperature":-40.2,"Humidity":0.0},"SHT3X-0x45":{"Temperature":-40.2,"Humidity":0.0},"SHTC3-0x70":{"Temperature":-42.9,"Humidity":0.0},"TempUnit":"C"}
03:58:17 WIF: Checking connection...
03:58:17 WIF: Connected
03:58:27 MQT: tele/hall_lux/STATE = {"Time":"2019-07-11T03:58:27","Uptime":"0T00:00:35","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"7A:8A:20:DD:A8:E4","Channel":1,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:05"}}
03:58:27 MQT: tele/hall_lux/SENSOR = {"Time":"2019-07-11T03:58:27","BH1750":{"Illuminance":13440},"SHT3X-0x44":{"Temperature":-40.2,"Humidity":0.0},"SHT3X-0x45":{"Temperature":-40.2,"Humidity":0.0},"SHTC3-0x70":{"Temperature":-34.7,"Humidity":0.0},"TempUnit":"C"}
03:58:37 MQT: tele/hall_lux/STATE = {"Time":"2019-07-11T03:58:37","Uptime":"0T00:00:45","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"7A:8A:20:DD:A8:E4","Channel":1,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:05"}}
03:58:37 MQT: tele/hall_lux/SENSOR = {"Time":"2019-07-11T03:58:37","BH1750":{"Illuminance":3200},"SHT3X-0x44":{"Temperature":-42.9,"Humidity":0.0},"SHT3X-0x45":{"Temperature":-40.2,"Humidity":0.0},"SHTC3-0x70":{"Temperature":-42.9,"Humidity":0.0},"TempUnit":"C"}
```

arendst, 2019-07-11:
Cannot reproduce
```
00:00:00 CFG: Loaded from flash at FB, Count 522
00:00:00 CFG: Saved to flash at FB, Count 523, Bytes 3584
00:00:00 SRC: Restart
00:00:00 Project sonoff Wemos7 v2 Version 6.6.0(release-sonoff)-2_3_0
00:00:00 SHT: Sensor did not ACK command
00:00:00 I2C: BH1750 found at 0x23
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId indebuurt1, Channel 1, BSSId 24:D3:F2:97:C0:A1, RSSI -58, Encryption 1
00:00:04 WIF: Network 1, AP-, SSId indebuurt2, Channel 5, BSSId A0:AB:1B:7D:42:AC, RSSI -81, Encryption 1
00:00:04 WIF: Network 2, AP2, SSId indebuurt3, Channel 12, BSSId 60:E3:27:58:77:E6, RSSI -72, Encryption 1
00:00:04 WIF: Patch issue 2186
00:00:04 WIF: Connecting to AP1 indebuurt1 in mode 11N as wemos7...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on wemos7 with IP address 192.168.2.225
00:00:09 MQT: Attempting connection...
00:00:09 MQT: Connected
00:00:09 MQT: tele/wemos7/LWT = Online (retained)
00:00:09 MQT: cmnd/wemos7/POWER = 
00:00:09 MQT: Subscribe to cmnd/wemos7/#
00:00:09 MQT: Subscribe to cmnd/sonoffs/#
00:00:09 MQT: Subscribe to cmnd/DVES_0802E9_fb/#
00:00:09 MQT: tele/wemos7/INFO1 = {"Module":"Generic","Version":"6.6.0(release-sonoff)","FallbackTopic":"cmnd/DVES_0802E9_fb/","GroupTopic":"sonoffs"}
00:00:09 MQT: tele/wemos7/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos7","IPAddress":"192.168.2.225"}
00:00:09 MQT: tele/wemos7/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: stat/wemos7/RESULT = {"POWER":"OFF"}
00:00:09 MQT: stat/wemos7/POWER = OFF
10:36:22 NTP: Drift 0, (UTC) Thu Jul 11 08:36:22 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
10:36:22 APP: Boot Count 219
10:36:22 CFG: Saved to flash at FA, Count 524, Bytes 3584
10:36:23 HTP: Main Menu
10:36:26 HTP: Console
10:36:31 MQT: tele/wemos7/STATE = {"Time":"2019-07-11T10:36:31","Uptime":"0T00:00:19","Heap":15,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":255,"POWER":"OFF","Wifi":{"AP":1,"SSId":"indebuurt1","BSSId":"24:D3:F2:97:C0:A1","Channel":1,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:36:31 MQT: domoticz/in = {"idx":201,"nvalue":0,"svalue":"39","Battery":200,"RSSI":7}
10:36:31 MQT: tele/wemos7/SENSOR = {"Time":"2019-07-11T10:36:31","BH1750":{"Illuminance":39}}
```
What is the output of command ``i2cscan``

xbmcnut, 2019-07-11:
I did a test with ESPEasy and that found a whole bunch of addresses (dozens) in the I2Scan but it picks only one to use, 0x23 the default. But even that does not seem to work right. The sensor value initially bounced around a bit but then just got stuck on one value. Maybe the new sensor I got today is no good?

I'll reflash with Tasmota in the morning and report the I2C values found.

Jason2866, 2019-07-11:
Just tested with my BH1750. Works as it should. Cant reproduce your issue.
So your sensor or setup is faulty.

ascillato2, 2019-07-11:
Closing issue as seems to be a hardware issue. Sorry.

If you need further assistant, please, do not hesitate on asking here or in the Tasmota Support Chat. Thanks.

closed 2019-07-11 by ascillato2

xbmcnut, 2019-07-12:
FYI, here is the result of the i2cscan. Faulty sensor? Tried this time with NodeMCU with proper SDA and SCL ports.

```
01:55:39 CMD: i2cscan
01:55:39 RSL: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x01 0x02 0x03 0x04 0x05 0x06 0x07 0x08 0x09 0x0a 0x0b 0x0c 0x0d 0x0e 0x0f 0x10 0x11 0x12 0x13 0x14 0x15 0x16 0x17 0x18 0x19 0x1a 0x1b 0x1c 0x1d 0x1e 0x1f 0x20 0x21 0x22 0x23 0x24 0x25 0x26 0x27 0x28 0x29 0x2a 0x2b 0x2c 0x2d 0x2e 0x2f 0x30 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38 0x39 0x3a 0x3b 0x3c 0x3d 0x3e 0x3f 0x40 0x41 0x42 0x43 0x44 0x45 0x46 0x47 0x48 0x49 0x4a 0x4b 0x4c 0x4d 0x4e 0x4f 0x50 0x51 0x52 0x53 0x54 0x55 0x56 0x57 0x58 0x59 0x5a 0x5b 0x5 ...
```

joba-1, 2019-07-12:
Really not much room for doubt.
And given the price of that sensor I‘d just order 2 more and try...

xbmcnut, 2019-07-12:
Agreed and done!

xbmcnut, 2019-09-06:
FYI, I got a replacement sensor today and that sorted it.