arendst/Tasmota #6068: MQTT Epoch Time stamp

pmendiuk, 2019-07-11:
**Have you look for this feature in other issues and in the wiki?**
I have looked in the wiki and I asked about MQTT Epoch time stamps in the forum.

**Is your feature request related to a problem? Please describe.**
The weewx MQTT extension requires Epoch time stamps for sensor data.  I would like to have Tasmota generate the MQTT payload with Epoch time stamps directly rather than republish the sensor packets.

**Describe the solution you'd like**
I would like to have the ability to choose Epoch time time stamps for publishing MQTT topics.

**Describe alternatives you've considered**
I considered modifying the weewx MQTT extension to accept the existing Tasmota MQTT time stamp format however weewx uses epoch time for most sensor data.  As an interim solution I  have been republishing MQTT packets with Epoch time stamps.  I then started looking at the Tasmota code and would like to suggest the following additions:

my_user_config.h
```
// -- MQTT - TimeStamp ----------------------------
#define MQTT_TIMESTAMP_EPOCH   0                 // [MQTT_TIMESTAMP_EPOCH] MQTT Time Stamp (0 = DT_LOCAL, 1 = EPOCH)
```
sonoff.h

`enum GetDateAndTimeOptions { DT_LOCAL, DT_UTC, DT_RESTART, DT_ENERGY, DT_UTC_EPOCH };`

sonoff.ino
```
bool MqttShowSensor(void)
{
  #if MQTT_TIMESTAMP_EPOCH
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s{\"" D_JSON_TIME "\":%s"), mqtt_data, GetDateAndTime(DT_UTC_EPOCH).c_str()); 
  #else
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s{\"" D_JSON_TIME "\":\"%s\""), mqtt_data, GetDateAndTime(DT_LOCAL).c_str());
  #endif
```

support_rtc.ino
```
String GetDateAndTime(uint8_t time_type)
{
  // "2017-03-07T11:08:02-07:00" - ISO8601:2004
  uint32_t time = local_time;

  switch (time_type) {
    case DT_ENERGY:
      time = Settings.energy_kWhtotal_time;
      break;
    case DT_UTC:
      time = utc_time;
      break;
    case DT_RESTART:
      if (restart_time == 0) {
        return "";
      }
      time = restart_time;
      break;
    case DT_UTC_EPOCH:
      time = utc_time;
      char str_utc_time[33];
      itoa(utc_time, str_utc_time, 10);
      return str_utc_time;
  }
```

Thanks,
Patrick

arendst, 2019-07-11:
How about adding the epoch as an optional JSON field like:
```
11:05:10 MQT: tele/wemos4/STATE = {"Time":"2019-07-11T11:05:10","Epoch":1562838655,"Uptime":"0T00:00:16","Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
```

Jason2866, 2019-07-11:
Adding Epoch time would be nice. There are now and then users asking for in Discord

ascillato2, 2019-07-11:
Closing this issue as has been added.

Thanks.

closed 2019-07-11 by ascillato2

pmendiuk, 2019-07-11:
The epoch value pair won't be recognized by weewx.  The time value pair needs to contain the epoch time.