arendst/Tasmota #6072: not all selected values will be send to the knx - bus

schlaubi666, 2019-07-11:
i use sonoff pow r2 with tasmota 6.6.0 and configured knxto send:

power
Energy Today
Energy Yesterday
Energy Total

 to seperate knx GA's. 

Console say that it was send:
16:02:41 KNX: Power sent to 0.7.21 
16:02:41 KNX: Energy Today sent to 0.7.23 
16:02:41 KNX: Energy Total sent to 0.7.24 
16:02:41 KNX: Energy Yesterday sent to 0.7.22 
16:03:46 RSL: sonoffPOWr2_01/tele/SENSOR = {"Time":"2019-07-11T16:03:41","ENERGY":{"TotalStartTime":"2019-07-06T20:26:32","Total":13.731,"Yesterday":4.451,"Today":1.907,"Period":1,"Power":194,"ApparentPower":200,"ReactivePower":48,"Factor":0.97,"Voltage":237,"Current":0.840}}

 If i open the busmonitor in my ETS it only shows that power and Energy Today was send, but not the rest . 

Is this a known problem ?

ascillato, 2019-07-11:
Hi,

I tested and works fine but with the KNX communication enhancement turned off.

As the troubleshooting template is missing I can't know if you have that enabled or not.

Please, turn KNX communication enhancement to off and try again.

I will fix the KNX communication enhancement to work properly also for KNX_TX slots. Thanks for reporting.

ascillato, 2019-07-11:
FYI:

The bug is just too much telegrams at once if using **KNX communication enhancement**.

The **KNX communication enhancement** sends 3 telegrams for every data, avoiding the drop out of multicast data of some wifi routers (Some wifi routers dismiss some multicast transmissions if they are busy. With this enhancement, this action is not performed).

The code structure of Tasmota is

Sensor -> Sensor Driver -> KNX Driver -> KNX IP Library -> WIFI UDP Library of Arduino -> SDK -> Wifi

So, what is happening is that the WIFI UDP Arduino library has a very limited buffer for all UDP packets, so when trying to send those 4 telegrams multiplied by 3 of the Enhancement, the KNX driver was trying to send 12 telegrams. The UDP buffer can manage at once about 5 to 6 telegrams (depending on the size), so that is why you saw just 2 sensor data.

Disabling the **KNX communication enhancement** will make the driver to just send 4 telegrams and that will work without issue.

I have to add an interim buffer in the KNX Driver so as to manage this case with the enhancement.
Will do later. Thanks.

ascillato2, 2019-07-13:
@schlaubi666 

Have you tested without the Enhancement enabled?

ascillato2, 2019-07-18:
Closing as there is no feedback.

Support for enhanced communication will be added soon for knx tx slots

closed 2019-07-18 by ascillato2