arendst/Tasmota #6073: Problem compiling with 2.3.0 core Atom/PlatformIO

andreamastro, 2019-07-11:
I have this problem when i try to compile with 2.3.0 core:
Linking .pio\build\sonoff\firmware.elf
c:/users/mastro/.platformio/packages/toolchain-xtensa@1.40802.0/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\sonoff\firmware.elf sect
ion `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\sonoff\firmware.elf] Error 1
========================= [ERROR] Took 50.17 seconds =========================
 
================================== [SUMMARY] ==================================
Environment sonoff-minimal      [IGNORED]
Environment sonoff-basic        [IGNORED]
Environment sonoff-classic      [IGNORED]
Environment sonoff-knx          [IGNORED]
Environment sonoff-sensors      [IGNORED]
Environment sonoff-display      [IGNORED]
Environment sonoff-BG           [IGNORED]
Environment sonoff-BR           [IGNORED]
Environment sonoff-CN           [IGNORED]
Environment sonoff-CZ           [IGNORED]
Environment sonoff-DE           [IGNORED]
Environment sonoff-ES           [IGNORED]
Environment sonoff-FR           [IGNORED]
Environment sonoff-GR           [IGNORED]
Environment sonoff-HE           [IGNORED]
Environment sonoff-HU           [IGNORED]
Environment sonoff-IT           [IGNORED]
Environment sonoff-KO           [IGNORED]
Environment sonoff-NL           [IGNORED]
Environment sonoff-PL           [IGNORED]
Environment sonoff-PT           [IGNORED]
Environment sonoff-RU           [IGNORED]
Environment sonoff              [FAILED]
=================== 1 failed, 0 succeeded in 50.17 seconds ===================
Environment sonoff-SE           [IGNORED]
Environment sonoff-SK           [IGNORED]
Environment sonoff-TR           [IGNORED]
Environment sonoff-TW           [IGNORED]
Environment sonoff-UK           [IGNORED]

Jason2866, 2019-07-11:
It is a strange setup error of the build environment.
I do have the same error when trying to compile with gitpod
Doing absolute the same on my local PC (with VSC / platformio) it compiles just fine!

andreamastro, 2019-07-11:
Tried with VSC. Same problem
[image]

Jason2866, 2019-07-11:
It does work for me with VSC
[image]

Jason2866, 2019-07-11:
What CPU do you have? Your compile time is only 49 seconds!
My i5 needs 106 seconds

andreamastro, 2019-07-11:
i5 6500

Jason2866, 2019-07-11:
Okay, i think it is 2 times faster....  I am out of ideas what the problem could be!

riotm0de, 2019-07-11:
same here, cant compile with ATOM!

gemu2015, 2019-07-11:
had the same  problem 
deleting  the folder .platformio fixed it

andreamastro, 2019-07-11:
Already tried. Doesn't work for me

ascillato, 2019-07-11:
Hi,

In my environment latest Tasmota (as default) is compiling fine with core 2.3.0

Have you enabled something else in my_user_config.h file?

riotm0de, 2019-07-11:
> Have you enabled something else in my_user_config.h file?

i did nothing beside of inserting " SSID + PASS " 
Still no success :(

curzon01, 2019-07-11:
I'd the same issue 2 weeks ago and it had nothing to do with own config or just used a fresh git clone.
The reason was the platformio install with installed libs.

 I didn't dig deep into the real issue, but my previous not working platformio environment was based on several platformio update/upgrade and displayed meantime 4.x 

I removed platformio completely, do again the steps written here in Wiki installed a fresh platformio. Now platformio is still v3.x even after command line update/upgrade (currently can not show real version, don't have access to my PC) and the error above is gone.

May be someone helps...

riotm0de, 2019-07-11:
yesterday platformio was released with core 4
i already tried to uninstall all and did a new fresh install of VSC and Platformio, no success :( 

please help

ascillato, 2019-07-11:
@curzon01

That makes sense. I can compile without issue and my Platformio is v3.6.7

Gitpod uses latest PIO also. So the problem seems to rely on platformio 4.

I was reading the release notes and they change several things like workdir

Jason2866, 2019-07-11:
@curzon01  Can confirm i am on Platformio v.3.6.7 too.
Gitpod uses v.4.x

riotm0de, 2019-07-11:
so is there a quick fix for me or do i need to wait for an update ? thanks

arendst, 2019-07-11:
I'm also still using 3 after my bad experience with a core 4 beta.

See docs here https://docs.platformio.org/en/latest/migration.html?utm_source=platformio&utm_medium=mail&utm_campaign=pio-core-4

I love this 

> PlatformIO Core 4.0 is fully backward compatible with v3.x. 

where it's obviously not...

ascillato2, 2019-07-11:
> so is there a quick fix for me or do i need to wait for an update ? thanks

Quick? You can compile in a different core than 2.3.0

arendst, 2019-07-11:
What might work is executing the following commands from a terminal:
```
pip unistall platformio
pip install platformio==3.6.7
```

Jason2866, 2019-07-12:
@arendst  Your suggestion does work! :-)
Just did PR #6078 to get gitpod working again with core 2.3.0

Jason2866, 2019-07-12:
Opened issue on PlatfrmIO Github https://github.com/platformio/platformio-core/issues/2780

nhorvath, 2019-07-12:
> What might work is executing the following commands from a terminal:
> 
> ```
> pip uninstall platformio
> pip install platformio==3.6.7
> ```

I made the mistake of upgrading pio this morning which resulted in the same error as the OP. This fixed my issue, but do we know why tasmota won't compile with PIO 4 (ie. is it a bug with tasmota or pio)?

arendst, 2019-07-12:
it's a bug in pio. Analysis of .map files shows huge usage of Iram. I guess a toolchain issue.

ivankravets, 2019-07-13:
Friends, please don't panic 🙏 

> I'm also still using 3 after my bad experience with a core 4 beta.

@arendst, next time please report any issues to https://github.com/platformio/platformio-core/issues
Do you still have any problems with PIO Core 4.0?

> I love this
>> PlatformIO Core 4.0 is fully backward compatible with v3.x.
>
> where it's obviously not...

Why it's not? Please check the "Compatibility" section in the migration guide:

> PlatformIO Core 4.0 is fully backward compatible with v3.x. The only major change is a new location for project build artifacts and library dependencies. The previous **.pioenvs** (build_dir) and .piolibdeps (libdeps_dir) folders were **moved to a new workspace_dir**.

-----------

Let me explain now. Espressif 8266 dev/platform uses own patched LS Scripts because existing were developed strictly for Arduino IDE build system and depends on it "tmp build object path". Later, together with Espressif Team, we adapted them to be compatible between Arduino IDE/PlatformIO. So, the latest versions of dev/platform (starting from v1.6.0) use original LD scripts from Arduino Core.

See details https://github.com/platformio/platform-espressif8266/blob/v1.5.0/ldscripts/esp8266.flash.common.ld#L3

Please note that we added [a special "warning" block](https://github.com/platformio/platform-espressif8266/releases/tag/v1.5.0) to this release and explained how to configure it properly for PlatformIO Core 4.0.

ivankravets, 2019-07-13:
I forgot to add. The current [platformio.ini](https://github.com/arendst/Sonoff-Tasmota/blob/development/platformio.ini) file looks so complicated for editing. It's actually our fault. However, in PlatforIO Core 4.0 we introduced [Extra Configuration Files](http://docs.platformio.org/en/latest/projectconf/section_platformio.html#extra-configs). Using them, we could significantly simplify this config and split whole logic into Arduino Cores. For example,

```ini
[platformio]
src_dir = sonoff

; *** Uncomment one of the lines below to build/upload only one environment
default_envs = sonoff 

extra_configs = 
   pio_core_2_3_0.ini

[env]
framework = arduino
board = xxx ; edit me

[env:sonoff]
build_flags =
  ${env.build_flags}
   -DSOME_CUSTOM_FLAGS
...

```

Where ` pio_core_2_3_0.ini` can exten [Global `env` Scope](http://docs.platformio.org/en/latest/projectconf/section_env.html#global-scope-env) data:

```
[env]
platform = espressif8266@~1.5.0
framework = arduino
```

See also different test use cases https://github.com/platformio/platformio-core/blob/develop/tests/test_projectconf.py

Jason2866, 2019-07-13:
The warning is not easy to find. The warning should be added to the actual release...

Jason2866, 2019-07-13:
Solved with PR #6087

closed 2019-07-13 by ascillato2

arendst, 2019-07-15:
@ivankravets thx for your insights and sorry for my blunt workaround ;-)

As of today I am magically using core 4.0.0 (It seems it installed itself as I was still using 3.5.6 yesterday...) and it works as it should.

ivankravets, 2019-07-16:
> thx for your insights and sorry for my blunt workaround ;-)
@arendst, everything is OK! Next time just ping me here or file a separate issue. PlatformIO Core 4.0 is very stable now and a lot of companies use it in production.

In any case, thank you so much for using PlatformIO for your project. It helps us to get better adoption in the market. Thanks again! 😊

smilexth, 2019-08-14:
How to install specific version of platformio in vscode for windows ?

ivankravets, 2019-08-14:
@smilexth why do you need it? The latest should work for old projects too.

ascillato, 2019-08-14:
Latest platform.io works really good.

lobocobra, 2019-10-19:
I had the same problem, as I wanted just to recompile an old code. I ended up with an broken CSV setup after the autoupdate, Below I summarized what I needed to do (includes what is written above).

1) reset Visual studio Code
    Open Run (Win + R)
    Enter %appdata%
    Press Enter
    Delete the folder Code.
2) Turns out the extensions are stored under %USER%.vscode\extensions. Deleting that gets rid of them. %USER%\\.vscode\extensions (or) %USERPROFILE%\.vscode\extensions
3) Install python 2.7: https://www.python.org/downloads/release/python-2716/
   => !! add the option to add python to the path.... REBOOT Windows
4) Open VSC install 
   pip-packages (not yet platformio)
5) open console and enter....
    pip install platformio==3.6.7
6) now install in VSC platformio
7) delete in the GIT directory of your code the .pio and .vscode (if there)
8) now add your working directory to the workspace and complie

=> Are there easier ways? Well maybe... but this one worked (above pip uninstall failed for me and ended in a broken setup).

ivankravets, 2019-10-19:
@lobocobra why do you need all these steps? why you don't use the latest PlatformIO Core 4.0? Why you don't use the latest Python 3?

@arendst do we have any problems with PlatformIO Core 4.0 and Python 3? Python 2.7 ends in December, we should move to Python 3 soon. It would be good if people start installing Python 3.  On Windows we install the latest Python 3 by default.

lobocobra, 2019-10-19:
The problems are solved if you use the new 6.6 firmware. 
=> But I needed to compile a code 6.5 which was modified (not yet ported to the lastest version)

If you want to use an old platformio version 3.6.7 you need Python 2.7.
=> I do not recommend to others what I did here, but it solved my problem, which was caused by a silly and unwnated autoupdate of platformio.

ivankravets, 2019-10-19:
But why do you need  PIO Core 3.6.7 ? PIo Core 4.0 is backward compatible with PIO Core 3 projects

ascillato, 2019-10-19:
@ivankravets

Latest Platform.io with latest python on VSC is working fine for me with latest Tasmota. In my setup, there was no need to use old python.

ivankravets, 2019-10-19:
@lobocobra It seems that I remember this issue. You need VERY OLD version ESP8266 dev/platform which depends on `.pioenvs`. See solution here https://github.com/platformio/platform-espressif8266/releases/tag/v1.5.0

You can switch back to PIo Core 4 and Python 3

Jason2866, 2019-10-19:
@lobocobra  Platformio 4 compiles fine with older Tasmota version(s) on all cores
Your setup of platformio.ini is probably not correct

ascillato, 2019-10-19:
@lobocobra 

It is not recommended to use old arduino cores due to security issues. See #6666

lobocobra, 2019-10-19:
@ivankravets
Thanks for your suggestions. I tried them and upgrade to the latest version. I also used a new platformio.ini (one that I use for my recent builds)

It ends always with: 
`
Copy/sonoff/xdrv_interface.ino:191:1: error: invalid conversion from 'boolean (*)(byte) {aka unsigned char (*)(unsigned char)}' to 'bool (*)(uint8_t) {aka bool (*)(unsigned char)}' [-fpermissive]
`
I made extra an portable VSC installation for this older firmware and platformio upgraded without asking to 4.03, which broke all. I do not understand why they do those upgrades without asking.
=> My solution above works somehow and I agree that it is not good solution.

@ascillato
Thanks for the tip. I will upgrade the code later, till then I life with the risk. Network is anyhow secured.

ascillato, 2019-10-19:
>I will upgrade the code later

Tasmota codebase and precompiled bins are already updated.

lobocobra, 2019-10-19:
Thanks but the new code contains part of my Shutter code I made last year, but not all the options I need. 
=> Thus I need to use a compiled version. 

Only Issue I have is related to PlatformIO upgrading itself without asking.

ascillato, 2019-10-19:
If you need other options for shutters that are not available in Tasmota, please, if you have time, do a PR to Tasmota in order to add them. Thanks.

lobocobra, 2019-10-20:
> 
> 
> But why do you need PIO Core 3.6.7 ? PIo Core 4.0 is backward compatible with PIO Core 3 projects

No it's not backwards compatible... if I compile the same code with
- 4.0.3 it throws errors
- 3.6.7 it works
Which means it is not backwards compatible. I have an few month old code, which I bugfix regularily and update it to the latest firmware of Arendst from time to time.

ivankravets, 2019-10-20:
Please provide a simple PlatformIO project to reproduce this issue.

johny-mnemonic, 2019-10-20:
@Jason2866 
> @lobocobra Platformio 4 compiles fine with older Tasmota version(s) on all cores
> Your setup of platformio.ini is probably not correct

Could you please reference a correct platfromio.ini to be able to build with 4.x ?
As building v6.5 with my modified platformio.ini failed, I tried to checkout v6.6, used default platformio.ini and I tried building with all cores (2.3.0, 2.4.2 and 2.5.2), but all were failing:-(
I also tried with adding "build_dir = .pioenvs", but it didn't help, just produced different error.

Downgrading to 3.6.7 fixed it though...

ivankravets, 2019-10-20:
@Jason2866 could you help here? I remember you did great research about this issue before. Thanks!

Jason2866, 2019-10-20:
@johny-mnemonic  use platformio.ini from actual development version of Tasmota
https://github.com/arendst/Sonoff-Tasmota/blob/development/platformio.ini

rosariopomillo, 2020-02-28:
Hi all,
after recent Win10 update I can't build Tasmota in vsCode nor in Atom. I'm not an expert but I imagine this is a Platformio issue. Tried to reinstall all packages but problem persists. The build outcome is:
**Can not remove temporary directory c:\...\Tasmota-8.1.0\.pioenvs. Please remove it manually to avoid build issues**
well: there is no .pioenvs directory, and
**Error: Invalid c:\...\Tasmota-8.1.0\platformio.ini (project configuration file): 'No option 'platform_packages' in section: 'common'**
which is too much for me!
Removed customization files and tried with a fresh Tasmota-8.1.0 download: no way.
How can I solve?
Thanks in advance

Jason2866, 2020-02-28:
You have probably updated PlatfformIO to latest version. There are breaking changes.
To compile the release version of Tasmota with latest Platformio use the Platformio config files
from the development branch (`platformio.ini` `platformio_tasmota_env.ini` and if you use the override file this too `platformio_override_sample.ini`).

Jason2866, 2020-02-28:
@rosariopomillo  We have updated the PlatformIO ini files in Tasmota master branch.
Just do a fresh git clone or download again.

rosariopomillo, 2020-02-28:
Very helpful, thanks a lot!