arendst/Tasmota #6074: IrRemote Repeat & Delay

marcelkordek, 2019-07-11:
Hi,

is it possible to repeat IR commands?
Some of my IR devices require the code to be sent longer than others.

Maybe something like that (xdrv_05_irremote.ino)?

```
int pulse = root[UpperCase_P(parm_uc, PSTR(D_JSON_IR_PULSE))];
int repeat = root[UpperCase_P(parm_uc, PSTR(D_JSON_IR_REPEAT))];
int rdelay = root[UpperCase_P(parm_uc, PSTR(D_JSON_IR_RDELAY))];
int pdelay = root[UpperCase_P(parm_uc, PSTR(D_JSON_IR_PDELAY))];

if (pulse <= 0)
    pulse = 1; // Make sure pulse isn't 0
if (repeat <= 0)
    repeat = 1; // Make sure repeat isn't 0
if (pdelay <= 0)
    pdelay = 100; // Default pdelay
if (rdelay <= 0)
    rdelay = 1000; // Default rdelay

if (protocol && bits) {
    char protocol_text[20];
    int protocol_code = GetCommandCode(protocol_text, sizeof(protocol_text), protocol, kIrRemoteProtocols);

    snprintf_P(log_data, sizeof(log_data), PSTR("IRS: protocol_text %s, protocol %s, bits %d, data %u (0x%lX), protocol_code %d"), protocol_text, protocol, bits, data, data, protocol_code);
    AddLog(LOG_LEVEL_DEBUG);

    // Repeat Loop
    for (int r = 0; r < repeat; r++) {
        // Pulse Loop
        for (int p = 0; p < pulse; p++) {
            switch (protocol_code) {
                case NEC:
                    irsend - > sendNEC(data, (bits > NEC_BITS) ? NEC_BITS : bits);
                    break;
                case SONY:
                    irsend - > sendSony(data, (bits > SONY_20_BITS) ? SONY_20_BITS : bits, 2);
                    break;
                case RC5:
                    irsend - > sendRC5(data, bits);
                    break;
                case RC6:
                    irsend - > sendRC6(data, bits);
                    break;
                case DISH:
                    irsend - > sendDISH(data, (bits > DISH_BITS) ? DISH_BITS : bits);
                    break;
                case JVC:
                    irsend - > sendJVC(data, (bits > JVC_BITS) ? JVC_BITS : bits, 1);
                    break;
                case SAMSUNG:
                    irsend - > sendSAMSUNG(data, (bits > SAMSUNG_BITS) ? SAMSUNG_BITS : bits);
                    break;
                case PANASONIC:
                    irsend - > sendPanasonic(bits, data);
                    break;
                default:
                    snprintf_P(mqtt_data, sizeof(mqtt_data), S_JSON_COMMAND_SVALUE, command, D_JSON_PROTOCOL_NOT_SUPPORTED);
            }
            if (p + 1 < pulse)
                delay(pdelay);
        }
        if (r + 1 < repeat)
            delay(rdelay);
    }
} else {
    error = true;
}
```
Thanks.

Regards,
Marcel

marcelkordek, 2019-07-11:
And for MQTT Commands:

```
IRsend {"Protocol":"NEC","Bits":32,"Data":2170978686,"Pulse":1,"Repeat":1,"RDelay":100,"PDelay":1000}
```
Regards,
Marcel

Jason2866, 2019-07-11:
Sending it twice via command backlog and delay?

ascillato2, 2019-07-11:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-11 by ascillato2

kobuki, 2019-08-01:
Would it be possible to add the same functionality for raw codes? My AC for example requires raw commands to be used, but about 1 in 10 remote command doesn't get through. It would be more robust if it was possible to add repeat/long press parameters similarly to the solution in the [PR](https://github.com/arendst/Sonoff-Tasmota/pull/6096) above. I can open a new issue if needed.

s-hadinger, 2019-08-01:
Makes sense. Will take a look.

s-hadinger, 2019-08-04:
@kobuki I have submitted PR #6176

kobuki, 2019-08-04:
@s-hadinger thanks, I was notified of the PR. How is it possible to set the inter-message delay? Some devices have a specific requirement regarding that.

s-hadinger, 2019-08-04:
Are you using IRsend with timing pulses or IRSend Raw with binary bitstream? If the later, I need to add a parameter for inter-message.

kobuki, 2019-08-04:
I'm using the latter - binary codes. Just a hint though, wouldn't it have been simpler to change not the command name, but the raw keyword that accepts additional parameterst? Something like:

`IRSend raw2,<repeat>,<repeatdelay>,0,8951,4543,593,542,1654,110000010110000000000001010000010000000000000000`

I admit I'm not very familiar with the internals of the code, adding an index-like number to the IRSend param (IRSend<X>) like how you implemented now, might be simpler.

arendst, 2019-08-04:
The chosen solution has the advantage of backward compatibility while allowing to have repeat for both ``IrSend <raw>`` and ``IrSend <freq>``.

kobuki, 2019-08-04:
I understand, though `IRSend raw2, ...` or similar change would't break that compatibility with `IRSend raw, ...`.

arendst, 2019-08-04:
Correct, but it wouldn't add repeat to ``IrSend <freq>...`` where you might want it too.

s-hadinger, 2019-08-04:
Actually `IRSend<x>` comes for free in Tasmota, `<x>` is already in `XdrvMailbox.index`

I'm reluctant to add a parameter to `IRSend raw...` because of backwards compatibility. Instead I propose to insert a computed inter-message pause of `3 * (header_mark + header_space)`. Comparing to existing protocols, il looks like a safe value.

kobuki, 2019-08-04:
Would something like passing a negative value after the frequency value work? Or some value that's too big to be a mark/space length, like 100000+pause_length from which the actual pause could be computed. I'd be fine with the latest proposed solution, it's simple and if easy to implement, all the better.

s-hadinger, 2019-08-04:
In your case, I'm not sure the inter-message gap has any importance. You're just covering the fact that the first message was not well received. It's not even equivalent to a long press.

In your case, it would compute to 40.5ms inter-message gap. Is this value suitable?

s-hadinger, 2019-08-04:
Before you ask, the current code uses `uint16_t` for values, so no negatives nor >65535 values.

kobuki, 2019-08-04:
I can only test it by converting to raw timing values. I'll attempt that later if you need me to, but I think 40.5 ms would be adequate. It's the same as used by the NEC protocol IIRC. Actually the custom AC IR proto is very similar with similar timings but with 48 bits of data so I can't send them as NEC (Tasmota cuts off the extra bits).

s-hadinger, 2019-08-04:
Ah, ok. It definitely looked like NEC encoding.

@arendst I'm pushing a complement to the PR.

PR #6096 (2019-07-15): Support IRSend long press ('repeat' feature from IRRemoteESP8266)
Some IR devices need a longer signal to be sent, equivalent to a long press on the remote controller.

IRRemoteESP8266 already supports the 'repeat' parameter, I added it to the JSON.

Now you can either send:

`IRsend { "protocol": "SAMSUNG", "bits": 32, "data": 551502015 }`

or

`IRsend { "protocol": "NEC", "bits": 32, "data":"0x02FDFE80", "repeat": 2 }`

By default `repeat` is 0 which means 'sent once', except when the protocol expects a minimal repeat (ex: SONY minimum repeat is 2).

No change to previous versions if you don [... cut]
PR #6176 (2019-08-04): Add allow repeat/longpress for IRSend raw, introduced IRSend<r> option
Description:

Requested by @kobuki in #6074 
Allows to send repeats/long press with IRSend raw in addition to IRSend json.

Introduced a new parameter: `IRSend<x> ...` where `<x>` indicates the number of times the message should be sent.

Warning: `IRSend1` means the command is sent once, which is equivalent to `"repeat" = 0`. `IRSend2` is equivalent to `"repeat" = 1`. 

If both ÌRSend<x>` and `"repeat"= <r>` are specified, the greater of both is used, i.e. either `<x>` or `<r>+1`.

**Related issue (if applicable):** fixes  [... cut]
PR #6178 (2019-08-04): Add inter-message gap for IRSend raw
## Description:

Complement to PR #6176 

This PR inserts inter-message gaps for `IRSend raw,...`.

If `header_mark` and `header_space` are specified, the gap will be computed as `(header_mark + header_space) * 3` with a high limit of 65ms to avoid 16 bits overflow.

If `header_mark` and `header_space` are not specified, the gap will be 40ms.

**Related issue (if applicable):** fixes #6074

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  -  [... cut]