arendst/Tasmota #6080: CMD dimmer increase does not work

matrix1000, 2019-07-12:
Hi,

The dimmer value increase with "+" does not work. 
The dimmer value decrease with "-" works fine.
Module: ARILUX® SL-LC 06
Tasmota Version: 6.6.0
LED Strip: RGB CW WW

For example:
Starting with:
curl --silent 'http://192.168.0.234/cm?cmnd=dimmer%2050'
RESULT: {"POWER":"ON","Dimmer":50,"Color":"0000000080","HSBColor":"0,0,0","Channel":[0,0,0,0,50],"CT":500}
-> set to 50%  Works fine.
curl --silent 'http://192.168.0.234/cm?cmnd=dimmer%20-'
RESULT: {"POWER":"ON","Dimmer":40,"Color":"0000000066","HSBColor":"0,0,0","Channel":[0,0,0,0,40],"CT":500}
-> decrease with "-" works fine too

curl --silent 'http://192.168.0.234/cm?cmnd=dimmer%20+'
RESULT: {"Dimmer":40}
-> increase with "+" does not work.

ascillato, 2019-07-12:
Hi,

A + symbol can't be passed just like that. you need to encode it the same way you enconde the space (%20)

The command dimmer + should be passed as:

http://192.168.0.234/cm?cmnd=dimmer%20%2b

arendst, 2019-07-12:
You need to do more debugging before initiating an issue.

Try de gui which will work just fine

Learn how to send html commands; you most likely will need to change the "+" into "%2B"

closed 2019-07-12 by ascillato2

matrix1000, 2019-07-12:
Many Thanks,

yes, that solves my problem. :-)